riscv-collab/riscv-gnu-toolchain #1415: The unimp instruction occurred while compiling the v instruction, causing QEMU to report an error: an illegal instruction error

24geng, 2024-01-31:
Can someone please tell me why that is? I would appreciate it!

My compile options:
`clang++ -g --target=riscv64-unknown-linux-gnu -march=rv64imafdcv -mrvv-vector-bits=128 20230329.cpp -o 20230329 -stdlib=libc++`

`00000000000016f8 <_start>:
    16f8: ef 00 20 02  	jal	0x171a <load_gp>
    16fc: aa 87        	mv	a5, a0
    16fe: 17 f5 00 00  	auipc	a0, 15
    1702: 03 35 25 a2  	ld	a0, -1502(a0)
    1706: 82 65        	ld	a1, 0(sp)
    1708: 30 00        	addi	a2, sp, 8
    170a: 13 71 01 ff  	andi	sp, sp, -16
    170e: 81 46        	li	a3, 0
    1710: 01 47        	li	a4, 0
    1712: 0a 88        	mv	a6, sp
    1714: ef f0 df d9  	jal	0x14b0 <.plt+0xe0>
    1718: 02 90        	ebreak

000000000000171a <load_gp>:
    171a: 97 f1 00 00  	auipc	gp, 15
    171e: 93 81 61 0e  	addi	gp, gp, 230
    1722: 82 80        	ret
    1724: 00 00        	unimp

    c444: 13 01 01 fc  	addi	sp, sp, -64
    c448: 23 3c 11 02  	sd	ra, 56(sp)
    c44c: 23 38 81 02  	sd	s0, 48(sp)
    c450: 13 04 01 04  	addi	s0, sp, 64
    c454: 13 01 01 fe  	addi	sp, sp, -32
    c458: 23 34 a4 fc  	sd	a0, -56(s0)
    c45c: 23 30 b4 fc  	sd	a1, -64(s0)
    c460: 03 35 84 fc  	ld	a0, -56(s0)
    c464: 83 35 04 fc  	ld	a1, -64(s0)
    c468: 57 70 02 cd  	vsetivli	zero, 4, e32, m1, ta, ma
    c46c: 07 e4 05 02  	vle32.v	v8, (a1)
    c470: 23 30 a4 fe  	sd	a0, -32(s0)
    c474: 93 05 04 fa  	addi	a1, s0, -96
    c478: 27 84 85 02  	vs1r.v	v8, (a1)
    c47c: 13 05 40 00  	li	a0, 4
    c480: 23 2e a4 fc  	sw	a0, -36(s0)
    c484: 03 25 c4 fd  	lw	a0, -36(s0)
    c488: 57 75 05 0d  	vsetvli	a0, a0, e32, m1, ta, ma
    c48c: 23 38 a4 fc  	sd	a0, -48(s0)
    c490: 03 35 04 fe  	ld	a0, -32(s0)
    c494: 07 e4 85 02  	vl1re32.v	v8, (a1)
    c498: 83 35 04 fd  	ld	a1, -48(s0)
    c49c: 57 f0 05 0d  	vsetvli	zero, a1, e32, m1, ta, ma
    c4a0: 27 64 05 02  	vse32.v	v8, (a0)
    c4a4: 73 10 00 c0  	unimp`

TommyMurphyTM1234, 2024-01-31:
Insufficient info provided.
How did you build your toolchain?
What is your code, or at least a small self-contained reproducible test case?
How are you running QEMU?

Since this seems to be an issue with QEMU why aren't you reporting it upstream where the development is actually done?

* https://www.qemu.org/
  * https://www.qemu.org/contribute/

TommyMurphyTM1234, 2024-01-31:
According to this:

* https://luplab.gitlab.io/rvcodecjs/#q=0xc0001073&abi=false&isa=AUTO

This instruction:
> ```
> c4a4: 73 10 00 c0  	unimp
> ```

when formatted to take account of endianess (i.e. `0xc0001073`) decodes to:
```
csrrw x0, cycle, x0
```
So maybe your toolchain or QEMU wasn't built/run to include support for the `Zicsr` ISA extension?

TommyMurphyTM1234, 2024-02-02:
Any update @24geng?

TommyMurphyTM1234, 2024-02-02:
I'm going to assume that the issue has been explained above and am closing it.
If that's not the case then please post back with more details and reopen it.