riscv-collab/riscv-gnu-toolchain #1419: qemu not available

R080R1d3R, 2024-02-02:
not able to install qemu the webpage doesn't exist 

fatal: unable to connect to github.com:
github.com[0: 20.207.73.82]: errno=Connection timed out

TommyMurphyTM1234, 2024-02-02:
You'll need to provide a bit of context and detail.
What exactly are you trying to do?
What commands are you running?
Any logfiles?
For what it's worth https://github.com/is working fine from my location (Ireland).

TommyMurphyTM1234, 2024-02-02:
FWIW [this](https://www.qemu.org/download/) worked fine for me just now:
```
git clone https://gitlab.com/qemu-project/qemu.git
Cloning into 'qemu'...
remote: Enumerating objects: 728678, done.
remote: Counting objects: 100% (17055/17055), done.
remote: Compressing objects: 100% (4483/4483), done.
remote: Total 728678 (delta 12547), reused 17003 (delta 12533), pack-reused 711623
Receiving objects: 100% (728678/728678), 315.90 MiB | 4.56 MiB/s, done.
Resolving deltas: 100% (606027/606027), done.
Updating files: 100% (10023/10023), done.
```
If you're having problems accessing the QEMU GitHub repo then maybe it's some local configuration/connectivity issue at your end?

R080R1d3R, 2024-02-02:
There is a GitHub link in the repo not a gitlab link

TommyMurphyTM1234, 2024-02-02:
> There is a GitHub link in the repo not a gitlab link

I don't know what you're referring to.
The official QEMU GitLab repo link:

* https://www.qemu.org/download/ - `https://gitlab.com/qemu-project/qemu.git`

is used by this (`riscv-gnu-toolchain`) repo:

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.gitmodules#L24-L26

as also shown here where retrieval of QEMU via the GitLab link works fine for me:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git submodule init qemu
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
git submodule update qemu
Cloning into '/home/user/Downloads/temp/riscv-gnu-toolchain/qemu'...
Submodule path 'qemu': checked out 'f48c205fb42be48e2e47b7e1cd9a2802e5ca17b0'
cd qemu
git config --get remote.origin.url
https://gitlab.com/qemu-project/qemu.git
```
As I said before, you need to explain your issue more clearly - what exactly you're trying to do, how you're trying to do it/what commands you are using. what goes wrong, any error messages/logs etc.

FWIW I also did this and it all worked fine:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1419
cd issue-1419
./configure --prefix=`pwd`/installed-tools --with-sim=qemu
make && make build-sim
ls installed-tools/bin
qemu-riscv32                   riscv64-unknown-elf-c++filt     riscv64-unknown-elf-gcc-ar      riscv64-unknown-elf-gdb            riscv64-unknown-elf-nm       riscv64-unknown-elf-size
qemu-riscv64                   riscv64-unknown-elf-cpp         riscv64-unknown-elf-gcc-nm      riscv64-unknown-elf-gdb-add-index  riscv64-unknown-elf-objcopy  riscv64-unknown-elf-strings
riscv64-unknown-elf-addr2line  riscv64-unknown-elf-elfedit     riscv64-unknown-elf-gcc-ranlib  riscv64-unknown-elf-gprof          riscv64-unknown-elf-objdump  riscv64-unknown-elf-strip
riscv64-unknown-elf-ar         riscv64-unknown-elf-g++         riscv64-unknown-elf-gcov        riscv64-unknown-elf-ld             riscv64-unknown-elf-ranlib
riscv64-unknown-elf-as         riscv64-unknown-elf-gcc         riscv64-unknown-elf-gcov-dump   riscv64-unknown-elf-ld.bfd         riscv64-unknown-elf-readelf
riscv64-unknown-elf-c++        riscv64-unknown-elf-gcc-13.2.0  riscv64-unknown-elf-gcov-tool   riscv64-unknown-elf-lto-dump       riscv64-unknown-elf-run

R080R1d3R, 2024-02-03:
okk i will tell the whole process here i am working on open-piton and using this github repo to install it for ease. 
https://github.com/kevinjohnmartin/nooman-openpiton.git

in this there is script :- https://github.com/kevinjohnmartin/nooman-openpiton/blob/master/piton/design/chip/tile/ariane/ci/build-riscv-gcc.sh
this script is used to install riscv toolchain but running this scripts gives me error in this repo that is :- https://github.com/riscv-collab/riscv-gnu-toolchain in the qemu section. i am attaching the whole log and error screenshot. please help me on this 
[image]
[error.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/14154456/error.txt)

TommyMurphyTM1234, 2024-02-04:
> using this github repo to install it for ease. https://github.com/kevinjohnmartin/nooman-openpiton.git

That script is referencing a `riscv-gnu-toolchain` commit from over 6 years ago!

* https://github.com/kevinjohnmartin/nooman-openpiton/blob/a7cd55cd7db115b5932a11b03fd5261d7366dca8/piton/design/chip/tile/ariane/ci/build-riscv-gcc.sh#L6
* https://github.com/kevinjohnmartin/nooman-openpiton/blob/a7cd55cd7db115b5932a11b03fd5261d7366dca8/piton/design/chip/tile/ariane/ci/build-riscv-gcc.sh#L23
```
commit 691e4e826251c7ec59f883cab18440c87baf45e7
Merge: 0ee6a5e ed957ad
Author: Palmer Dabbelt <palmer@dabbelt.com>
Date:   Thu Dec 21 10:51:45 2017 -0800

    Merge pull request #302 from zongbox/master

    Add enable-obsolete-rpc option for glibc
```
That commit references a stale QEMU repo link (`git://github.com/riscv/riscv-qemu.git`).
Back then local forks of dependent projects like QEMU were used rather than the upstream projects that are used today.
That is because a lot of RISC-V specific experimental development was required on those dependent projects at the time.
Those local forks no longer exist hence the script that you're using will not work.

I would modify that outdated script to remove this `git checkout` so that it uses the latest toolchain sources:

* https://github.com/kevinjohnmartin/nooman-openpiton/blob/a7cd55cd7db115b5932a11b03fd5261d7366dca8/piton/design/chip/tile/ariane/ci/build-riscv-gcc.sh#L23

closed 2024-02-04 by TommyMurphyTM1234