riscv-collab/riscv-gnu-toolchain #1420: no clang executables generated after configuration with --enable-llvm

wanghuibin0, 2024-02-03:
I have followed the instruction in the "LLVM/Clang" section in README.
```
./configure --prefix=$RISCV --enable-llvm --enable-linux --with-arch=rv64gc --with-abi=lp64d
make -j$(nproc) all build-sim SIM=qemu
```
But when I checked the $RISCV directory after the build process is finished, there was no clang executable. 
```bash
$ ls ../install-llvm/bin/
riscv64-unknown-linux-gnu-addr2line  riscv64-unknown-linux-gnu-elfedit  riscv64-unknown-linux-gnu-nm       riscv64-unknown-linux-gnu-readelf
riscv64-unknown-linux-gnu-ar         riscv64-unknown-linux-gnu-gprof    riscv64-unknown-linux-gnu-objcopy  riscv64-unknown-linux-gnu-size
riscv64-unknown-linux-gnu-as         riscv64-unknown-linux-gnu-ld       riscv64-unknown-linux-gnu-objdump  riscv64-unknown-linux-gnu-strings
riscv64-unknown-linux-gnu-c++filt    riscv64-unknown-linux-gnu-ld.bfd   riscv64-unknown-linux-gnu-ranlib   riscv64-unknown-linux-gnu-strip
```
I am wondering which part I do is wrong.
Thanks for helping me.

TommyMurphyTM1234, 2024-02-03:
You'll probably need to attach the build log.
Ideally for a non-parallel build without any `-j$(nproc)` as parallel build logs are difficult to analyse due to the interleaving of logging from the different parallel jobs.
Looks to me like the build may not have run to completion.
Maybe it failed prematurely due to some error?

TommyMurphyTM1234, 2024-02-03:
This worked for me:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1420
cd issue-1420
./configure --prefix=`pwd`/installed-tools --enable-llvm --enable-linux --with-arch=rv64gc --with-abi=lp64d
make all build-sim SIM=qemu 2>&1 | tee build.log

...

ls installed-tools/bin/
amdgpu-arch             clang-rename      llvm-dwp       llvm-symbolizer                      riscv64-unknown-linux-gnu-gcc            riscv64-unknown-linux-gnu-lto-dump
analyze-build           clang-repl        llvm-lib       nvptx-arch                           riscv64-unknown-linux-gnu-gcc-13.2.0     riscv64-unknown-linux-gnu-nm
clang                   clang-scan-deps   llvm-ml        qemu-riscv32                         riscv64-unknown-linux-gnu-gcc-ar         riscv64-unknown-linux-gnu-objcopy
clang++                 diagtool          llvm-nm        qemu-riscv64                         riscv64-unknown-linux-gnu-gcc-nm         riscv64-unknown-linux-gnu-objdump
clang-17                git-clang-format  llvm-objcopy   riscv64-unknown-linux-gnu-addr2line  riscv64-unknown-linux-gnu-gcc-ranlib     riscv64-unknown-linux-gnu-ranlib
clang-check             hmaptool          llvm-objdump   riscv64-unknown-linux-gnu-ar         riscv64-unknown-linux-gnu-gcov           riscv64-unknown-linux-gnu-readelf
clang-cl                intercept-build   llvm-pdbutil   riscv64-unknown-linux-gnu-as         riscv64-unknown-linux-gnu-gcov-dump      riscv64-unknown-linux-gnu-run
clang-cpp               ld.lld            llvm-profdata  riscv64-unknown-linux-gnu-c++        riscv64-unknown-linux-gnu-gcov-tool      riscv64-unknown-linux-gnu-size
clang-extdef-mapping    ld64.lld          llvm-ranlib    riscv64-unknown-linux-gnu-c++filt    riscv64-unknown-linux-gnu-gdb            riscv64-unknown-linux-gnu-strings
clang-format            lld               llvm-rc        riscv64-unknown-linux-gnu-clang      riscv64-unknown-linux-gnu-gdb-add-index  riscv64-unknown-linux-gnu-strip
clang-linker-wrapper    lld-link          llvm-readobj   riscv64-unknown-linux-gnu-clang++    riscv64-unknown-linux-gnu-gfortran       scan-build
clang-offload-bundler   llvm-ar           llvm-size      riscv64-unknown-linux-gnu-cpp        riscv64-unknown-linux-gnu-gprof          scan-build-py
clang-offload-packager  llvm-cov          llvm-strings   riscv64-unknown-linux-gnu-elfedit    riscv64-unknown-linux-gnu-ld             scan-view
clang-refactor          llvm-cxxfilt      llvm-strip     riscv64-unknown-linux-gnu-g++        riscv64-unknown-linux-gnu-ld.bfd         wasm-ld

```
Zipped `build.log`: [build.log.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/14154208/build.log.zip)

TommyMurphyTM1234, 2024-02-03:
I also did a parallel build using `make -j$(nproc) all build-sim SIM=qemu` and it also worked fine and produced all expected artifacts.

wanghuibin0, 2024-02-04:
Thanks for your reply.
I built the project from scratch again. This time I did a serial make. Unfortunately it failed again. 
The log says that the LD_LIBRARY_PATH environment variable contains the current directory. 
But when I checked it, I did not find anything wrong.
```bash
$ echo $LD_LIBRARY_PATH
/home/wanghuibin/.local/lib/:/home/wanghuibin/.local/lib/:
$ pwd
/SSD/wanghb/projs/riscv-gnu-toolchain/build-llvm
```
zipped log here: [build.log.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/14155157/build.log.zip)

TommyMurphyTM1234, 2024-02-04:
> Thanks for your reply. I built the project from scratch again. This time I did a serial make. Unfortunately it failed again. The log says that the LD_LIBRARY_PATH environment variable contains the current directory.
> ...
> But when I checked it, I did not find anything wrong.
>```
> $ echo $LD_LIBRARY_PATH
> /home/wanghuibin/.local/lib/:/home/wanghuibin/.local/lib/:
> ```

The trailing `:` in your `LD_LIBRARY_PATH` is going to be converted into the current working directory.

* https://github.com/Linuxbrew/legacy-linuxbrew/issues/807#issuecomment-193656929
* https://github.com/Linuxbrew/legacy-linuxbrew/issues/807#issuecomment-307288187

My own `LD_LIBRARY_PATH` is blank/not set so maybe just do `unset LD_LIBRARY_PATH` before your toolchain build to avoid such issues?

Configuration/environment issues such as this is one reason that I generally do builds in a clean virtual machine/containerised installations rather than a regular machine used for other day to day purposes which may have clutter that impacts the build.

wanghuibin0, 2024-02-04:
OK, I'll try it in a fresh envrironment. Thank you very much~

TommyMurphyTM1234, 2024-02-04:
> OK, I'll try it in a fresh envrironment. Thank you very much~

Maybe just try this first?
```
unset LD_LIBRARY_PATH
make distclean
rm -rf $RISCV
./configure --prefix=$RISCV --enable-llvm --enable-linux --with-arch=rv64gc --with-abi=lp64d
make -j$(nproc) all build-sim SIM=qemu
```

closed 2024-02-04 by TommyMurphyTM1234

wanghuibin0, 2024-02-04:
> > OK, I'll try it in a fresh envrironment. Thank you very much~
> 
> Maybe just try this first?
> 
> ```
> unset LD_LIBRARY_PATH
> make distclean
> rm -rf $RISCV
> ./configure --prefix=$RISCV --enable-llvm --enable-linux --with-arch=rv64gc --with-abi=lp64d
> make -j$(nproc) all build-sim SIM=qemu
> ```

It works. Thank you for helping me.