riscv-collab/riscv-gnu-toolchain #1422: cc1: error: no include path in which to search for stdc-predef.h

hassan-xy, 2024-02-06:
Hi guys,
I wanted to load the "riscv-gnu-toolchain" in my desktop and was facing an error with the compilation. 
I have executed the below commands.
"./configure --with-multilib-generator="rv32i-ilp32-rv32im-c""
sudo make linux

the error I get "cc1: error: no include path in which to search for stdc-predef.h"
[image]
[image]

Do let me know what changes I need to make to resolve this issue.

Thanks

TommyMurphyTM1234, 2024-02-06:
Does it make any difference if you do a clean build?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1411

```
make distclean
make linux
```
Note that --with-multilib-generator` only works with the Newlib/bare-metal toolchain and not the Linux toolchain:

* https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#build-with-customized-multi-lib-configure

TommyMurphyTM1234, 2024-02-06:
> Just to make it clear, I am running ubuntu 20.04, and needed riscv-none-embed-gcc compiler for one my cross compilations. 

It's a bit confusing that you were previously doing `make linux` so - this builds the RISC-V Linux toolchain and not the bare-metal/Newlib toolchain. The former is for compiling user mode programs to run on a RISC-V system that runs Linux. The latter is for targeting bare-metal/lightweight RTOS systems.

> I have riscv64 toolchain with me right now.

I don't know what this means. What toolchain are you referring to here?

> I need riscv-none-embed-gcc in my system as the final output.
> I still dont see riscv-none compilers.

You're not going to see any `riscv-none-embed` prefixed tools from the `riscv-gnu-toolchain` build.
The bare-metal/Newlib toolchain built by the following:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --enable-multilib
make
```
will build the bare-metal/Newlib toolchain that has all tools prefixed by the string `riscv64-unknown-elf` - e.g. `riscv64-unknown-elf-gcc`. If, for some reason, you want/need the prefix `riscv-none-embed` then you may have to use some `configure` option or parameterise/modify the generated `Makefile`.

Alternatively you could the xPack Project's Docker container approach to building the toolchain as it uses the `riscv-none-embed` prefix by default:

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/#riscv-none-elf-gcc

Or maybe you could just use that project's binary builds to avoid building the toolchain yourself (lots of people seem to think that they need to do this when there are perfectly good pre-built toolchains available already).

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/#riscv-none-elf-gcc

> I dont think the build was a success.

I can barely read the screenshot and when I click on it for a bigger view it comes up as "private" (`This private-user-images.githubusercontent.com page can’t be found`) but it seems to me that the build did complete and you should find the `riscv64-unknown-elf` prefixed tools in the `bin` directory of your target install directory (`/opt/riscv/bin`) by default.

To summarise:

1. If you are still having problems building the toolchain then you need to provide more details such as the build log (e.g. `make 2>&1 | tee build.log`). If you do this then please compress and attach the log rather than copying and pasting it. Also - it's better to use logs rather than screenshots as the latter are not easy to read for everybody and do not facilitate easy analysis. In fact some of your screeshots are not appearing in "zoomed" mode for me because they are coming up as "private",
2. If you really need your toolchain to be `riscv-none-embed` rather than `riscv64-unknown-elf` prefixed then you need to figure out how to do that via `configure` (I can't see anything obvious) or by parameterising/editing the generated `Makefile` or use the xPack Project approach or pre-built binaries.
3. I can't see that you have identified any issue with the `riscv-gnu-toolchain` here and the issue seems to have deviated away from the error message originally reported.

TommyMurphyTM1234, 2024-02-06:
> This is what I have done till now.
> 
> 1. ./configure --prefix=/opt/riscv
> 2. ./configure --enable-multilib
> 3. sudo make linux
>    ![image](https://private-user-images.githubusercontent.com/125990448/302570727-5570659b-8801-4fe9-baf2-2a5d2c45a56d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcyMTQ2OTAsIm5iZiI6MTcwNzIxNDM5MCwicGF0aCI6Ii8xMjU5OTA0NDgvMzAyNTcwNzI3LTU1NzA2NTliLTg4MDEtNGZlOS1iYWYyLTJhNWQyYzQ1YTU2ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwNlQxMDEzMTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04MGE5ZWQwM2QyYzZiOWFkZjkxNWYyMzg5OWE1MTgwMmE5YzUzNjZiNmVkZThjNDhmMzc1NmExNDVlZjVmZDJkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.8nW3iSK7P9aRU9sRk0XSg0o3orSSmfOI4UzxPFiQbsI)
> 
> I still dont see riscv-none compilers. I dont think the build was a success. Please guide me.
> 
> To prevent any misunderstandings, I need **riscv-none-embed-gcc** in my system as the final output.
> 
> Thanks

For some reason I CAN see the screenshot zoomed now - weird.
Anyway - your toolchain build did complete but you're still building the Linux rather than the bare-metal/Newlib toolchain.
You need to to `make` not `make linux` for the latter.
This will give you the `riscv64-unknown-elf` prefixed bare-metal/Newlib toolchain.
As I explained above if you want a different prefix such as `riscv-none-embed` then you'll need to look at the suggestions that I posted above.

FWIW I tried this in order to override the default toolchain prefix but obviously it failed.
```
NEWLIB_TUPLE=riscv-none-embed make 2>&1 | tee build.log
mkdir -p /home/user/Downloads/mailing-list/installed-tools/.test || \
        (echo "Sorry, you don't have permission to write to" \
         "'/home/user/Downloads/mailing-list/installed-tools'. Please make sure that the location is " \
         "writable or use --prefix to specify another destination.'" \
         && exit 1)
rm -r /home/user/Downloads/mailing-list/installed-tools/.test
mkdir -p stamps/ && touch stamps/check-write-permission
rm -rf stamps/build-binutils-newlib build-binutils-newlib
mkdir build-binutils-newlib
cd build-binutils-newlib && CC_FOR_TARGET=riscv-none-embed-gcc /home/user/Downloads/mailing-list/binutils/configure \
        --target=riscv-none-embed \
         \
        --prefix=/home/user/Downloads/mailing-list/installed-tools \
         \
        --disable-werror \
        --with-expat=yes  \
        --disable-gdb \
        --disable-sim \
        --disable-libdecnumber \
        --disable-readline \
        --with-isa-spec=20191213
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... Invalid configuration 'riscv-none-embed': OS 'embed' not recognized
configure: error: /bin/bash /home/user/Downloads/mailing-list/binutils/config.sub riscv-none-embed failed
make: *** [Makefile:561: stamps/build-binutils-newlib] Error 1
```
I can't remember offhand how xPack does it but all of the scripts etc. are available for you to find out if necessary.

TommyMurphyTM1234, 2024-02-06:
Small correction - I incorrectly stated that the xPack RISC-V toolchain uses the `riscv-none-embed` prefix.
It does not - it uses `riscv-none-elf`:

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/#riscv-none-elf-gcc

So if you really want/need the `riscv-none-embed` prefix and cannot do with the `riscv-gnu-toolchain`'s `riscv64-unknown-elf` or xPack's `riscv-none-elf` prefixes for some reason then you will need to figure out how to use a non-default toolchain prefix string.

hassan-xy, 2024-02-06:
Ok I get your point. But I only see riscv64 based compilers and node riscv-none-elf. 
[image]

TommyMurphyTM1234, 2024-02-06:
> Ok I get your point. But I only see riscv64 based compilers and node riscv-none-elf. 

I have explained this already!
A build of the `riscv-gnu-toolchain` bare-metal/Newlib toolchain results in a toolchain that prefixes the tools with `riscv64-unknown-elf` (or `riscv32-unknown-elf` if `configure --with-arch=rv32...` is used). 
If you need a different prefix such as `riscv-none-embed` (why?) then you need to make changes to the build process.
Details of which are outside the scope of this repo.

Since you are not having any problems building the toolchains after all - your original error message seems to have disappeared without explanation - I am closing this issue as it does not identify any problem with the `riscv-gnu-toolchain` repo.

closed 2024-02-06 by TommyMurphyTM1234