riscv-collab/riscv-gnu-toolchain #1425: What is the difference between riscv64-unknown-elf-gcc and riscv-none-elf-gcc?

Juan-Gg, 2024-02-16:
As the title says. I haven't been able to find information about this, where could I find such information?
Thank you.

TommyMurphyTM1234, 2024-02-16:
What is the `riscv-none-elf-gcc` that you're referring to?
The xPack GNU RISC-V Embedded GCC perhaps?

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/

If so then they are both versions of the RISC-V Newlib/bare-metal toolchain but with some configuration/build differences (and, obviously, a different toolchain name prefix).

TommyMurphyTM1234, 2024-02-16:
Discussion of general issues such as this is better suited to other forums such as the mailing lists rather than being logged as an Issue here which is primarily intended for problems or bugs with the `riscv-gnu-toolchain` repo/tools (excepting those that should be logged against the upstream [submodule projects](https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.gitmodules) where ongoing development takes place).

closed 2024-02-16 by TommyMurphyTM1234

Juan-Gg, 2024-02-16:
> What is the riscv-none-elf-gcc that you're referring to?

In my case, it was the toolchain built by the scripts provided in a [RISC-V core implementation](https://github.com/openhwgroup/cva6/tree/master/util/gcc-toolchain-builder). Your links are helpful. I will ask elsewhere next time.
Thank you.

TommyMurphyTM1234, 2024-02-16:
> > What is the riscv-none-elf-gcc that you're referring to?
> 
> In my case, it was the toolchain built by the scripts provided in a [RISC-V core implementation](https://github.com/openhwgroup/cva6/tree/master/util/gcc-toolchain-builder). 

In that case you might want to ask on that repo how their toolchain compares/contrasts with the one built from the `riscv-gnu-toolchain` repo. Or maybe just study their build scripts.

As all of this illustrates, there are different approachs to building a RISC-V toolchain - e.g. the current `riscv-gnu-toolchain` repo's approach, the xPack approach, the OpenHW Group approach etc. And there are many others these days. And also LLVM/Clang based toolchains too.

ilg-ul, 2024-02-16:
> there are different approaches to building a RISC-V toolchain

I don't know about the other approaches, but the xPack RISC-V GCC approach is to use the original upstream GCC sources, without changes (except unusual cases when bugs are identified and fixed, before the fixes are upstreamed). 

The selection of multi-libs is a difficult choice for a general purpose toolchain, and currently a combinatorial approach is used, but this results in a huge archive, and this decision might be reconsidered in the future. 

As for the name, it does not include the word size (like `riscv32-` or `riscv64-`) because this is redundant and confusing, the same compiler is able to generate both 32/64-bit binaries.

TommyMurphyTM1234, 2024-02-16:
> > there are different approaches to building a RISC-V toolchain
> 
> I don't know about the other approaches, but the xPack RISC-V GCC approach is to use the original upstream GCC sources, without changes (except unusual cases when bugs are identified and fixed, before the fixes are upstreamed).

Thanks @ilg-ul - I was vaguely referring to things like what sources (and specific repos/branches/commits etc.) are used, what configuration and build options are used, what multilibs are built etc.

> The selection of multi-libs is a difficult choice for a general purpose toolchain, and currently a combinatorial approach is used, but this results in a huge archive, and this decision might be reconsidered in the future.

Indeed - I recall the long discussion on this topic:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1315

> As for the name, it does not include the word size (like `riscv32-` or `riscv64-`) because this is redundant and confusing, the same compiler is able to generate both 32/64-bit binaries.

I agree that this makes more sense than having `riscv64-...` or `riscv32-...`.

TommyMurphyTM1234, 2024-02-17:
I just noticed another difference between the OpenHW CVA6 toolchain and the `riscv-gnu-toolchain` and xPack toolchains - it only builds C support and not also C++ support as the latter two do.

* https://github.com/openhwgroup/cva6/blob/b4c287a18eafcc6c25f456efb42b956c45036af2/util/gcc-toolchain-builder/config/gcc-13.1.0-baremetal.sh#L48

Juan-Gg, 2024-02-18:
Thank you for providing those details.