riscv-collab/riscv-gnu-toolchain #1426: glibc repository does not exist

siserte, 2024-02-18:
I am missing something since I cannot clone the glibc repository in the toolchain.
The link provided for glibc is broken: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/glibc

How can I get glibc?

TommyMurphyTM1234, 2024-02-18:
You must be cloning an old commit or something.
For a long time now this repo uses submodules to refer up to the master projects including Glibc:

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/59ab58e8a4aed4ed8f711ebab307757a5ebaa1f5/.gitmodules#L9-L11

What `git clone` command line are you using?

siserte, 2024-02-18:
I am using 2021.03.17 since I need to install GCC10.
Thanks, I'll update .gitmodules as you suggest taking into account the tag peculiarities.

TommyMurphyTM1234, 2024-02-18:
> I am using 2021.03.17 since I need to install GCC10. Thanks, I'll update .gitmodules as you suggest taking into account the tag peculiarities.

The `.gitmodules` file is a bit confusing and changing the details there may not work (or suffice?).
See here for example:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1286

You may need to change the actual Git submodule details.

Or maybe just pull the relevant upstream sources and use out of tree sources - e.g.:

* https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#use-source-tree-other-than-riscv-gnu-toolchain

```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://sourceware.org/git/glibc.git glibc-upstream
cd glibc-upstream
git checkout <relevant-commit>
cd ..
./configure ... --with-glibc-src=`pwd`/glibc-upstream
...
```

TommyMurphyTM1234, 2024-02-18:
> I am using 2021.03.17 since I need to install GCC10. 

Could you use the pre-built CI binary release?

* https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2021.03.17

siserte, 2024-02-18:
That would save me a lot of time but I need it for Fedora, and I only see Ubuntu packages :(

Thanks!

TommyMurphyTM1234, 2024-02-18:
> That would save me a lot of time but I need it for Fedora, and I only see Ubuntu packages :(
> 
> Thanks!

Did you try running the CI build on Fedora in case it just works?

siserte, 2024-02-18:
It was worth it to try, but no luck :(

`-bash: ./riscv64-unknown-linux-gnu-gcc-10.2.0: cannot execute binary file: Exec format error`

TommyMurphyTM1234, 2024-02-18:
> It was worth it to try, but no luck :(
> 
> `-bash: ./riscv64-unknown-linux-gnu-gcc-10.2.0: cannot execute binary file: Exec format error`

Strange - I did this and it worked:
```
# On CentOS 7.9.2009...
cat /etc/centos-release
CentOS Linux release 7.9.2009 (Core)

cd ~/Downloads
wget riscv64-glibc-ubuntu-20.04-nightly-2021.03.17-nightly.tar.gz
tar xzvf riscv64-glibc-ubuntu-20.04-nightly-2021.03.17-nightly.tar.gz
./riscv/bin/riscv64-unknown-linux-gnu-gcc -v
Using built-in specs.
COLLECT_GCC=./riscv/bin/riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/home/user/Downloads/riscv/bin/../libexec/gcc/riscv64-unknown-linux-gnu/10.2.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: /home/runner/work/riscv-gnu-toolchain/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/opt/riscv --with-sysroot=/opt/riscv/sysroot --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=.././riscv-gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gc --with-tune=rocket 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (GCC) 
```
Are you sure that you're running an x86_64 version of Fedora?

siserte, 2024-02-18:
No! I am running the riscv64 GNU/Linux Fedora 38 (Thirty Eight).

TommyMurphyTM1234, 2024-02-18:
> No! I am running the riscv64 GNU/Linux Fedora 38 (Thirty Eight).

OK - in that case an x86_64 build of the toolchain was obviously never going to work.

I presume that you'll need to do `make linux-native` when compiling the toolchain on a RISC-V Linux host to run on that host?
As opposed to doing `make linux` for a cross toolchain build (e.g. build on x86_64 to run on x86_64 to compile for RISC-V.

Unfortunately I could never find much in the way of documentation of the `make linux-native` option other than this:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues?q=is%3Aissue+%22make+linux-native%22+
* https://github.com/riscv-collab/riscv-gnu-toolchain/pulls?q=is%3Apr+%22make+linux-native%22

TommyMurphyTM1234, 2024-02-18:
Closing the issue as the original problem has been explained.

closed 2024-02-18 by TommyMurphyTM1234