riscv-collab/riscv-gnu-toolchain #1428: Is that normal or error that GNU toolchain is generating 2-byte and some  4-byte Instructions ?

zakirhussaingit, 2024-02-26:
After compiling the c code the object code I am getting has got some  2-byte and some 4-byte Instructions. The following are my observations on the latest UBUNTU[Ubuntu 22.04.4 LTS]:

For the installation of the RISC V GNU toolchain, I have followed the following steps.

step1: git clone https://github.com/riscv/riscv-gnu-toolchain

step2: Follow what is mentioned in the above-mentioned git repository [which I have put below for reference]. To build the Newlib cross-compiler, pick an install path (that is writeable). If you choose, say, /opt/riscv, then add /opt/riscv/bin to your PATH. Then, simply run the following command:

./configure --prefix=/opt/riscv --enable-multilib
make

--enable-multilib this one is required to avoid errors related to multilib

TommyMurphyTM1234, 2024-02-26:
What is your toolchain default `arch/abi`?
You should be able to get this from `riscv64-unknown-elf-gcc -v` via the `--with-arch=... --with-abi=...` settings.
I presume that it's `rv64gc/lp64d` which means that it will, by default, generate compressed instructions and link with `rv64gc/lp64d` libraries/startup code which will contain same.
If you don't want compressed instructions then you need to specify an architecture without the `C` extension and ensure that the toolchain has multilibs for that architecture.
E.g. `riscv64-unknown-elf-gcc -march=rv64g -mabi= lp64d ...`.

zakirhussaingit, 2024-02-26:
Upon typing riscv64-unknown-elf-gcc -v in the terminal, I got this as in the screenshot.

[image]
Where I have 
--with-abi=lp64d --with-arch=rv64imafdc

zakirhussaingit, 2024-02-26:
But, to do what you have suggested, I must have them right! When, I type in the terminal riscv64-unknown-elf-gcc -print-multi-lib
I do not see these -march=rv64g -mabi=lp64d . What should I do? 
[image]

zakirhussaingit, 2024-02-26:
I will try, thank you for all of your answers.