riscv-collab/riscv-gnu-toolchain #1433: riscv64-unknown-elf-gcc or riscv64-unknown-elf-g++ not detecting libraries

DanielLopezUCM, 2024-03-11:
When trying to compile using this compiller the libraries are not able to include. This is the output.
The gcc variant:
```
riscv64-unknown-elf-gcc -std=c++14 hola.cpp -o hola -lstdc++ -lm
hola.cpp:1:10: fatal error: vector: No such file or directory
    1 | #include <vector>
      |          ^~~~~~~~
compilation terminated.
```
The g++ variant:
```
riscv64-unknown-elf-g++ -std=c++14 hola.cpp -o hola
hola.cpp:1:10: fatal error: vector: No such file or directory
    1 | #include <vector>
      |          ^~~~~~~~
compilation terminated.
```
The same error on both. If someone could help I would really apreciate it.
Thanks

TommyMurphyTM1234, 2024-03-11:
I did this without any problems:

Download and extract the latest CI build of the toolchain from here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/releases
* https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2024.03.01

Specifically:

* https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.03.01/riscv64-elf-ubuntu-22.04-gcc-nightly-2024.03.01-nightly.tar.gz

Create a simple `test.cpp` program:
```
// https://www.programiz.com/cpp-programming/vectors

#include <iostream>
#include <vector>
using namespace std;

int main() {

  // initializer list
  vector<int> vector1 = {1, 2, 3, 4, 5};

  // uniform initialization
  vector<int> vector2{6, 7, 8, 9, 10};

  // method 3
  vector<int> vector3(5, 12);

  cout << "vector1 = ";

  // ranged loop
  for (const int& i : vector1) {
    cout << i << "  ";
  }

  cout << "\nvector2 = ";

  // ranged loop
  for (const int& i : vector2) {
    cout << i << "  ";
  }

  cout << "\nvector3 = ";

  // ranged loop
  for (int i : vector3) {
    cout << i << "  ";
  }

  return 0;
}
```
Compile it:
```
riscv64-unknown-elf-g++ -std=c++14 test.cpp -o test`
file test
test: ELF 64-bit LSB executable, UCB RISC-V, RVC, double-float ABI, version 1 (SYSV), statically linked, with debug_info, not stripped
```

So you'll probably need to explain where you got or exactly how you built your toolchain.

And maybe provide verbose log output from the attempted compilation - e.g.
```
./riscv64-unknown-elf-g++ -std=c++14 test.cpp -o test -v
Using built-in specs.
COLLECT_GCC=./riscv64-unknown-elf-g++
COLLECT_LTO_WRAPPER=/home/user/Downloads/riscv/bin/../libexec/gcc/riscv64-unknown-elf/13.2.0/lto-wrapper
Target: riscv64-unknown-elf
[... 11 lines cut ...]
ignoring duplicate directory "/home/user/Downloads/riscv/bin/../lib/gcc/../../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/include/c++/13.2.0/backward"
ignoring duplicate directory "/home/user/Downloads/riscv/bin/../lib/gcc/../../lib/gcc/riscv64-unknown-elf/13.2.0/include"
ignoring nonexistent directory "/home/user/Downloads/riscv/bin/../riscv64-unknown-elf/usr/local/include"
ignoring duplicate directory "/home/user/Downloads/riscv/bin/../lib/gcc/../../lib/gcc/riscv64-unknown-elf/13.2.0/include-fixed"
ignoring duplicate directory "/home/user/Downloads/riscv/bin/../lib/gcc/../../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/include"
ignoring duplicate directory "/home/user/Downloads/riscv/bin/../riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 /home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/include/c++/13.2.0
 /home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/include/c++/13.2.0/riscv64-unknown-elf/.
 /home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/include/c++/13.2.0/backward
 /home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/include
 /home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/include-fixed
 /home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/include
End of search list.
Compiler executable checksum: 6fc17400916a1acf0b3f3ecc0711725d
COLLECT_GCC_OPTIONS='-std=c++14' '-o' 'test' '-v' '-mtune=rocket' '-march=rv64imafdc_zicsr_zifencei' '-mabi=lp64d' '-misa-spec=20191213' '-march=rv64imafdc_zicsr_zifencei'
 /home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/bin/as -v --traditional-format -march=rv64imafdc_zicsr_zifencei -march=rv64imafdc_zicsr_zifencei -mabi=lp64d -misa-spec=20191213 -o /tmp/ccfP14nu.o /tmp/ccJsHbyB.s
GNU assembler version 2.42 (riscv64-unknown-elf) using BFD version (GNU Binutils) 2.42
COMPILER_PATH=/home/user/Downloads/riscv/bin/../libexec/gcc/riscv64-unknown-elf/13.2.0/:/home/user/Downloads/riscv/bin/../libexec/gcc/:/home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/bin/
LIBRARY_PATH=/home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/:/home/user/Downloads/riscv/bin/../lib/gcc/:/home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/lib/:/home/user/Downloads/riscv/bin/../riscv64-unknown-elf/lib/
COLLECT_GCC_OPTIONS='-std=c++14' '-o' 'test' '-v' '-mtune=rocket' '-march=rv64imafdc_zicsr_zifencei' '-mabi=lp64d' '-misa-spec=20191213' '-march=rv64imafdc_zicsr_zifencei' '-dumpdir' 'test.'
 /home/user/Downloads/riscv/bin/../libexec/gcc/riscv64-unknown-elf/13.2.0/collect2 -plugin /home/user/Downloads/riscv/bin/../libexec/gcc/riscv64-unknown-elf/13.2.0/liblto_plugin.so -plugin-opt=/home/user/Downloads/riscv/bin/../libexec/gcc/riscv64-unknown-elf/13.2.0/lto-wrapper -plugin-opt=-fresolution=/tmp/ccllDVrf.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgloss -plugin-opt=-pass-through=-lgcc --sysroot=/home/user/Downloads/riscv/bin/../riscv64-unknown-elf -melf64lriscv -o test /home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/lib/crt0.o /home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/crtbegin.o -L/home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0 -L/home/user/Downloads/riscv/bin/../lib/gcc -L/home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/../../../../riscv64-unknown-elf/lib -L/home/user/Downloads/riscv/bin/../riscv64-unknown-elf/lib /tmp/ccfP14nu.o -lstdc++ -lm -lgcc --start-group -lc -lgloss --end-group -lgcc /home/user/Downloads/riscv/bin/../lib/gcc/riscv64-unknown-elf/13.2.0/crtend.o
COLLECT_GCC_OPTIONS='-std=c++14' '-o' 'test' '-v' '-mtune=rocket' '-march=rv64imafdc_zicsr_zifencei' '-mabi=lp64d' '-misa-spec=20191213' '-march=rv64imafdc_zicsr_zifencei' '-dumpdir' 'test.'
```

DanielLopezUCM, 2024-03-11:
I'll try downloading the lastest version. I did obtain the toolchain from this github and built it. As far as my testing I've seen that if I include the directory `/usr/include/c++/11` it detects vector, but not the `bit` content as here shows.

```
riscv64-unknown-elf-g++ hola.cpp -o hola -Im -I/usr/include/c++/11 -I/usr/include -I/usr/include/c++/11/bits
In file included from /usr/include/c++/11/vector:60,
                 from hola.cpp:1:
/usr/include/c++/11/bits/stl_algobase.h:59:10: fatal error: bits/c++config.h: No such file or directory
   59 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

This is the vervose output.

```
riscv64-unknown-elf-g++ hola.cpp -o hola -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/riscv64-unknown-elf/10.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/include' --mandir='/usr/share/man' --infodir='/usr/share/info' --sysconfdir=/etc --localstatedir=/var --disable-option-checking --disable-silent-rules --libdir='/usr/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --target=riscv64-unknown-elf --prefix=/usr --infodir=/usr/share/doc/gcc-riscv64-unknown-elf/info --mandir=/usr/share/man --htmldir=/usr/share/doc/gcc-riscv64-unknown-elf/html --pdfdir=/usr/share/doc/gcc-riscv64-unknown-elf/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --with-pkgversion= --disable-shared --disable-threads --enable-languages=c,c++ --enable-tls --with-newlib --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --with-system-zlib --enable-checking=yes --enable-multilib --with-abi=lp64d --disable-libstdcxx-pch --disable-libstdcxx --disable-fixinc --with-arch=rv64imafdc --with-gnu-as --with-gnu-ld --with-as=/usr/lib/riscv64-unknown-elf/bin/as --with-ld=/usr/lib/riscv64-unknown-elf/bin/ld AR_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ar AS_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/as NM_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/nm LD_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ld OBJDUMP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/objdump RANLIB_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/ranlib READELF_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/readelf STRIP_FOR_TARGET=/usr/lib/riscv64-unknown-elf/bin/strip CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-LkcieF/gcc-riscv64-unknown-elf-10.2.0=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-LkcieF/gcc-riscv64-unknown-elf-10.2.0=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-LkcieF/gcc-riscv64-unknown-elf-10.2.0=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-LkcieF/gcc-riscv64-unknown-elf-10.2.0=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-LkcieF/gcc-riscv64-unknown-elf-10.2.0=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,now' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-LkcieF/gcc-riscv64-unknown-elf-10.2.0=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-riscv64-unknown-elf-LkcieF/gcc-riscv64-unknown-elf-10.2.0=. -fstack-protector-strong' 'CFLAGS_FOR_TARGET=-Os -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medany'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.0 () 
COLLECT_GCC_OPTIONS='-o' 'hola' '-v' '-march=rv64imafdc' '-mabi=lp64d'
 /usr/lib/gcc/riscv64-unknown-elf/10.2.0/cc1plus -quiet -v -imultilib rv64imafdc/lp64d hola.cpp -quiet -dumpbase hola.cpp -march=rv64imafdc -mabi=lp64d -auxbase hola -version -o /tmp/cctX2C3K.s
GNU C++14 () version 10.2.0 (riscv64-unknown-elf)
        compiled by GNU C version 10.2.0, GMP version 6.2.0, MPFR version 4.1.0, MPC version 1.2.0-rc1, isl version none
warning: GMP header version 6.2.0 differs from library version 6.2.1.
warning: MPC header version 1.2.0-rc1 differs from library version 1.2.1.
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
ignoring nonexistent directory "/usr/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/include/c++/10.2.0"
ignoring nonexistent directory "/usr/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/include/c++/10.2.0/riscv64-unknown-elf/rv64imafdc/lp64d"
ignoring nonexistent directory "/usr/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/include/c++/10.2.0/backward"
ignoring nonexistent directory "/usr/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/sys-include"
ignoring nonexistent directory "/usr/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/gcc/riscv64-unknown-elf/10.2.0/include
 /usr/lib/gcc/riscv64-unknown-elf/10.2.0/include-fixed
End of search list.
GNU C++14 () version 10.2.0 (riscv64-unknown-elf)
        compiled by GNU C version 10.2.0, GMP version 6.2.0, MPFR version 4.1.0, MPC version 1.2.0-rc1, isl version none
warning: GMP header version 6.2.0 differs from library version 6.2.1.
warning: MPC header version 1.2.0-rc1 differs from library version 1.2.1.
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
Compiler executable checksum: 66d90f180db8d7de16552e86d6a35f1b
hola.cpp:1:10: fatal error: vector: No such file or directory
    1 | #include <vector>
      |          ^~~~~~~~
compilation terminated.
```

DanielLopezUCM, 2024-03-11:
> That's very old!
That was the problem, solved with the new release, thank you very much. :)
I'm closing the issue, thanks for the help.

closed 2024-03-11 by DanielLopezUCM