riscv-collab/riscv-gnu-toolchain #1436: gdb set breakpoint auto-hw on failed

IZBH, 2024-03-15:
Hello everyone!hope you're all doing well.

I'm currently working on my own chip. To verify the functionality, I run GDB, openOCD on PC(windwos) to communicate with FPGA.  Everything works well, except when I try to `set breakpoint auto-hw on`.

"The GDB version I'm using is 13.2.0, from https://www.embecosm.com/resources/tool-chain-downloads/#riscv-stable.
The command I use is `riscv32-unknown-elf-gdb.exe -command=.gdbinit -x flash.gdb`.

Here is the content of my .gdbinit file:

```
set breakpoint auto-hw on
mem 0x28000000 0x280FFFFF ro
```

As you can see, GDB correctly obtained the memory attributes and It provided a Note(automatically using hardware breakpoints for read-only addresses).But when I use breakpoints, gdb does not automatically convert them to hardware breakpoints.

[image]

Is my configuration correct, or does riscv-gdb not support auto-hw?

I'm looking forward to your feedback, thanks a lot!

TommyMurphyTM1234, 2024-03-15:
> or does riscv-gdb not support auto-hw?

I'm pretty sure that it does.

You should probably share (capture and attach) your GDB remote trace log and OpenOCD verbose (`openocd -d3`) log to help with analysis.

IZBH, 2024-03-15:
This is the log generated after I re-executed the above operations.

[gdb_log.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/14614658/gdb_log.txt)
[openocd_log.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/14614660/openocd_log.txt)

> > or does riscv-gdb not support auto-hw?
> 
> I'm pretty sure that it does.
> 
> You should probably share (capture and attach) your GDB remote trace log and OpenOCD verbose () log to help with analysis.`openocd -d3`

TommyMurphyTM1234, 2024-03-15:
> This is the log generated after I re-executed the above operations.
> 
> [gdb_log.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/14614658/gdb_log.txt) [openocd_log.txt]

I meant a log of the GDB Remote Serial protocol interactions with OpenOCD - e.g. using `set debug remote on`:

* https://sourceware.org/gdb/current/onlinedocs/gdb.html/Debugging-Output.html#:~:text=set%20debug%20remote,stream.%20The%20default%20is%20off.

> (https://github.com/riscv-collab/riscv-gnu-toolchain/files/14614660/openocd_log.txt)

I can't see any breakpoints being set there but this may not happen until the target is resumed.

TommyMurphyTM1234, 2024-03-15:
> > or does riscv-gdb not support auto-hw?
> 
> I'm pretty sure that it does.

Ok - I may be wrong.
A quick test at my end suggests that it does not!
I'll need to check in more detail and maybe with an Arm target to see if that behaves any differently.

TommyMurphyTM1234, 2024-03-15:
> ![image-20240315152256617](https://private-user-images.githubusercontent.com/61223505/313109984-3e865691-5c4f-49b3-aa2f-a19bc5613515.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA1MDY4MzUsIm5iZiI6MTcxMDUwNjUzNSwicGF0aCI6Ii82MTIyMzUwNS8zMTMxMDk5ODQtM2U4NjU2OTEtNWM0Zi00OWIzLWFhMmYtYTE5YmM1NjEzNTE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE1VDEyNDIxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUwZTRmOGJkOGQ5OGFmOGZhZDAxMzliNjE2ZThkNDc4ZTI0Y2QyYTJhNGZiMzBhZGI3NjhmZWI0MzE0ZGQ4MjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Ah2dLlXmBAuMdSLsIzRMMyqmv9UsTqoR5M8Nouyyq_8)

Just thinking - is it possible that GDB's local cached list of breakpoints always reflects the nature of the requested breakpoint (e.g. `break ...` being a software breakpoint by default) and any translation of this into a hardware breakpoint (via `set breakpoint auto-hw on` and definition of read-only memory regions) only happens when the breakpoint is "flushed" to the remote target? 

What happens in your case when you resume the target so that this happens? Does the ostensibly soft breakpoint in a read-only memory region get translated into a hardware breakpoint via the GDB Remote Serial Protocol and what OpenOCD does?

A quick text against the SiFive HiFive1 Rev A01 board suggests that this is the case but I'm doing it in a bit of a rush so can't be 100% sure.

IZBH, 2024-03-18:
Yes, you are right.I'm sorry for only replying now.When I set the attribute of the memory region to read-only, the behaviors of b and hb are same.
It looks like GDB's local cached list of breakpoints and any translation of them into a hardware breakpoint  only happens when the breakpoint is "flushed" to the remote target.

IZBH, 2024-03-18:
> > ![image-20240315152256617](https://private-user-images.githubusercontent.com/61223505/313109984-3e865691-5c4f-49b3-aa2f-a19bc5613515.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA1MDY4MzUsIm5iZiI6MTcxMDUwNjUzNSwicGF0aCI6Ii82MTIyMzUwNS8zMTMxMDk5ODQtM2U4NjU2OTEtNWM0Zi00OWIzLWFhMmYtYTE5YmM1NjEzNTE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE1VDEyNDIxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUwZTRmOGJkOGQ5OGFmOGZhZDAxMzliNjE2ZThkNDc4ZTI0Y2QyYTJhNGZiMzBhZGI3NjhmZWI0MzE0ZGQ4MjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Ah2dLlXmBAuMdSLsIzRMMyqmv9UsTqoR5M8Nouyyq_8)
> 
> Just thinking - is it possible that GDB's local cached list of breakpoints always reflects the nature of the requested breakpoint (e.g. `break ...` being a software breakpoint by default) and any translation of this into a hardware breakpoint (via `set breakpoint auto-hw on` and definition of read-only memory regions) only happens when the breakpoint is "flushed" to the remote target?
> 
> What happens in your case when you resume the target so that this happens? Does the ostensibly soft breakpoint in a read-only memory region get translated into a hardware breakpoint via the GDB Remote Serial Protocol and what OpenOCD does?
> 
> A quick text against the SiFive HiFive1 Rev A01 board suggests that this is the case but I'm doing it in a bit of a rush so can't be 100% sure. When I do this in GDB:
> 
> ```
> (gdb) mem 0x80000000 0x8001ffff ro
> (gdb) i mem
> Using user-defined memory regions.
> Num Enb Low Addr   High Addr  Attrs 
> 1   y  	0x80000000 0x8001ffff ro nocache 
> (gdb) b main
> Breakpoint 1 at 0x8000032c: file programs/semihosting.c, line 66.
> Note: automatically using hardware breakpoints for read-only addresses.
> (gdb) si
> halted at 0x404 due to step
> halted at 0x404 due to step
> 0x00000404 in ?? ()
> (gdb) 
> ```
> 
> I see in the OpenOCD log that it uses a trigger to presumably set up a hardware breakpoint rather than using a software breakpoint by writing a `BKPT` instruction into code memory having saved the original instruction.
> 
> ```
> Debug: 4673 83682 riscv.c:1105 maybe_add_trigger_t2_t6_for_bp(): [riscv.cpu] trying to setup equality match trigger
> Debug: 4674 83682 riscv.c:847 try_setup_single_match_trigger(): [riscv.cpu] trying to set up a match trigger
> Debug: 4675 83682 riscv.c:747 log_trigger_request_info(): tdata1=28001044, tdata2=8000032c, tdata1_ignore_mask=7e00000
> Debug: 4676 83682 riscv.c:615 find_next_free_trigger(): [riscv.cpu] 1 trigger(s) of type 2 found on index 0, chained == false
> Debug: 4677 83682 riscv.c:5186 riscv_set_or_write_register(): [riscv.cpu] Writing to target: tselect <- 0x0 (cacheable=false, valid=false, dirty=false)
> ```
Thank you very much for your help

closed 2024-03-18 by IZBH