riscv-collab/riscv-gnu-toolchain #1437: How to compile the newlib toolchain with PIE

DynamicLoader, 2024-03-15:
As the title, I'm trying to link my bare metal application with newlib which built with this toolchian. I noticed that  the toolchain's ld does not support -pie, and while debugging, some functions in the newlib do use absolute address.

I wonder if the toolchain can be configured with PIE support, (i.e. add some argument to `configure` or so on). Also, if not linked with -pie, would it really be PIE? 

* The ultimate goal for this is to run the application from RAM or directly from flash, which has different address. Thanks.

TommyMurphyTM1234, 2024-03-15:
Is this of any relevance?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/905#issuecomment-846562481

> Are you running program on bare-metal environment? (or in another word, not on Linux/*BSD), if so you need a program loader for resolve those relocation, PIE means you can load program to anywhere of memory, but the price is you need to fix up those address when you load program into memory.

DynamicLoader, 2024-03-16:
A little, but the linker just not support PIE (When using -Wl,-pie it just said that `can't use pie without -shared`; but passing `-shared` would cause the linker to complain that `-shared is not supported`). I'd try to resolve symbols later, but I need a toolchain that supports...

TommyMurphyTM1234, 2024-03-16:
> A little, but the linker just not support PIE (When using -Wl,-pie it just said that `can't use pie without -shared`; but passing `-shared` would cause the linker to complain that `-shared is not supported`). I'd try to resolve symbols later, but I need a toolchain that supports...

I'm not sure that PIE is appropriate for a bare-metal toolchain in the first instance? Maybe the upstream projects/forums for GCC/Binutils might be able to advise better?

DynamicLoader, 2024-03-16:
Umm, it seems to hard. Maybe I'd try to use a bootloader to do this...

closed 2024-03-16 by DynamicLoader