riscv-collab/riscv-gnu-toolchain #1438: Don't have texinfo on RHEL 8

dioncullum29, 2024-03-21:
I am on an RHEL 8 system, and this system does not have any repos available through subscriptions. This means I do not have the texinfo package installed. And because of not having repos available through subscriptions, I cannot just enable the CodeReadyBuilder repo that supposedly has texinfo included. I wanted to know if there is any way around this? Or if there is a prebuilt riscv64-unknown-glf-gcc for RHEL 8 anywhere?

Also, I apologize if this is a dumb question... I also apologize if this is not where I should be asking this. I am very new to linux, installing linux libraries, etc. All I know is I have to get some riscv commands and command options onto one of our computers so I can build an old project that utilizes those things, and this git repo is where I landed in my research. Any help would be greatly appreciated.

TommyMurphyTM1234, 2024-03-21:
> Or if there is a prebuilt riscv64-unknown-glf-gcc for RHEL 8 anywhere?

Maybe try @ilg-ul's xPack binary builds?
They're done in such a way that they're pretty portable across different Linux distros with only a dependency on the version of GLIBC.

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/
  * https://xpack.github.io/dev-tools/riscv-none-elf-gcc/
    * https://xpack.github.io/dev-tools/riscv-none-elf-gcc/releases/

dioncullum29, 2024-03-21:
Yes it is limiting, but I am dealing with my company's computers and I was trying to see if I could get by without asking one of the higher ups if we can add that subscription to what we already have. Its one of those situations where this part of the project was initially started a decade ago, on an earlier version of RHEL, and was abandoned until now where someone remembered it existed and wants it to work on our new systems. So, I am trying to configure the our new system to build the existing make files for this old project, if that makes sense.

I will look into the things you have suggested and I will message back if any of them work for me! Thank you.

ilg-ul, 2024-03-21:
> Maybe try @ilg-ul's xPack binary builds?

The xPack builds currently require GLIBC 2.27 or higher, so they are expected to run on RHEL 8 (which uses 2.28).

TommyMurphyTM1234, 2024-03-30:
Closing this since it's not an issue with the `riscv-gnu-toolchain` repo per se.

closed 2024-03-30 by TommyMurphyTM1234