riscv-collab/riscv-gnu-toolchain #1442: How to build on Windows?

inxksp, 2024-03-29:
How can I compile a riscv-tolchain on Windows?

kito-cheng, 2024-03-29:
WSL and then use mingw cross compiler (by something like `--with-host=x86_64-w64-mingw32`)...that's the best way I know.

TommyMurphyTM1234, 2024-03-29:
> How can I compile a riscv-tolchain on Windows for nuclei N300 (arch=rv32imafcp)?

Is that even a valid architecture string?
When I try to do a Linux build I get this:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1442
cd issue-1442
./configure --prefix=`pwd`/installed-tools --with-arch=rv32imafcp --with-abi=ilp32
make 2>&1 | tee build.log
...
checking for riscv32-unknown-elf-gcc... riscv32-unknown-elf-gcc -B/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv32-unknown-elf/newlib/ -isystem /home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv32-unknown-elf/newlib/targ-include -isystem /home/user/Downloads/riscv-gnu-toolchain/newlib/newlib/libc/include -B/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv32-unknown-elf/libgloss/riscv32 -L/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv32-unknown-elf/libgloss/libnosys -L/home/user/Downloads/riscv-gnu-toolchain/newlib/libgloss/riscv32
checking whether we are cross compiling... yes
checking for suffix of object files... configure: error: in `/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv32-unknown-elf/newlib':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make[2]: *** [Makefile:8448: configure-target-newlib] Error 1
make[2]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-newlib'
make[1]: *** [Makefile:879: all] Error 2
make[1]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-newlib'
make: *** [Makefile:661: stamps/build-newlib] Error 2
```

TommyMurphyTM1234, 2024-03-30:
I also tried this but it too failed:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1442
cd issue-1442
./configure --prefix=`pwd`/installed-tools --with-multilib-generator="rv32imafcp-ilp32--"
make 2>&1 | tee build.log
...
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc -B/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv32imafcp_zicsr/ilp32/newlib/ -isystem /home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv32imafcp_zicsr/ilp32/newlib/targ-include -isystem /home/user/Downloads/riscv-gnu-toolchain/newlib/newlib/libc/include -B/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv32imafcp_zicsr/ilp32/libgloss/riscv64 -L/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv32imafcp_zicsr/ilp32/libgloss/libnosys -L/home/user/Downloads/riscv-gnu-toolchain/newlib/libgloss/riscv64  -march=rv32imafcp_zicsr -mabi=ilp32
checking whether we are cross compiling... yes
checking for suffix of object files... configure: error: in `/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv32imafcp_zicsr/ilp32/newlib':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make[2]: *** [Makefile:8448: configure-target-newlib] Error 1
make[2]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-newlib'
make[1]: *** [Makefile:879: all] Error 2
make[1]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-newlib'
make: *** [Makefile:661: stamps/build-newlib] Error 2
```
Just to reitereate - I'm just trying to sanity check an `rv32imafcp/ilp32` capable toolchain build on Linux before trying to build it for Windows.

TommyMurphyTM1234, 2024-03-30:
> How can I compile a riscv-tolchain on Windows for nuclei N300 (arch=rv32imafcp)?

Is that even possible at the moment?
As far as I can see the `P` extension (which presumably encompasses some or all of `Zbpbo`, `Zpn`, `Zpsfoperand`) is not yet ratified so I'm not sure if/where any toolchain support has been prototyped?

* https://jira.riscv.org/browse/RVS-1135?src=confmacro

Presumably not in the upstream GCC and related projects where only support for ratified extensions is implemented?

TommyMurphyTM1234, 2024-03-30:
> WSL and then use mingw cross compiler (by something like `--with-host=x86_64-w64-mingw32`)...that's the best way I know.

I tried this but it failed because of unmet GCC dependencies and I'm not sure if/how to tell it to build these via the top level `riscv-gnu-toolchain` `configure/make` process? Or maybe the `riscv-gnu-toolchain` `Makefile` needs to be edited to pass `--with-gmp` etc. to build the required dependencies from sources that have been checked out a priori?
```
sudo apt install gcc-mingw-w64-x86-64
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1442
cd issue-1442
git clone https://github.com/plctlab/riscv-gcc -b riscv-gcc-p-ext riscv-gcc-p-ext
git clone https://github.com/plctlab/riscv-binutils-gdb -b riscv-binutils-p-ext riscv-binutils-p-ext
./configure --prefix=`pwd`/installed-tools --with-arch=rv32imafcp --with-abi=ilp32f --with-gcc-src=`pwd`/riscv-gcc-p-ext --with-binutils-src=`pwd`/riscv-binutils-p-ext --disable-gdb --with-host=x86_64-w64-mingw32
...
checking for the correct version of gmp.h... no
configure: error: Building GCC requires GMP 4.2+, MPFR 3.1.0+ and MPC 0.8.0+.
Try the --with-gmp, --with-mpfr and/or --with-mpc options to specify
their locations.  Source code for these libraries can be found at
their respective hosting sites as well as at
https://gcc.gnu.org/pub/gcc/infrastructure/.  See also
http://gcc.gnu.org/install/prerequisites.html for additional info.  If
you obtained GMP, MPFR and/or MPC from a vendor distribution package,
make sure that you have installed both the libraries and the header
files.  They may be located in separate packages.
make: *** [Makefile:626: stamps/build-gcc-newlib-stage1] Error 1
```
Just to note that the above build process for a native Linux build of the toolchain having a default `arch/abi` of `rv32imafcp/ilp32f` completes successfully.

TommyMurphyTM1234, 2024-03-30:
Just trying this now...
```
...
git clone https://github.com/plctlab/riscv-gcc -b riscv-gcc-p-ext riscv-gcc-p-ext
cd riscv-gcc-p-ext
./contrib/download_prerequisites
cd ..
...
```

TommyMurphyTM1234, 2024-03-30:
> Just trying this now...
> 
> ```
> ...
> git clone https://github.com/plctlab/riscv-gcc -b riscv-gcc-p-ext riscv-gcc-p-ext
> cd riscv-gcc-p-ext
> ./contrib/download_prerequisites
> cd ..
> ...
> ```

Got further but still fails unfortunately...
```
config.status: executing depdir commands
mkdir -p -- .deps
make[2]: Entering directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage1/libcpp'
g++  -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/libcpp -I. -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/libcpp/../include -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/libcpp/include  -g -O2 -W -Wall -Wno-narrowing -Wwrite-strings -Wmissing-format-attribute -pedantic -Wno-long-long  -fno-exceptions -fno-rtti -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/libcpp -I. -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/libcpp/../include -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/libcpp/include   -c -o charset.o -MT charset.o -MMD -MP -MF .deps/charset.Tpo /home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/libcpp/charset.c
In file included from /home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/libcpp/system.h:374,
                 from /home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/libcpp/charset.c:21:
/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/libcpp/../include/libiberty.h:112:14: error: ambiguating new declaration of ‘char* basename(const char*)’
  112 | extern char *basename (const char *) ATTRIBUTE_RETURNS_NONNULL ATTRIBUTE_NONNULL(1);
      |              ^~~~~~~~
In file included from /home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/libcpp/system.h:205,
                 from /home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/libcpp/charset.c:21:
/usr/include/string.h:524:26: note: old declaration ‘const char* basename(const char*)’
  524 | extern "C++" const char *basename (const char *__filename)
      |                          ^~~~~~~~
make[2]: *** [Makefile:224: charset.o] Error 1
make[2]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage1/libcpp'
make[1]: *** [Makefile:8293: all-libcpp] Error 2
make[1]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage1'
make: *** [Makefile:627: stamps/build-gcc-newlib-stage1] Error 2
```

TommyMurphyTM1234, 2024-03-30:
> WSL and then use mingw cross compiler (by something like `--with-host=x86_64-w64-mingw32`)...that's the best way I know.

I tried a few different tacks to get this working but failed.
```
sudo apt install gcc-mingw-w64-x86-64 g++-mingw-w64-x86-64
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issues-1442
cd issues-1442
git clone https://github.com/plctlab/riscv-gcc -b riscv-gcc-p-ext riscv-gcc-p-ext
git clone https://github.com/plctlab/riscv-binutils-gdb -b riscv-binutils-p-ext riscv-binutils-p-ext
cd riscv-gcc-p-ext
./contrib/download_prerequisites
cd ..
./configure  --prefix=`pwd`/installed-tools --with-arch=rv32imafcp --with-abi=ilp32f --with-gcc-src=`pwd`/riscv-gcc-p-ext --with-binutils-src=`pwd`/riscv-binutils-p-ext --disable-gdb --with-host=x86_64-w64-mingw32 --without-system-zlib
make 2>&1 | tee build.log
...

cc -Wl,--stack,12582912 -o xgcc.exe gcc.o gcc-main.o ggc-none.o \
  c/gccspec.o driver-mingw32.o  libcommon-target.a \
   libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a
cp xgcc.exe gcc-cross.exe
x86_64-w64-mingw32-g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wno-error=format-diag -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -Ic-family -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/gcc -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/gcc/c-family -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/gcc/../include -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/gcc/../libcpp/include -I/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage1/./gmp -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/gmp -I/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage1/./mpfr/src -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/mpfr/src -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/mpc/src  -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/gcc/../libdecnumber -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/gcc/../libbacktrace -I/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage1/./isl/include -I/home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/isl/include  -o c-family/cppspec.o -MT c-family/cppspec.o -MMD -MP -MF c-family/.deps/cppspec.TPo /home/user/Downloads/riscv-gnu-toolchain/riscv-gcc-p-ext/gcc/c-family/cppspec.c
x86_64-w64-mingw32-g++ -no-pie   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wno-error=format-diag -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc -Wl,--stack,12582912 -o cpp.exe gcc.o gcc-main.o ggc-none.o \
  c-family/cppspec.o driver-mingw32.o  libcommon-target.a \
   libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a
riscv32-unknown-elf-gcc -dumpspecs > tmp-specs
/bin/bash: line 1: riscv32-unknown-elf-gcc: command not found
make[2]: *** [Makefile:2090: specs] Error 127
make[2]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc'
make[1]: *** [Makefile:4414: all-gcc] Error 2
make[1]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage1'
make: *** [Makefile:627: stamps/build-gcc-newlib-stage1] Error 2
```
I gave up and had better success with crosstool-NG.
I followed the usual instructions for installing CT-NG:

* https://crosstool-ng.github.io/docs/install/
```
cd ~/Downloads
git clone https://github.com/crosstool-ng/crosstool-ng
cd crosstool-ng
./bootstrap
# Install any prerequisites and repeat until bootstrap succeeds
./configure --enable-local
# Install any prerequisites and repeat until configure succeeds
make
```
As above I installed the Windows cross compilers and cloned the `P` extension sources:
```
sudo apt install gcc-mingw-w64-x86-64 g++-mingw-w64-x86-64
cd ~/Downloads/crosstool-ng
git clone https://github.com/plctlab/riscv-gcc -b riscv-gcc-p-ext riscv-gcc-p-ext
git clone https://github.com/plctlab/riscv-binutils-gdb -b riscv-binutils-p-ext riscv-binutils-p-ext
```
I then ran `ct-ng menuconfig` to create this `.config` (renamed to allow uploading here):

* [riscv-gcc-win.config.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/14813466/riscv-gcc-win.config.txt)

I then ran `ct-ng build` and the toolchain built successfully.

To avoid problems with symbolic links I did the following:
```
cd ~/x-tools/HOST-x86_64-w64-mingw32
tar --dereference -czvf riscv32-unknown-elf.tgz riscv32-unknown-elf
# I'm using a Linux guest VM on Windows host so copy the compressed toolchain to Windows
cp riscv32-unknown-elf.tgz /media/sf_C_DRIVE/temp
# Extract the toolchain on Windows
```
In a Windows command shell:
```
c:
cd \temp\riscv32-unknown-elf\bin
riscv32-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=c:/temp/riscv32-unknown-elf/bin/../libexec/gcc/riscv32-unknown-elf/10.2.0/lto-wrapper.exe
Target: riscv32-unknown-elf
[... cut ...] -unknown-elf/buildtools/complibs-host --with-mpc=/home/user/Downloads/crosstool-ng/.build/HOST-x86_64-w64-mingw32/riscv32-unknown-elf/buildtools/complibs-host --with-isl=/home/user/Downloads/crosstool-ng/.build/HOST-x86_64-w64-mingw32/riscv32-unknown-elf/buildtools/complibs-host --enable-lto --enable-target-optspace --disable-nls --disable-multilib --enable-languages=c,c++
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.0 (crosstool-NG 1.26.0.66_6ad7b96)
```

QBos07, 2024-03-31:
Hello, ct-ng contributer here. I just wanted to say that MSYS2 is a great wsl alternative using native windows. ct-ng and gcc can but successfully compiled and used in this environment.
The toolchains don't have any dependency on the environment apart from a few DLLs wich you can just copy to the bin folder of the toolchain.

TommyMurphyTM1234, 2024-03-31:
> Hello, ct-ng contributer here. I just wanted to say that MSYS2 is a great wsl alternative using native windows. ct-ng and gcc can but successfully compiled and used in this environment. The [...]

Thanks @QBos07. Maybe things have changed in recent years but, in the past, I always found MSYS2 difficult to correctly configure and use in order to build a GCC toolchain or other stuff such as OpenOCD. I always found a CT-NG Canadian Cross compilation from Linux more effective. Or the old xPack Docker based approach. I'm only trying the latest xPack approach now to see if/how that can be used to build a custom toolchain such as the one mentioned here. Maybe I should give MSYS2 another go too? :-)

QBos07, 2024-03-31:
Definitely, I wouldn't say it works flawlessly 101% of the time but for that is autoconf. I can recommend it.

TommyMurphyTM1234, 2024-04-01:
> Definitely, I wouldn't say it works flawlessly 101% of the time but for that is autoconf. I can recommend it.

Thanks @QBos07 - unfortunately I spent several hours yesterday trying to get things to build under MSYS2 but failed miserably. As I said, this is pretty typical of my experience with MSYS2 in the past. In the end I gave up at this point:
```
gcc -c -DHAVE_CONFIG_H -g -O2      -I. -I/home/temp/riscv-gnu-toolchain/binutils/libiberty/../include  -W -Wall -Wwrite-strings -Wc++-compat -Wstrict-prototypes -Wshadow=local -pedantic  -D_GNU_SOURCE  /home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c -o pex-win32.o
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c: In function ‘pex_win32_open_read’:
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:122:10: warning: implicit declaration of function ‘_open’; did you mean ‘open’? [-Wimplicit-function-declaration]
  122 |   return _open (name, _O_RDONLY | (binary ? _O_BINARY : _O_TEXT));
      |          ^~~~~
      |          open
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:122:23: error: ‘_O_RDONLY’ undeclared (first use in this function); did you mean ‘O_RDONLY’?
  122 |   return _open (name, _O_RDONLY | (binary ? _O_BINARY : _O_TEXT));
      |                       ^~~~~~~~~
      |                       O_RDONLY
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:122:23: note: each undeclared identifier is reported only once for each function it appears in
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:122:45: error: ‘_O_BINARY’ undeclared (first use in this function); did you mean ‘O_BINARY’?
  122 |   return _open (name, _O_RDONLY | (binary ? _O_BINARY : _O_TEXT));
      |                                             ^~~~~~~~~
      |                                             O_BINARY
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:122:57: error: ‘_O_TEXT’ undeclared (first use in this function); did you mean ‘O_TEXT’?
  122 |   return _open (name, _O_RDONLY | (binary ? _O_BINARY : _O_TEXT));
      |                                                         ^~~~~~~
      |                                                         O_TEXT
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c: In function ‘pex_win32_open_write’:
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:136:18: error: ‘_O_WRONLY’ undeclared (first use in this function); did you mean ‘O_WRONLY’?
  136 |                 (_O_WRONLY | _O_CREAT | _O_TRUNC
      |                  ^~~~~~~~~
      |                  O_WRONLY
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:136:30: error: ‘_O_CREAT’ undeclared (first use in this function); did you mean ‘O_CREAT’?
  136 |                 (_O_WRONLY | _O_CREAT | _O_TRUNC
      |                              ^~~~~~~~
      |                              O_CREAT
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:136:41: error: ‘_O_TRUNC’ undeclared (first use in this function); did you mean ‘O_TRUNC’?
  136 |                 (_O_WRONLY | _O_CREAT | _O_TRUNC
      |                                         ^~~~~~~~
      |                                         O_TRUNC
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:137:30: error: ‘_O_BINARY’ undeclared (first use in this function); did you mean ‘O_BINARY’?
  137 |                  | (binary ? _O_BINARY : _O_TEXT)),
      |                              ^~~~~~~~~
      |                              O_BINARY
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:137:42: error: ‘_O_TEXT’ undeclared (first use in this function); did you mean ‘O_TEXT’?
  137 |                  | (binary ? _O_BINARY : _O_TEXT)),
      |                                          ^~~~~~~
      |                                          O_TEXT
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:138:17: error: ‘_S_IREAD’ undeclared (first use in this function); did you mean ‘S_IREAD’?
  138 |                 _S_IREAD | _S_IWRITE);
      |                 ^~~~~~~~
      |                 S_IREAD
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:138:28: error: ‘_S_IWRITE’ undeclared (first use in this function); did you mean ‘S_IWRITE’?
  138 |                 _S_IREAD | _S_IWRITE);
      |                            ^~~~~~~~~
      |                            S_IWRITE
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c: In function ‘pex_win32_close’:
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:146:10: warning: implicit declaration of function ‘_close’; did you mean ‘close’? [-Wimplicit-function-declaration]
  146 |   return _close (fd);
      |          ^~~~~~
      |          close
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c: In function ‘win32_spawn’:
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:670:13: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  670 |       pid = (pid_t) pi->hProcess;
      |             ^
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c: In function ‘spawn_script’:
/home/temp/riscv-gnu-toolchain/binutils/libiberty/pex-win32.c:695:31: error: ‘_O_RDONLY’ undeclared (first use in this function); did you mean ‘O_RDONLY’?
  695 |   int fd = _open (executable, _O_RDONLY);
      |                               ^~~~~~~~~
      |                               O_RDONLY
[... cut ...]
If you know of any reliable instructions for building the RISC-V GCC toolchain - or maybe even any GCC toolchain - using MSYS2 then please link to them. Thanks.

It also looks like one needs to be careful of the location of the MSYS home directory as mine ended up using spaces which messed things up, and probably also needs to enable NTFS case sensitivity for the MSYS2 directory in order to avoid problems with Linux headers with the same names but different case:

* https://learn.microsoft.com/en-us/windows/wsl/case-sensitivity

TommyMurphyTM1234, 2024-04-06:
I was able to build a bare-metal toolchain (but not including GDB yet) using MSYS2 as follows:

1. Create an install directory for MSYS2 - e.g. `C:\msys64`.
2. Enable NTFS filesystem case sensitivity in order to avoid problems with files with the same name but different case (e.g. Linux headers): https://learn.microsoft.com/en-us/windows/wsl/case-sensitivity. In a PowerShell terminal with administrative privileges run this command: `fsutil.exe file setCaseSensitiveInfo C:\msys64\ enable`.
3. Download the MSYS2 installer from https://www.msys2.org/ run it and install into `C:\msys64`.
4. Uncomment `MSYS=winsymlinks:nativestrict` in `C:\msys64\ucrt64.ini` to avoid problems with `ln -s` symbolic links that the build process creates.
5. Run an MSYS2 UCRT64 terminal: _Start Menu > MSYS2 > MSYS2 UCRT64_.
6. Install these dependencies: `pacman -S mingw-w64-ucrt-x86_64-gcc autoconf automake autotools curl python3 python3-pip gawk bison flex texinfo gperf libtool patchutils bc libexpat expat ninja git cmake`.
7. Run `pwd` to display the home directory - if it contains spaces ("e.g. mine was `/home/Tommy Murphy`) thense these will cause problems so create and use a directory without spaces - e.g. `cd /home && mkdir riscv && cd riscv`.
8. Do a build as normal (disable GDB for now until I figure that out):
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --disable-gdb
make 2>&1 | tee build.log
```
TODO:

1. Figure out how to build GDB - haven't been able to figure out what `expat` package meets its dependency requirements: https://packages.msys2.org/search?t=pkg&q=expat
2. Build Linux toolchain.
3. Build other stuff - e.g. Spike/QEMU, LLVM etc.

Notes:

1. crosstool-NG seems much easier to configure/use and builds a toolchain much quicker.

TommyMurphyTM1234, 2024-04-07:
> 1. Figure out how to build GDB - haven't been able to figure out what `expat` package meets its dependency requirements: https://packages.msys2.org/search?t=pkg&q=expat

For some reason just installing the `expat` package (`pacman -S expat`) didn't help and the build failed looking for `libexpat`. I tried `pacman -S mingw-w64-expat` that failed:
```
pacman -S mingw-w64-expat
error: target not found: mingw-w64-expat
```
In the end I just downloaded the `expat` sources and built them myself and then the build worked:
```
cd /home/temp
# https://github.com/libexpat/libexpat/releases
wget https://github.com/libexpat/libexpat/releases/download/R_2_6_2/expat-2.6.2.tar.gz
tar xzvf expat-2.6.2.tar.gz
cd expat-2.6.2
./configure 
make install

cd /home/temp
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools
make

ls installed-tools/bin/
riscv64-unknown-elf-addr2line.exe  riscv64-unknown-elf-cpp.exe         riscv64-unknown-elf-gcc-nm.exe      riscv64-unknown-elf-gcov.exe       riscv64-unknown-elf-ld.exe        riscv64-unknown-elf-ranlib.exe   riscv64-unknown-elf-strip.exe
riscv64-unknown-elf-ar.exe         riscv64-unknown-elf-elfedit.exe     riscv64-unknown-elf-gcc-ranlib.exe  riscv64-unknown-elf-gdb-add-index  riscv64-unknown-elf-lto-dump.exe  riscv64-unknown-elf-readelf.exe
riscv64-unknown-elf-as.exe         riscv64-unknown-elf-g++.exe         riscv64-unknown-elf-gcc.exe         riscv64-unknown-elf-gdb.exe        riscv64-unknown-elf-nm.exe        riscv64-unknown-elf-run.exe
riscv64-unknown-elf-c++.exe        riscv64-unknown-elf-gcc-13.2.0.exe  riscv64-unknown-elf-gcov-dump.exe   riscv64-unknown-elf-gprof.exe      riscv64-unknown-elf-objcopy.exe   riscv64-unknown-elf-size.exe
riscv64-unknown-elf-c++filt.exe    riscv64-unknown-elf-gcc-ar.exe      riscv64-unknown-elf-gcov-tool.exe   riscv64-unknown-elf-ld.bfd.exe     riscv64-unknown-elf-objdump.exe   riscv64-unknown-elf-strings.exe
```
I did also had to manually copy the following DLLs in order for GDB to launch correctly:
```
cp /ucrt64/bin/libgmp-10.dll /home/temp/riscv-gnu-toolchain/installed-tools/bin
cp /ucrt64/bin/libmpfr-6.dll /home/temp/riscv-gnu-toolchain/installed-tools/bin
cp /ucrt64/bin/libzstd.dll /home/temp/riscv-gnu-toolchain/installed-tools/bin
cp /ucrt64/bin/libgcc_s_seh-1.dll /home/temp/riscv-gnu-toolchain/installed-tools/bin
cp /ucrt64/bin/libiconv-2.dll /home/temp/riscv-gnu-toolchain/installed-tools/bin
cp /ucrt64/bin/libwinpthread-1.dll /home/temp/riscv-gnu-toolchain/installed-tools/bin
cp /ucrt64/bin/libstdc++-6.dll /home/temp/riscv-gnu-toolchain/installed-tools/bin
cp /ucrt64/bin/libexpat-1.dll /home/temp/riscv-gnu-toolchain/installed-tools/bin
```
I haven't tried all of the toolchain executables so it's possible that there are other dependent DLLs but I think that the principle of compiling the Windows native bare-metal toolchain using MSYS2 has been illustrated adequately.

QBos07, 2024-04-07:
The binary version of expat is distrubuted as [mingw-w64-ucrt-x86_64-expat](https://packages.msys2.org/package/mingw-w64-ucrt-x86_64-expat?repo=ucrt64) for this platform.

TommyMurphyTM1234, 2024-04-07:
> The binary version of expat is distrubuted as [mingw-w64-ucrt-x86_64-expat](https://packages.msys2.org/package/mingw-w64-ucrt-x86_64-expat?repo=ucrt64) for this platform.

Thanks @QBos07 - I didn't realise that I was searching only base packages and not all packages earlier! 

* https://packages.msys2.org/search?t=binpkg&q=expat

TommyMurphyTM1234, 2024-04-08:
> I haven't tried all of the toolchain executables so it's possible that there are other dependent DLLs but I think that the principle of compiling the Windows native bare-metal toolchain using MSYS2 has been illustrated adequately.

I guess that these DLLs probably need to be copied to the toolchain `bin` directory...
```
 ldd installed-tools/bin/*.exe | grep ucrt64 | rev | cut -c17- | rev | sort | uniq
        libexpat-1.dll => /ucrt64/bin/libexpat-1.dll
        libgcc_s_seh-1.dll => /ucrt64/bin/libgcc_s_seh-1.dll
        libgmp-10.dll => /ucrt64/bin/libgmp-10.dll
        libiconv-2.dll => /ucrt64/bin/libiconv-2.dll
        libisl-23.dll => /ucrt64/bin/libisl-23.dll
        libmpc-3.dll => /ucrt64/bin/libmpc-3.dll
        libmpfr-6.dll => /ucrt64/bin/libmpfr-6.dll
        libstdc++-6.dll => /ucrt64/bin/libstdc++-6.dll
        libwinpthread-1.dll => /ucrt64/bin/libwinpthread-1.dll
        libzstd.dll => /ucrt64/bin/libzstd.dll
        zlib1.dll => /ucrt64/bin/zlib1.dll
```

TommyMurphyTM1234, 2024-04-09:
Yes, but I'm thinking of the situation where one builds a native Windows toolchain for use by somebody who doesn't have/want to install MSYS2.

ilg-ul, 2024-04-09:
> It's possible to just add `C:\msys64\ucrt\bin` to the PATH

Depending on the position in the PATH, this might break other programs using different versions of these libraries.

I'm no Windows specialist, so I might be wrong, but I think that the safest location is the same folder as the executable that uses the library.

TommyMurphyTM1234, 2024-06-01:
Two methods for building a Windows toolchain have been shown to work:

1. Canadian Cross toolchain buid using crosstool-NG on Linux to build a toolchain to run on Windows that generates code for RISC-V
2. Building natively on Windows using MSYS2

I haven't had time to pursue a third way:

3. xPack

I will try to look into that at some point but for now I will close this issue as the original question has been addressed.

closed 2024-06-01 by TommyMurphyTM1234