riscv-collab/riscv-gnu-toolchain #1443: Where to look at if buildroot/linux does not recognize vector extension instructions with illegal instruction trap even after enabling vector extension patch?

NazerkeT, 2024-03-30:
Hi there,

This question might be a bit sideways, however I wanted to ask in case someone has a clue about this. I was trying to run vector extension enabled app on linux+spike, however while launching the application it keeps giving an Illegal Instruction error whenever it sees the first vector instruction/intrinsic. In order to solve this problem, I tried to configure buildroot/linux to choose rv64gcv according to this [commit](https://github.com/buildroot/buildroot/commit/4d70454754142a6334ec4a7e8cc6582bbf2f4b9d)  and I could see how OpenSBI shows vector extension as part of base ISA as well. Nevertheless, I am still keep having this illegal instruction error. 

Do you have any suggestions on where I can look at it?

Thanks,
Best wishes

TommyMurphyTM1234, 2024-03-30:
It might help to show how you're invoking Spike and what options, in particular what `--isa=...` and maybe `--varch=...` options, you're passing?

* https://github.com/riscv-software-src/riscv-isa-sim/blob/3427b459f88d2334368a1abbdf5a3000957f08e8/spike_main/spike.cc#L44
* https://github.com/riscv-software-src/riscv-isa-sim/blob/3427b459f88d2334368a1abbdf5a3000957f08e8/spike_main/spike.cc#L48

It might be a better idea to ask about this sort stuff on the Spike GitHub repo if it's not addressed there already?

* https://github.com/riscv-software-src/riscv-isa-sim/issues

TommyMurphyTM1234, 2024-04-02:
Any update @NazerkeT?

NazerkeT, 2024-04-02:
Hi @TommyMurphyTM1234 ,

Thanks for your response and related links!

In general, I build and run spike with respective --isa=rv64gcv, also set varch to be of vlen 256, and elen of 32 and 64 at various times. However, it did not work :(

I did not had a time to debug it after this post unfortunately, so do not have updates yet. 

Nevertheless, so far due to other deadlines, seems like I will not get to debug this project of mine in some time, so we can close the issue here and I can follow up on it when the time comes. 

Thanks,
Best wishes