riscv-collab/riscv-gnu-toolchain #1444: Why does my custom instruction simulation run incorrectly？

niuzhi1221, 2024-03-31:
After seeing this C function:
        `double clamp (double x ) { return x < 0 ? 0 : x > 1 ? 1 : x ; }`
I want to design a custom instruction named 'clp' , it uses F extesnion. The instruction format I designed is as follows:
`clp       rd rs1 rs2 rs3 rm 26..25=1 6..2=0x15 1..0=3`
First , I add the MASK_CLP and MATCH_CLP to `riscv-gnu-toolchain/riscv-binutils /include/opcode/riscv-opc.h`. 
Then, in `riscv-gnu-toolchain/riscv-binutils/opcodes/riscv-opc.c` I add the following content:
`{"clp",        0, INSN_CLASS_D_INX,   "D,S,T,R",       MATCH_CLP|MASK_RM, MASK_CLP|MASK_RM, match_opcode, 0},
{"clp",        0, INSN_CLASS_D_INX,   "D,S,T,R,m",       MATCH_CLP, MASK_CLP, match_opcode, 0},`
After that , I recompiled my toolchain.
To achieve command functionality, I want to simulate the `clp` instruction in Spike.So I firstly add MATCH_CLP and MASK_CLP in `spike/riscv/encoding.h`, then I make a file named clp.h in `spike/riscv/insns`, I add the following content:
`require_either_extension('D', EXT_ZDINX);`
`require_fp;`
`softfloat_roundingMode = RM;`
`if(f64_lt(FRS1_D, FRS2_D)){`
	`WRITE_FRD_D(FRS1_D);`
`}`
`else if(f64_lt(FRS3_D,FRS1_D)){`
	`WRITE_FRD_D(FRS3_D);`
`}`
`else `
	`WRITE_FRD_D(FRS2_D);`
`set_fp_exceptions;`
Then , in `spike/riscv/riscv.mk.in` file, I add this:
[image]
in `spike/disasm/disasm.cc` file , I add this:
[image]
Then I recompiled my spike.
I write a test C program named `clamp.c`:
[image]
But I can't use command `spike pk clamp` to simulate it:
 
[image]
Sincerely looking forward to your help！

TommyMurphyTM1234, 2024-03-31:
This issue tracker is for reporting problems and bugs with the `riscv-gnu-toolchain` tools repo.
It's not for general support on adding custom instructions to the toolchain or troubleshooting Spike issues.
There is ample information about adding custom instructions to GCC generally and for RISC-V specifically elsewhere (including in the open and closed issues here but also upstream and on third party sites).
You need to research these in order to ensure that you are adding your custom instruction to GCC correctly.
It might also be worthwhile using the relevant upstream GCC and Spike forums to raise such questions.

niuzhi1221, 2024-03-31:
> This issue tracker is for reporting problems and bugs with the `riscv-gnu-toolchain` tools repo. It's not for general support on adding custom instructions to the toolchain or troubleshooting Spike [...]

> This issue tracker is for reporting problems and bugs with the `riscv-gnu-toolchain` tools repo. It's not for general support on adding custom instructions to the toolchain or troubleshooting Spike [...]

OK，thanks！！！

niuzhi1221, 2024-03-31:
> This issue tracker is for reporting problems and bugs with the `riscv-gnu-toolchain` tools repo. It's not for general support on adding custom instructions to the toolchain or troubleshooting Spike [...]

Can you recommend several upstream websites? I really want to solve this problem.

TommyMurphyTM1234, 2024-03-31:
> Can you recommend several upstream websites? I really want to solve this problem.

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues?q=is%3Aissue+add+custom+instruction
* https://gcc.gnu.org/
* https://www.google.com/search?q=riscv+custom+instruction

closed 2024-03-31 by TommyMurphyTM1234