riscv-collab/riscv-gnu-toolchain #1446: riscv64-unknown-elf-gcc rejects x0 as destination register

europe2024, 2024-04-02:
riscv64-unknown-elf-gcc will translate

     la x1,symbol

into machine code.

riscv64-unknown-elf-gcc should also translate

     la x0,symbol

into machine code but will reject the pseudo-instruction by displaying the following message.

     Error: illegal operands `la x0,symbol'

TommyMurphyTM1234, 2024-04-02:
> riscv64-unknown-elf-gcc should also translate
> 
> ```
>  la x0,symbol
> ```
> 
> into machine code but will reject the pseudo-instruction by displaying the following message.
> 
> ```
>  Error: illegal operands `la x0,symbol'
> ```

Works for me:

* https://godbolt.org/z/M5Ev8ncjn

cmuellner, 2024-04-02:
`la` is not an instruction but a pseudo-instruction that is defined here: https://github.com/riscv-non-isa/riscv-asm-manual/blob/master/riscv-asm.md#-a-listing-of-standard-risc-v-pseudoinstructions

`la rd, symbol` expands to `auipc rd, symbol[31:12]; addi rd, rd, symbol[11:0]`.
In the case of `rd=x0`, we get `auipc x0, ...`, a reserved opcode for the HINT instruction.

To avoid unintended use of a `HINT` instruction, there is no support for `x0` as `rd` for `la` in binutils (GCC does not care about that).

europe2024, 2024-04-02:
host processor: ARM
host OS: variant of Debian (version 12)
GNU assembler (GNU Binutils) version: 2.42

riscv64-unknown-elf-as also displayed

     Error: illegal operands `la x0,symbol'

for

     la x0,symbol

TommyMurphyTM1234, 2024-04-02:
> To avoid unintended use of a `HINT` instruction, there is no support for `x0` as `rd` for `la` in binutils (GCC does not care about that).

Ah - OK - so the issue doesn't appear on Compiler Explorer because it's just generating assembler code from the C code and not actually assembling it to object format?

I see that when I do that using a local toolchain I do get the same error messages:
```
cat test.c

// Type your code here, or load an example.
void f(int num) {
    asm("la x0, num");
}

riscv64-unknown-elf-gcc -O0 -c test.c
test.c: Assembler messages:
test.c:3: Error: illegal operands `la x0,num'

riscv64-unknown-elf-gcc -O0 -S test.c

cat test.s

        .file   "test.c"
        .option nopic
        .attribute arch, "rv64i2p1_m2p0_a2p1_f2p2_d2p2_c2p0_zicsr2p0"
        .attribute unaligned_access, 0
        .attribute stack_align, 16
        .text
        .align  1
        .globl  f
        .type   f, @function
f:
        addi    sp,sp,-32
        sd      s0,24(sp)
        addi    s0,sp,32
        mv      a5,a0
        sw      a5,-20(s0)
 #APP
# 3 "test.c" 1
        la x0, num
# 0 "" 2
 #NO_APP
        nop
        ld      s0,24(sp)
        addi    sp,sp,32
        jr      ra
        .size   f, .-f
        .ident  "GCC: () 13.2.0"

riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/Downloads/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv64-unknown-elf/13.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/user/Downloads/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-elf --prefix=/home/user/Downloads/riscv-gnu-toolchain/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 13.2.0 ()
```

TommyMurphyTM1234, 2024-04-02:
> `la rd, symbol` expands to `auipc rd, symbol[31:12]; addi rd, rd, symbol[11:0]`. In the case of `rd=x0`, we get `auipc x0, ...`, a reserved opcode for the HINT instruction.
>
> To avoid unintended use [...]

Just for reference purposes the `HINT` instructions are described in the ISA specification.

* https://github.com/riscv/riscv-isa-manual/releases/tag/riscv-isa-release-af07c84-2024-03-26
  * Section 2.9 - RV32I `HINT` instructions
  * Section 4.4 - RV64I `HINT` instructions
  * Section 26.7 - RVC (Compressed) extension `HINT` instructions

TommyMurphyTM1234, 2024-04-03:
Closing this issue since the reason for the error messages has been explained by @cmuellner here: 

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1446#issuecomment-2032701401

closed 2024-04-03 by TommyMurphyTM1234