riscv-collab/riscv-gnu-toolchain #1447: adding custom branch instruction

jammyb1, 2024-04-02:
I am adding a custom branch instruction which should be assembled into machine code in exactly the same way as any of the other B-type instructions (except with a func3 of 010).

I had added `{"bir", 32, INSN_CLASS_I, "s,t,p", 0x2063, MASK_BEQ, match_opcode, 0 },` and then `DECLARE_INSN(bir, 0x2063, MASK_BEQ)` to set this all up but when I go to write some C code with inline assembly (`asm volatile ("bir zero, zero, 50");`, don't worry about the fact both registers are 0, the branch instruction is not using registers and using custom hardware) it produces `.insn 4, 0x3463`.

I have added lots of other custom instructions using the above method and they all seem to work fine - it's just this branch that doesn't. Does it have something to do with the assembler not knowing how to work out what immediate value to create in the machine code from what the PC will be and what the number you pass in (50 in the code above)?

TommyMurphyTM1234, 2024-04-02:
You'll probably need to explain exactly (all steps/changes) how you're adding your custom instruction to the toolchain.

jammyb1, 2024-04-02:
> You'll probably need to explain exactly (all steps/changes) how you're adding your custom instruction to the toolchain.

Updated the original post

TommyMurphyTM1234, 2024-04-05:
SInce you're not getting any feedback here you might be better off raising this issue on an upstream GCC forum/mailing list where there may be more people better placed to offer advice.

* https://gcc.gnu.org/lists.html

pz9115, 2024-04-08:
I think your way of adding the new branch instruction is totally right.
One thing that you omit is when you try to disassemble the branch instructions,
PC will always be active in it.

> I had added `{"bir", 32, INSN_CLASS_I, "s,t,p", 0x2063, MASK_BEQ, match_opcode, 0 },` 

You used 's,t,p' as the operand for instruction `bir`, when the 'p' operand disassemble,
it will add pc back to your immediate (defines in [riscv-dis.c](https://github.com/bminor/binutils-gdb/blob/261d6a6776b6d0652eab37552450e94cb9eec780/opcodes/riscv-dis.c#L488) ).

So here it an assemble code example(branch.s):
```
branch:
        .option norvc
        beq a0, a1, 0
        bne a0, a1, 0
```
when we use as to assemble it (like `riscv32-unknown-elf-as -c branch.s -o branch.o`) and we will get such disassemble code by use objdump(`riscv32-unknown-elf-objdump -d branch.o`):
```
Disassembly of section .text:

00000000 <branch>:
   0:   00b51463                bne     a0,a1,8 <branch+0x8>
   4:   0000006f                j       4 <branch+0x4>
   8:   00b50463                beq     a0,a1,10 <branch+0x10>
   c:   0000006f                j       c <branch+0xc>
```
PC changes the output instruction name sequence and use the current address symbol at all.

If you want to make disassembler print a name form instruction but not an opcode one, my suggestion is that you should add one more branch instruction with match equal '0x3063'(eg. 'bor'), it will be paired with your 'bir'. When the 'bir' added PC, it will show the instruction 'bor' in the disassemble output, and vice versa.

pz9115, 2024-04-08:
I think your way of adding the new branch instruction is totally right.
One thing that you omit is when you try to disassemble the branch instructions,
PC will always be active in it.

> I had added `{"bir", 32, INSN_CLASS_I, "s,t,p", 0x2063, MASK_BEQ, match_opcode, 0 },` 

You used 's,t,p' as the operand for instruction `bir`, when the 'p' operand disassemble,
it will add pc back to your immediate (defines in [riscv-dis.c](https://github.com/bminor/binutils-gdb/blob/261d6a6776b6d0652eab37552450e94cb9eec780/opcodes/riscv-dis.c#L488) ).

So here it an assemble code example(branch.s):
```
branch:
        .option norvc
        beq a0, a1, 0
        bne a0, a1, 0
```
when we use as to assemble it (like `riscv32-unknown-elf-as -c branch.s -o branch.o`) and we will get such disassemble code by use objdump(`riscv32-unknown-elf-objdump -d branch.o`):
```
Disassembly of section .text:

00000000 <branch>:
   0:   00b51463                bne     a0,a1,8 <branch+0x8>
   4:   0000006f                j       4 <branch+0x4>
   8:   00b50463                beq     a0,a1,10 <branch+0x10>
   c:   0000006f                j       c <branch+0xc>
```
PC changes the output instruction name sequence and use the current address symbol at all.

If you want to make disassembler print a name form instruction but not an opcode one, my suggestion is that you should add one more branch instruction with match equal '0x3063'(eg. 'bor'), it will be paired with your 'bir'. When the 'bir' added PC, it will show the instruction 'bor' in the disassemble output, vice versa.

TommyMurphyTM1234, 2024-04-11:
Any update @jammyb1 ?
Did @pz9115's suggestion help at all?
Have you asked about this upstream?

TommyMurphyTM1234, 2024-04-29:
Since there's no update for several weeks I'm closing this issue.

closed 2024-04-29 by TommyMurphyTM1234