riscv-collab/riscv-gnu-toolchain #1448: How to compile with Zicond?

Ye-Jinhong, 2024-04-03:
I want to compare difference between rv64gc and rv64gc_zicond, but there is no instruction of Zicond generated by gcc. 

**toolchain config**
./configure --march=rv64gc_zba_zbb_zbc_zbs_zifence_zicsr_zicond_zcb_zfa

**code**
coremark

**compile options**
CC = riscv64-unknown-linux-gnu-gcc
CFLAGS = -march=rv64gc{_zicond}

**others**
I heard that GCC13.2 does not choose to generate Zicond instruction but GCC14.0 dose. [link](https://stackoverflow.com/questions/77350365/how-to-compile-for-riscv-zicond-extension-in-gcc)

TommyMurphyTM1234, 2024-04-03:
Did you try this already?
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --with-arch=rv64gc_zba_zbb_zbc_zbs_zifence_zicsr_zicond_zcb_zfa --with-abi=lp64d
make 2>&1 | tee build.log
```
If that doesn't build, or does build but doesn't generate `Zicond` instructions, then try building with the latest upstream GCC sources:

* https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#use-source-tree-other-than-riscv-gnu-toolchain

It looks like `Zicond` support development is ongoing in upstream GCC:

* https://gcc.gnu.org/pipermail/gcc-cvs/2023-August/387594.html

And is already supported by Binutils 2.41:

* https://www.phoronix.com/news/GNU-Binutils-2.41-Branched

And `riscv-gnu-toolchain` already uses 2.42:

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/9a28c809a5d355874def06a414aa5f272fae564d/.gitmodules#L1-L4
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc gcc-master
./configure --with-arch=rv64gc_zba_zbb_zbc_zbs_zifence_zicsr_zicond_zcb_zfa --with-abi=lp64d --with-gcc-src=`pwd`/gcc-master
make 2>&1 | tee build.log
```

TommyMurphyTM1234, 2024-04-04:
Has `Zicond` support actually been added to GCC?

`Zicond` was ratified November 2023:

* https://wiki.riscv.org/display/HOME/Ratified+Extensions#:~:text=RISC%2DV%20Integer%20Conditional%20(Zicond,Zicond

There hasn't been a GCC release since then:

* https://gcc.gnu.org/releases.html

Perhaps support is in the upstream GCC master repo?

* https://www.mail-archive.com/search?l=gcc-patches%40gcc.gnu.org&q=zicond
* https://github.com/gcc-mirror/gcc

Or maybe implementation is being done in some other GCC forked repo?

As with other extensions, trying to keep track of where support is developed is very unclear/confusing (to me at least!).

pz9115, 2024-04-04:
Zicond has already supported on GCC upstream.
https://gcc.gnu.org/git/?p=gcc.git&a=search&h=HEAD&st=commit&s=zicond

So you can update your gcc branch and use '-march' option with 'zicond' to enable Zicond extension, like '-march=rv64gc_zicond'

Here is a gcc test example:
https://gcc.gnu.org/git/?p=gcc.git;a=blob;f=gcc/testsuite/gcc.target/riscv/cset-sext-zicond-rtl.c;h=f6c82d5a5d055fc4e2d4bd8f07f90dc3c51a06d8;hb=5a874dec6006357cca4e5f10a1c399157b96b837

TommyMurphyTM1234, 2024-04-04:
> Zicond has already supported on GCC upstream. https://gcc.gnu.org/git/?p=gcc.git&a=search&h=HEAD&st=commit&s=zicond
>
> So you can update your gcc branch and use '-march' option with 'zicond' to enable [...]

Thanks a lot @pz9115.

So if this is @Ye-Jinhong's intention:

> I want to compare difference between rv64gc and rv64gc_zicond,

then this should be feasible:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc gcc-master
# Not sure if it should maybe be rv64gc_zicsr_zifencei_zicond?
./configure --prefix=`pwd`/installed-tools --with-multilib-generator="rv64gc_zicond-lp64d--" --with-gcc-src=`pwd`/gcc-master
make

# Test using rv64gc/lp64d (toolchain default arch/abi)
riscv64-unknown-elf-gcc test.c -o test

# Test using rv64gc_zicond/lp64d
riscv64-unknown-elf-gcc -march=rv64gc_zicond test.c -o test
```

TommyMurphyTM1234, 2024-04-04:
> Here is a gcc test example: https://gcc.gnu.org/git/?p=gcc.git;a=blob;f=gcc/testsuite/gcc.target/riscv/cset-sext-zicond-rtl.c;h=f6c82d5a5d055fc4e2d4bd8f07f90dc3c51a06d8;hb=5a874dec6006357cca4e5f10a1c399157b96b837

Hi @pz9115 - that example seems to require the use of DejaGnu in the context of the GCC test suite, and isn't really usable directly as far as I can see. Are you aware of any "plain" C code examples that can be used "standalone" by any chance?

pz9115, 2024-04-04:
> ee. Are you aware of any "plain" C code examples that can be used "standalone" by any chance?

Ok, here is another simple c C code example that compare between use 'Zicond' and without it: https://godbolt.org/z/1PMq58Khs

TommyMurphyTM1234, 2024-04-05:
Unfortunately building from upstream GCC sources doesn't seem to work right now for some reason (note that I'm doing a regular build here and not even trying to exercise `Zicond`):
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc gcc-master
./configure --prefix=`pwd`/installed-tools --with-gcc-src=`pwd`/gcc-master
make 2>&1 | tee build.log

...
make[4]: Entering directory '/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib'
  GEN      stmp-targ-include
make[5]: Entering directory '/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib'
  GEN      targ-include
  GEN      targ-include/_newlib_version.h
  GEN      targ-include/newlib.h
  GEN      targ-include/sys
  GEN      targ-include/machine
  GEN      targ-include/bits
make[5]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib'
make[5]: Entering directory '/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib'
  GEN      targ-include/_newlib_version.h
  GEN      targ-include/newlib.h
make[5]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib'
  CC       libm/math/libm_a-k_standard.o
  CC       libm/math/libm_a-k_rem_pio2.o

...

  CC       libm/complex/libm_a-ccoshl.o
/home/user/Downloads/riscv-gnu-toolchain/newlib/newlib/libm/complex/ccoshl.c: In function 'ccoshl':
/home/user/Downloads/riscv-gnu-toolchain/newlib/newlib/libm/complex/ccoshl.c:43:13: error: implicit declaration of function 'coshl'; did you mean 'coshf'? [-Wimplicit-function-declaration]
   43 |         w = coshl(x) * cosl(y) + (sinhl(x) * sinl(y)) * I;
      |             ^~~~~
      |             coshf
/home/user/Downloads/riscv-gnu-toolchain/newlib/newlib/libm/complex/ccoshl.c:43:24: error: implicit declaration of function 'cosl'; did you mean 'cosf'? [-Wimplicit-function-declaration]
   43 |         w = coshl(x) * cosl(y) + (sinhl(x) * sinl(y)) * I;
      |                        ^~~~
      |                        cosf
/home/user/Downloads/riscv-gnu-toolchain/newlib/newlib/libm/complex/ccoshl.c:43:35: error: implicit declaration of function 'sinhl'; did you mean 'sinhf'? [-Wimplicit-function-declaration]
   43 |         w = coshl(x) * cosl(y) + (sinhl(x) * sinl(y)) * I;
      |                                   ^~~~~
      |                                   sinhf
/home/user/Downloads/riscv-gnu-toolchain/newlib/newlib/libm/complex/ccoshl.c:43:46: error: implicit declaration of function 'sinl'; did you mean 'sinf'? [-Wimplicit-function-declaration]
   43 |         w = coshl(x) * cosl(y) + (sinhl(x) * sinl(y)) * I;
      |                                              ^~~~
      |                                              sinf
make[4]: *** [Makefile:41572: libm/complex/libm_a-ccoshl.o] Error 1
make[4]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib'
make[3]: *** [Makefile:5038: all] Error 2
make[3]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib'
make[2]: *** [Makefile:8492: all-target-newlib] Error 2
make[2]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-newlib'
make[1]: *** [Makefile:879: all] Error 2
make[1]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-newlib'
make: *** [Makefile:661: stamps/build-newlib] Error 2
```
I've tried this several times and from a clean base but I get the same error every time - oddly while compiling Newlib libs. Maybe upstream GCC requires some later version of Newlib?

Compressed `build.log`: [build.log.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/14881266/build.log.zip)

TommyMurphyTM1234, 2024-04-05:
> Maybe upstream GCC requires some later version of Newlib?

Looks like that's the case.
A test build with the latest Newlib master sources worked.
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc gcc-master
git clone https://github.com/bminor/newlib newlib-master
./configure --prefix=`pwd`/installed-tools --with-gcc-src=`pwd`/gcc-master --with-newlib-src=`pwd`/newlib-master
make 2>&1 | tee build.log
```
I'm just trying again with this:
```
./configure --prefix=`pwd`/installed-tools --with-gcc-src=`pwd`/gcc-master --with-newlib-src=`pwd`/newlib-master --with-multilib-generator="rv64gc_zicond-lp64d--"
make 2>&1 | tee build.log
```

TommyMurphyTM1234, 2024-04-05:
OK - so this worked for me:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/bminor/newlib newlib-master
git clone https://github.com/gcc-mirror/gcc gcc-master
./configure --prefix=`pwd`/installed-tools --disable-gdb --with-gcc-src=`pwd`/gcc-master --with-newlib-src=`pwd`/newlib-master --with-multilib-generator="rv64gc_zicond-lp64d--"
make 2>&1 | tee build.log

cd installed-tools/bin
./riscv64-unknown-elf-gcc -print-multi-lib
.;
rv64imafdc_zicond_zicsr_zifencei/lp64d;@march=rv64imafdc_zicond_zicsr_zifencei@mabi=lp64d

# Note that ".;" refers to the default arch/abi = rv64imafdc/lp64d
./riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/Downloads/issue-1448/installed-tools/libexec/gcc/riscv64-unknown-elf/14.0.1/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/user/Downloads/issue-1448/gcc-master/configure --target=riscv64-unknown-elf --prefix=/home/user/Downloads/issue-1448/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=g4c8b3600c48 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/Downloads/issue-1448/installed-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=/home/user/Downloads/issue-1448/gcc-master --enable-multilib --with-multilib-generator=rv64gc_zicond-lp64d-- --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 14.0.1 20240405 (experimental) (g4c8b3600c48)

# The default arch/abi is actually synonymous with rv64imafdc_zicsr/lp64d
./riscv64-unknown-elf-gcc -dumpspecs

...

*multilib_defaults:
march=rv64imafdc_zicsr mabi=lp64d

...
```
And then a test:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1448#issuecomment-2037398902
* https://godbolt.org/z/1PMq58Khs
```
# test.c
int foo (long a, long b)
{
  if (!a)
    return 0;
  else if (b)
    return 1;
  else
    return 0;
}
```
Using `rv64gc[_zicsr]/lp64d`:
```
./riscv64-unknown-elf-gcc -O2 -c test.c

./riscv64-unknown-elf-objdump -dS test.o

test.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <foo>:
   0:   c501                    beqz    a0,8 <.L3>
   2:   00b03533                snez    a0,a1
   6:   8082                    ret

0000000000000008 <.L3>:
   8:   4501                    li      a0,0
   a:   8082                    ret
```
And now using `rv64gc_zicond/lp64d`:
```
./riscv64-unknown-elf-gcc -march=rv64gc_zicond -O2 -c test.c

./riscv64-unknown-elf-objdump -dS test.o

test.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <foo>:
   0:   00b035b3                snez    a1,a1
   4:   0ea5d533                czero.eqz       a0,a1,a0
   8:   8082                    ret
```
So - finally! - the original question has been addressed and I think that we can close this. 😄

closed 2024-04-05 by TommyMurphyTM1234

cmuellner, 2024-04-05:
> @TommyMurphyTM1234 I tried building it and recurrent the same problem. You are right, we also need a latest Newlib, change it to release 4.40 or into master.

Bump to newlib 4.4.0 is in #1451.

Ye-Jinhong, 2024-04-06:
@TommyMurphyTM1234 @pz9115 thanks for much. I finally get Zicond code!

> OK - so this worked for me:
> 
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain
> cd riscv-gnu-toolchain
> git clone https://github.com/bminor/newlib newlib-master
> git clone https://github.com/gcc-mirror/gcc gcc-master
> ./configure --prefix=`pwd`/installed-tools --disable-gdb --with-gcc-src=`pwd`/gcc-master --with-newlib-src=`pwd`/newlib-master --with-multilib-generator="rv64gc_zicond-lp64d--"
> make 2>&1 | tee build.log
> 
> cd installed-tools/bin
> ./riscv64-unknown-elf-gcc -print-multi-lib
> .;
> rv64imafdc_zicond_zicsr_zifencei/lp64d;@march=rv64imafdc_zicond_zicsr_zifencei@mabi=lp64d
> 
> # Note that ".;" refers to the default arch/abi = rv64imafdc/lp64d
> ./riscv64-unknown-elf-gcc -v
> Using built-in specs.
> COLLECT_GCC=./riscv64-unknown-elf-gcc
> COLLECT_LTO_WRAPPER=/home/user/Downloads/issue-1448/installed-tools/libexec/gcc/riscv64-unknown-elf/14.0.1/lto-wrapper
> Target: riscv64-unknown-elf
> Configured with: /home/user/Downloads/issue-1448/gcc-master/configure --target=riscv64-unknown-elf --prefix=/home/user/Downloads/issue-1448/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=g4c8b3600c48 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/Downloads/issue-1448/installed-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=/home/user/Downloads/issue-1448/gcc-master --enable-multilib --with-multilib-generator=rv64gc_zicond-lp64d-- --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
> Thread model: single
> Supported LTO compression algorithms: zlib
> gcc version 14.0.1 20240405 (experimental) (g4c8b3600c48)
> 
> # The default arch/abi is actually synonymous with rv64imafdc_zicsr/lp64d
> ./riscv64-unknown-elf-gcc -dumpspecs
> 
> ...
> 
> *multilib_defaults:
> march=rv64imafdc_zicsr mabi=lp64d
> 
> ...
> ```
> 
> And then a test:
> 
> * [How to compile with Zicond? #1448 (comment)](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1448#issuecomment-2037398902)
> * https://godbolt.org/z/1PMq58Khs
> 
> ```
> # test.c
> int foo (long a, long b)
> {
>   if (!a)
>     return 0;
>   else if (b)
>     return 1;
>   else
>     return 0;
> }
> ```
> 
> Using `rv64gc[_zicsr]/lp64d`:
> 
> ```
> ./riscv64-unknown-elf-gcc -O2 -c test.c
> 
> ./riscv64-unknown-elf-objdump -dS test.o
> 
> test.o:     file format elf64-littleriscv
> 
> Disassembly of section .text:
> 
> 0000000000000000 <foo>:
>    0:   c501                    beqz    a0,8 <.L3>
>    2:   00b03533                snez    a0,a1
>    6:   8082                    ret
> 
> 0000000000000008 <.L3>:
>    8:   4501                    li      a0,0
>    a:   8082                    ret
> ```
> 
> And now using `rv64gc_zicond/lp64d`:
> 
> ```
> ./riscv64-unknown-elf-gcc -march=rv64gc_zicond -O2 -c test.c
> 
> ./riscv64-unknown-elf-objdump -dS test.o
> 
> test.o:     file format elf64-littleriscv
> 
> Disassembly of section .text:
> 
> 0000000000000000 <foo>:
>    0:   00b035b3                snez    a1,a1
>    4:   0ea5d533                czero.eqz       a0,a1,a0
>    8:   8082                    ret
> ```
> 
> So - finally! - the original question has been addressed and I think that we can close this. 😄

TommyMurphyTM1234, 2024-04-06:
Note that `riscv-gnu-toolchain` has since been updated to use Newlib 4.4.0:

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1451

so this is not longer necessary when building with upstream work in progress GCC 14 sources:
> ```
> git clone https://github.com/bminor/newlib newlib-master
> ...
> ./configure ... --with-newlib-src=`pwd`/newlib-master --with-multilib-generator="rv64gc_zicond-lp64d--"
>```