riscv-collab/riscv-gnu-toolchain #1449: configure: error: cannot compute suffix of object files: cannot compile

rajsureshgeorge, 2024-04-03:
I configured using this ```./configure --prefix=/home/hdg/Office/Tools/newlibc/ --with-multilib-generator="rv64imafdb-lp64--;rv64imafdzba_zbb_zbc_zbs-lp64--;rv64imafdczba_zbb_zbc_zbs-lp64--;"```

while using ```make newlib```
```
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc -B/home/Office/Tools/RISCV/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv64imafdb_zicsr/lp64/newlib/ -isystem /home/Office/Tools/RISCV/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv64imafdb_zicsr/lp64/newlib/targ-include -isystem /home/Office/Tools/RISCV/riscv-gnu-toolchain/newlib/newlib/libc/include -B/home//Office/Tools/RISCV/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv64imafdb_zicsr/lp64/libgloss/riscv64 -L/home/Office/Tools/RISCV/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv64imafdb_zicsr/lp64/libgloss/libnosys -L/home/Office/Tools/RISCV/riscv-gnu-toolchain/newlib/libgloss/riscv64  -march=rv64imafdb_zicsr -mabi=lp64
checking whether we are cross compiling... yes
checking for suffix of object files... configure: error: in `/home/Office/Tools/RISCV/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/rv64imafdb_zicsr/lp64/newlib':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make[2]: *** [Makefile:8448: configure-target-newlib] Error 1
make[2]: Leaving directory '/home/Office/Tools/RISCV/riscv-gnu-toolchain/build-newlib'
make[1]: *** [Makefile:879: all] Error 2
make[1]: Leaving directory '/home/Office/Tools/RISCV/riscv-gnu-toolchain/build-newlib'
make: *** [Makefile:661: stamps/build-newlib] Error 2

```

Error while compiling Toolchain.

TommyMurphyTM1234, 2024-04-03:
> Please capture and attach the full build log - e.g.:
> 
> ```
> make 2>&1 | tee build.log
> ```

Never mind - I can reproduce it without the full log.

As far as I know, this error:
> ```
> configure: error: cannot compute suffix of object files: cannot compile
> ```
occurs when an invalid architecture string is used.

I don't think that `b` as an alias for `zba_zbb_zbs` it recognised by GCC 13.2 currently used by `riscv-gnu-toolchain`:

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/9a28c809a5d355874def06a414aa5f272fae564d/.gitmodules#L5-L8

and to use it you may need to use upstream GCC sources:

* https://www.mail-archive.com/gcc-patches@gcc.gnu.org/msg333693.html
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc gcc-master
./configure --prefix=/home/hdg/Office/Tools/newlibc/ --with-multilib-generator="rv64imafdb-lp64--;rv64imafdzba_zbb_zbc_zbs-lp64--;rv64imafdczba_zbb_zbc_zbs-lp64--;" --with-gcc-src=`pwd`/gcc-master
make newlib
```

TommyMurphyTM1234, 2024-04-03:
Building from upstream GCC sources gives a different error:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc gcc-master
./configure --prefix=/home/hdg/Office/Tools/newlibc/ --with-multilib-generator="rv64imafdb-lp64--;rv64imafdzba_zbb_zbc_zbs-lp64--;rv64imafdczba_zbb_zbc_zbs-lp64--;" --with-gcc-src=`pwd`/gcc-master
make newlib

...

/home/user/Downloads/issue-1449/build-gcc-newlib-stage1/./gcc/xgcc -B/home/user/Downloads/issue-1449/build-gcc-newlib-stage1/./gcc/  -xc -nostdinc /dev/null -S -o /dev/null -fself-test=/home/user/Downloads/issue-1449/gcc-master/gcc/testsuite/selftests
xgcc: error: '-march=rv64imafdb_zicsr': extension 'b' is unsupported standard single letter extension
make[2]: *** [/home/user/Downloads/issue-1449/gcc-master/gcc/c/Make-lang.in:153: s-selftest-c] Error 1
rm gfdl.pod gcc.pod gcov-dump.pod gcov-tool.pod fsf-funding.pod gpl.pod cpp.pod gcov.pod lto-dump.pod
make[2]: Leaving directory '/home/user/Downloads/issue-1449/build-gcc-newlib-stage1/gcc'
make[1]: *** [Makefile:4689: all-gcc] Error 2
make[1]: Leaving directory '/home/user/Downloads/issue-1449/build-gcc-newlib-stage1'
make: *** [Makefile:627: stamps/build-gcc-newlib-stage1] Error 2
```

TommyMurphyTM1234, 2024-04-03:
FWIW this works:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=/home/hdg/Office/Tools/newlibc/ --with-multilib-generator="rv64imafdzba_zbb_zbs-lp64--;rv64imafdzba_zbb_zbc_zbs-lp64--;rv64imafdczba_zbb_zbc_zbs-lp64--;"
make newlib

...

./installed-tools/bin/riscv64-unknown-elf-gcc -print-multi-lib
.;
rv64imafd_zicsr_zba_zbb_zbs/lp64;@march=rv64imafd_zicsr_zba_zbb_zbs@mabi=lp64
rv64imafd_zicsr_zba_zbb_zbc_zbs/lp64;@march=rv64imafd_zicsr_zba_zbb_zbc_zbs@mabi=lp64
rv64imafdc_zicsr_zba_zbb_zbc_zbs/lp64;@march=rv64imafdc_zicsr_zba_zbb_zbc_zbs@mabi=lp64

./installed-tools/bin/riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./installed-tools/bin/riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/Downloads/issue-1449/installed-tools/libexec/gcc/riscv64-unknown-elf/13.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/user/Downloads/issue-1449/gcc/configure --target=riscv64-unknown-elf --prefix=/home/user/Downloads/issue-1449/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=gc891d8dc23e --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/Downloads/issue-1449/installed-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --enable-multilib --with-multilib-generator='rv64imafdzba_zbb_zbs-lp64--;rv64imafdzba_zbb_zbc_zbs-lp64--;rv64imafdczba_zbb_zbc_zbs-lp64--;' --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 13.2.0 (gc891d8dc23e)
```

TommyMurphyTM1234, 2024-04-03:
For some reason, when I try the same approach using upstream GCC sources it fails:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc gcc-master
./configure --prefix=/home/hdg/Office/Tools/newlibc/ --with-multilib-generator="rv64imafdzba_zbb_zbs-lp64--;rv64imafdzba_zbb_zbc_zbs-lp64--;rv64imafdczba_zbb_zbc_zbs-lp64--;" --with-gcc-src=`pwd`/gcc-master
make newlib

...

  CC       libm/complex/libm_a-ccoshl.o
/home/user/Downloads/issue-1449/newlib/newlib/libm/complex/ccoshl.c: In function 'ccoshl':
/home/user/Downloads/issue-1449/newlib/newlib/libm/complex/ccoshl.c:43:13: error: implicit declaration of function 'coshl'; did you mean 'coshf'? [-Wimplicit-function-declaration]
   43 |         w = coshl(x) * cosl(y) + (sinhl(x) * sinl(y)) * I;
      |             ^~~~~
      |             coshf
/home/user/Downloads/issue-1449/newlib/newlib/libm/complex/ccoshl.c:43:24: error: implicit declaration of function 'cosl'; did you mean 'cosf'? [-Wimplicit-function-declaration]
   43 |         w = coshl(x) * cosl(y) + (sinhl(x) * sinl(y)) * I;
      |                        ^~~~
      |                        cosf
/home/user/Downloads/issue-1449/newlib/newlib/libm/complex/ccoshl.c:43:35: error: implicit declaration of function 'sinhl'; did you mean 'sinhf'? [-Wimplicit-function-declaration]
   43 |         w = coshl(x) * cosl(y) + (sinhl(x) * sinl(y)) * I;
      |                                   ^~~~~
      |                                   sinhf
/home/user/Downloads/issue-1449/newlib/newlib/libm/complex/ccoshl.c:43:46: error: implicit declaration of function 'sinl'; did you mean 'sinf'? [-Wimplicit-function-declaration]
   43 |         w = coshl(x) * cosl(y) + (sinhl(x) * sinl(y)) * I;
      |                                              ^~~~
      |                                              sinf
make[4]: *** [Makefile:41572: libm/complex/libm_a-ccoshl.o] Error 1
make[4]: Leaving directory '/home/user/Downloads/issue-1449/build-newlib/riscv64-unknown-elf/newlib'
make[3]: *** [Makefile:5038: all] Error 2
make[3]: Leaving directory '/home/user/Downloads/issue-1449/build-newlib/riscv64-unknown-elf/newlib'
make[2]: *** [Makefile:8492: all-target-newlib] Error 2
make[2]: Leaving directory '/home/user/Downloads/issue-1449/build-newlib'
make[1]: *** [Makefile:879: all] Error 2
make[1]: Leaving directory '/home/user/Downloads/issue-1449/build-newlib'
make: *** [Makefile:661: stamps/build-newlib] Error 2
```
My conclusions are:

1. You should stick to building from this `riscv-gnu-toolchain` repo.
2. But you should avoid the `b` extension specifier in the architecture string because it doesn't work - use `zba_zbb_zbs` instead.
3. At some point in the futhre the upstream sources will build properly and support the `b` extension specifier as synonymous with `zba_zbb_zbs`.

rajsureshgeorge, 2024-04-04:
Thanks @TommyMurphyTM1234 for the response.Could you provide some details on how to generate bit manipulation instructions? I'm trying to compile a program, but I'm not seeing many bit manipulation instructions in the resulting binary. Is there a way to generate more bit manipulation instructions during compilation? For example, could adding extra multilibs or changing the ABI help? I've attached 2 programs for reference 
[pos.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/14861728/pos.txt) 
[polynomial.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/14861731/polynomial.txt)
_Both didn't generate much bit manipulation instruction._

rajsureshgeorge, 2024-04-04:
My configuration was this : ```./configure --prefix=/home/user/ToolChain/MULTILIB/ --enable-multilib --with-cmodel=medany``` then while ```make linux``` 

I got this error:
```
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv32-unknown-linux-gnu
checking for riscv32-unknown-linux-gnu-gcc... riscv64-unknown-linux-gnu-gcc -march=rv32imac -mabi=ilp32
checking for suffix of object files... configure: error: in `/home/user/ToolChain/riscv-gnu-toolchain/build-glibc-linux-rv32imac-ilp32':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make: *** [Makefile:360: stamps/build-glibc-linux-rv32imac-ilp32] Error 1
```

TommyMurphyTM1234, 2024-04-04:
> My configuration was this : `./configure --prefix=/home/user/ToolChain/MULTILIB/ --enable-multilib --with-cmodel=medany` then while `make linux`
>
> I got this error:
>
> ```
> checking build system type... [...]

Did you do a clean build?
```
make distclean
rm -rf /home/user/ToolChain/MULTILIB/
make Linux
```

TommyMurphyTM1234, 2024-04-04:
> Thanks @TommyMurphyTM1234 for the response.Could you provide some details on how to generate bit manipulation instructions?

Sorry, I don't know. 

> For example, could adding extra multilibs or changing the ABI help? 

I don't think so. I think the important thing is that you pass the appropriate `arch/abi` for your needs/target RISC-V implementation at compile time. E.g. `-march=rv64gc_zicsr_zba_zbb_zbc_zbs -mabi=lp64d`. Unless your code is completely standalone you will also need the toolchain to provide the startup code/multilib for the selected `arch/abi`.

You could probably search for information about the GCC patches that added bit-manipulation extension support to see if there's any information there about what kind of C code patterns are best suited to causing these instructions to be generated by the compiler.

TommyMurphyTM1234, 2024-04-04:
> You could probably search for information about the GCC patches that added bit-manipulation extension support to see if there's any information there about what kind of C code patterns are best suited to causing these instructions to be generated by the compiler.

Perhaps there is something useful in this test suite?

* https://github.com/riscv-software-src/riscv-tests/tree/master/isa

But I think that most or all of those tests are assembly language based?

TommyMurphyTM1234, 2024-04-07:
@rajsureshgeorge, are you still having issues with any of this or can the issue be closed now? 

If you're still getting the `cannot compute suffix of object files` error then please make sure that you are doing a clean build and that your environment isn't picking up another RISC-V toolchain etc. 

If the issue persists then please provide full details on how to reproduce the problem.

rajsureshgeorge, 2024-04-08:
@TommyMurphyTM1234  I have effectively constructed the tool-chain without encountering any errors by adhering to the provided instructions. We can now consider closing it for the time being.

closed 2024-04-08 by rajsureshgeorge