riscv-collab/riscv-gnu-toolchain #1450: Cannot access submodules

kper, 2024-04-05:
~~~
 => CACHED [ 5/20] RUN git clone https://github.com/riscv-collab/riscv-gnu-toolchain                                                                                                                     0.0s
 => CACHED [ 6/20] WORKDIR riscv-gnu-toolchain                                                                                                                                                           0.0s
 => ERROR [ 7/20] RUN git submodule update --init --depth=1 -- gcc                                                                                                                                      49.9s
------
 > [ 7/20] RUN git submodule update --init --depth=1 -- gcc:
0.535 Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
0.541 Cloning into '/root/riscv-gnu-toolchain/gcc'...
49.56 error: Server does not allow request for unadvertised object c891d8dc23e1a46ad9f3e757d09e57b500d40044
49.57 Fetched in submodule path 'gcc', but it did not contain c891d8dc23e1a46ad9f3e757d09e57b500d40044. Direct fetching of that commit failed.
------
Dockerfile:13
--------------------
  11 |     RUN git clone https://github.com/riscv-collab/riscv-gnu-toolchain
  12 |     WORKDIR riscv-gnu-toolchain
  13 | >>> RUN git submodule update --init --depth=1 -- gcc
  14 |     RUN git submodule update --init --depth=1 -- glibc
  15 |     RUN git submodule update --init --depth=1 -- binutils
~~~

Unfortunately, this script has worked before. Also when I click on gcc's reference in the browser, I get a 404 error.

TommyMurphyTM1234, 2024-04-05:
> Unfortunately, this script has worked before. 

What script are you referring to?
This seems to be some Docker script?
This repo (`riscv-gnu-toolchain`) doesn't use/provide any Docker scripts as far as I know.

> Also when I click on gcc's reference in the browser, I get a 404 error.

What "reference"?
Do you mean `https://gcc.gnu.org/git/gcc.git`?

Please clarify exactly what you are doing here that results in this problem - detail all commands that you are running, and any other relevant context, that results in this error.

For what it's worth I did this just now and had no problem in case that's of any relevance:
```
git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain issue-1450
Cloning into 'issue-1450'...
remote: Enumerating objects: 9157, done.
remote: Counting objects: 100% (9157/9157), done.
remote: Compressing objects: 100% (3778/3778), done.
remote: Total 9157 (delta 4922), reused 8839 (delta 4864), pack-reused 0
Receiving objects: 100% (9157/9157), 5.36 MiB | 3.43 MiB/s, done.
Resolving deltas: 100% (4922/4922), done.
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Submodule 'llvm' (https://github.com/llvm/llvm-project.git) registered for path 'llvm'
Submodule 'musl' (https://git.musl-libc.org/git/musl) registered for path 'musl'
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Submodule 'pk' (https://github.com/riscv-software-src/riscv-pk.git) registered for path 'pk'
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Cloning into '/home/user/Downloads/issue-1450/binutils'...
remote: Enumerating objects: 1219507, done.
remote: Counting objects: 100% (44589/44589), done.
remote: Compressing objects: 100% (39114/39114), done.
remote: Total 1219507 (delta 8069), reused 7750 (delta 5425), pack-reused 1174918
Receiving objects: 100% (1219507/1219507), 454.73 MiB | 7.44 MiB/s, done.
Resolving deltas: 100% (1013178/1013178), done.
Cloning into '/home/user/Downloads/issue-1450/dejagnu'...
remote: Counting objects: 9058, done.
remote: Compressing objects: 100% (2040/2040), done.
remote: Total 9058 (delta 6922), reused 9013 (delta 6890)
Receiving objects: 100% (9058/9058), 4.70 MiB | 5.03 MiB/s, done.
Resolving deltas: 100% (6922/6922), done.
Cloning into '/home/user/Downloads/issue-1450/gcc'...
remote: Enumerating objects: 3023828, done.
remote: Counting objects: 100% (224219/224219), done.
remote: Compressing objects: 100% (17856/17856), done.
remote: Total 3023828 (delta 217018), reused 208105 (delta 206287), pack-reused 2799609
Receiving objects: 100% (3023828/3023828), 1.15 GiB | 8.44 MiB/s, done.
Resolving deltas: 100% (2487484/2487484), done.
Cloning into '/home/user/Downloads/issue-1450/gdb'...
remote: Enumerating objects: 1219507, done.
remote: Counting objects: 100% (44589/44589), done.
remote: Compressing objects: 100% (39114/39114), done.
remote: Total 1219507 (delta 8071), reused 7750 (delta 5425), pack-reused 1174918
Receiving objects: 100% (1219507/1219507), 454.77 MiB | 7.49 MiB/s, done.
Resolving deltas: 100% (1013180/1013180), done.
Cloning into '/home/user/Downloads/issue-1450/glibc'...
remote: Enumerating objects: 719881, done.
remote: Counting objects: 100% (7751/7751), done.
remote: Compressing objects: 100% (2724/2724), done.
remote: Total 719881 (delta 6310), reused 5313 (delta 4908), pack-reused 712130
Receiving objects: 100% (719881/719881), 222.69 MiB | 6.21 MiB/s, done.
Resolving deltas: 100% (618661/618661), done.
Cloning into '/home/user/Downloads/issue-1450/llvm'...
remote: Enumerating objects: 5879411, done.
[... 327 lines cut ...]
Submodule path 'qemu/roms/qemu-palcode': checked out '99d9b4dcf27d7fbcbadab71bdc88ef6531baf6bf'
Submodule path 'qemu/roms/seabios': checked out 'a6ed6b701f0a57db0569ab98b0661c12a6ec3ff8'
Submodule path 'qemu/roms/seabios-hppa': checked out 'e4eac85880e8677f96d8b9e94de9f2eec9c0751f'
Submodule path 'qemu/roms/skiboot': checked out '24a7eb35966d93455520bc2debdd7954314b638b'
Submodule path 'qemu/roms/u-boot': checked out '840658b093976390e9537724f802281c9c8439f5'
Submodule path 'qemu/roms/u-boot-sam460ex': checked out '60b3916f33e617a815973c5a6df77055b2e3a588'
Submodule path 'qemu/roms/vbootrom': checked out '0c37a43527f0ee2b9584e7fb2fdc805e902635ac'
Submodule path 'qemu/tests/lcitool/libvirt-ci': checked out '77c800186f34b21be7660750577cc5582a914deb'
Submodule path 'spike': checked out 'd1efcdffffee57bab0fdbd2b377c6132b37556fd'
```
And if I go to `https://gcc.gnu.org/git/gcc.git` in a web browser I get this:

[image]

Finally:
> ```
> 49.56 error: Server does not allow request for unadvertised object c891d8dc23e1a46ad9f3e757d09e57b500d40044
> 49.57 Fetched in submodule path 'gcc', but it did not contain c891d8dc23e1a46ad9f3e757d09e57b500d40044. Direct fetching of that commit failed.
> ```

`c891d8dc23e1a46ad9f3e757d09e57b500d40044` is the commit hash for the `releases/gcc-13.2.0` tag:

* https://gcc.gnu.org/git/?p=gcc.git;a=commit;h=c891d8dc23e1a46ad9f3e757d09e57b500d40044

Maybe you are experiencing some local connectivity issues or happened to hit some problem with the upstream repos?

kper, 2024-04-05:
Ah sry to have bother you. I was working on a custom script which packages the toolchain into a docker container so I don't have to have the dependencies on my system.

I have run 
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain 
> git submodule update --init --depth=1 -- gcc

This has produced the mentioned exception. But it appears that it was only temporarily.

Regarding:

> Also when I click on gcc's reference in the browser, I get a 404 error.

I clicked the gcc folder link on the github UI which leads to the mentioned 404 error. But now, my impression is that it is not supposed to work because it points to a wrong URL.

[image]

Anyway, thank you for your time! I am closing this issue.

closed 2024-04-05 by kper

TommyMurphyTM1234, 2024-04-05:
> Ah sry to have bother you. I was working on a custom script which packages the toolchain into a docker container so I don't have to have the dependencies on my system.

Maybe @ilg-ul's xPack Docker based build scripts are of interest to you and might save you some work?

* https://xpack.github.io/dev-tools/riscv-none-elf-gcc/
  * https://xpack.github.io/dev-tools/riscv-none-elf-gcc/#build-details
    * https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/blob/xpack/README-MAINTAINER.md

kper, 2024-04-05:
Thank you
I wasn't aware that this project exists