riscv-collab/riscv-gnu-toolchain #1453: Can't clone llvm in riscv-gnu-toolchain

shane1604, 2024-04-06:
Does anyone know this problem? I try to follow the ins of risc-v. But somehow I can't clone this llvm. I tried many times, many solution but I get the same error. Pls help. No this file I can't `make`
[image]

pz9115, 2024-04-06:
It seems that you meet some network problems, please try to clone it again.

TommyMurphyTM1234, 2024-04-06:
Worked fine for me right now - both as part of a clone of the `riscv-gnu-toolchain` repo and standalone.
I even did them at the same time.
```
git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain issue-1453
Cloning into 'issue-1453'...
remote: Enumerating objects: 9163, done.
remote: Counting objects: 100% (9163/9163), done.
remote: Compressing objects: 100% (3784/3784), done.
remote: Total 9163 (delta 4926), reused 8839 (delta 4864), pack-reused 0
Receiving objects: 100% (9163/9163), 5.36 MiB | 2.70 MiB/s, done.
Resolving deltas: 100% (4926/4926), done.
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Submodule 'llvm' (https://github.com/llvm/llvm-project.git) registered for path 'llvm'
Submodule 'musl' (https://git.musl-libc.org/git/musl) registered for path 'musl'
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Submodule 'pk' (https://github.com/riscv-software-src/riscv-pk.git) registered for path 'pk'
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Cloning into '/home/user/Downloads/issue-1453/binutils'...
remote: Enumerating objects: 1219566, done.
remote: Counting objects: 100% (44648/44648), done.
remote: Compressing objects: 100% (39171/39171), done.
remote: Total 1219566 (delta 8114), reused 7739 (delta 5425), pack-reused 1174918
Receiving objects: 100% (1219566/1219566), 454.83 MiB | 9.46 MiB/s, done.
Resolving deltas: 100% (1013223/1013223), done.
Cloning into '/home/user/Downloads/issue-1453/dejagnu'...
remote: Counting objects: 9058, done.
remote: Compressing objects: 100% (2040/2040), done.
remote: Total 9058 (delta 6922), reused 9013 (delta 6890)
Receiving objects: 100% (9058/9058), 4.70 MiB | 5.04 MiB/s, done.
Resolving deltas: 100% (6922/6922), done.
Cloning into '/home/user/Downloads/issue-1453/gcc'...
remote: Enumerating objects: 3023970, done.
remote: Counting objects: 100% (224361/224361), done.
remote: Compressing objects: 100% (17998/17998), done.
remote: Total 3023970 (delta 217114), reused 208104 (delta 206287), pack-reused 2799609
Receiving objects: 100% (3023970/3023970), 1.16 GiB | 10.72 MiB/s, done.
Resolving deltas: 100% (2487580/2487580), done.
Cloning into '/home/user/Downloads/issue-1453/gdb'...
remote: Enumerating objects: 1219566, done.
remote: Counting objects: 100% (44648/44648), done.
remote: Compressing objects: 100% (39171/39171), done.
remote: Total 1219566 (delta 8110), reused 7738 (delta 5425), pack-reused 1174918
Receiving objects: 100% (1219566/1219566), 454.85 MiB | 9.44 MiB/s, done.
Resolving deltas: 100% (1013219/1013219), done.
Cloning into '/home/user/Downloads/issue-1453/glibc'...
remote: Enumerating objects: 719914, done.
remote: Counting objects: 100% (7784/7784), done.
remote: Compressing objects: 100% (2754/2754), done.
remote: Total 719914 (delta 6332), reused 5308 (delta 4908), pack-reused 712130
Receiving objects: 100% (719914/719914), 222.70 MiB | 7.04 MiB/s, done.
Resolving deltas: 100% (618683/618683), done.
Cloning into '/home/user/Downloads/issue-1453/llvm'...
remote: Enumerating objects: 5881001, done.
remote: Counting objects: 100% (3531/3531), done.
remote: Compressing objects: 100% (1590/1590), done.
remote: Total 5881001 (delta 2150), reused 2734 (delta 1888), pack-reused 5877470
Receiving objects: 100% (5881001/5881001), 2.25 GiB | 7.01 MiB/s, done.
Resolving deltas: 100% (4801463/4801463), done.gitResolving deltas:  71% (3450296/4801463)
Cloning into '/home/user/Downloads/issue-1453/musl'...
...
```
```
git clone https://github.com/llvm/llvm-project.git
Cloning into 'llvm-project'...
remote: Enumerating objects: 5881001, done.
remote: Counting objects: 100% (3988/3988), done.
remote: Compressing objects: 100% (2048/2048), done.
remote: Total 5881001 (delta 2560), reused 2666 (delta 1887), pack-reused 5877013
Receiving objects: 100% (5881001/5881001), 2.26 GiB | 6.93 MiB/s, done.
Resolving deltas: 100% (4797092/4797092), done.
Updating files: 100% (142107/142107), done.
```
As @pz9115 says, it's most likely a network connectivity issue between your machine and the GitHub repo.
Maybe try another computer and/or network connection if possible.

shane1604, 2024-04-06:
[image]
somehow I still get error. I tried to change my connection to other network. :<

TommyMurphyTM1234, 2024-04-06:
Unfortunately there's nothing that anybody here can do. 
It's not a problem with the `riscv-gnu-toolchain` or LLVM repositories since they're working fine for others.
You'll just have to try and diagnose the problem with your network administrators/providers.

In the meantime maybe you can download a suitable tarball of the sources and use `configure ... --with-llvm-src=...` until the issue is resolved?

* https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#use-source-tree-other-than-riscv-gnu-toolchain
* https://releases.llvm.org/download.html

Closing the issue since it's not a valid `riscv-gni-toolchain` problem.

closed 2024-04-06 by TommyMurphyTM1234

TommyMurphyTM1234, 2024-04-06:
Maybe some of the suggestions here could be useful for diagnosing the problem?

* https://stackoverflow.com/questions/66366582/github-unexpected-disconnect-while-reading-sideband-packet

shane1604, 2024-04-06:
thanks @TommyMurphyTM1234 for your support. I tried every method on internet from postBuff, uninstall,... and even the  post you referred to. It didn't work for me. But in the end, I was able to clone by using mobile data (4G) - (other network like u said). 
I'm quite new to this, so there's still a lot for me to learn. Thank you for your enthusiastic assistance. Wishing you lot of joy =))...

TommyMurphyTM1234, 2024-04-06:
> thanks @TommyMurphyTM1234 for your support. I tried every method on internet from postBuff, uninstall,... and even the post you referred to. It didn't work for me. But in the end, I was able to clone by using mobile data (4G) - (other network like u said). 

You should maybe ask your original network administrator/provider if they know of any issue preventing successful cloning of the LLVM repo.