riscv-collab/riscv-gnu-toolchain #1465: error while compiling with latest gcc

RohanAhmed1, 2024-04-24:
Hi,

I am trying to build the riscv-gnu-toolchain with **gcc at master branch**. I knew, this is not released yet, but I am just checking that, when the gcc14 will release, I will not face any problem. is there any way, we can try that

this error is coming.
[image]

I am following these steps;
> make distclean
> ./configure -with-gcc-src=[gcc-at-master] --prefix=[obj-folder] --with-cmodel=medany
> make

Thanks

TommyMurphyTM1234, 2024-04-24:
Please provide your full commands and build log.
E.g. all relevant `git clone`, `configure`, `make` commands and build log from something like `make 2>&1 | tee build.log`.
Logs and transcripts are preferable to screenshots as not everybody can view the latter easily and they can't be easily searched/analysed.

TommyMurphyTM1234, 2024-04-24:
This worked fine for me:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc gcc-master
./configure --prefix=`pwd`/installed-tools --with-gcc-src=`pwd`/gcc-master --with-cmodel=medany
make 2>&1 | tee build.log
```
* `build.log.zip`: [build.log.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/15097190/build.log.zip)

So I presume that the issue is something in your local setup?

RohanAhmed1, 2024-04-25:
Thanks @TommyMurphyTM1234 , let me try from scratch

RohanAhmed1, 2024-04-29:
Actually, I tried it with toolchain with this gcc that you tried;
git clone https://github.com/gcc-mirror/gcc
and it got same error. 

Now, I am cloning toolchain from scratch(without any change) and will try with this gcc. lets see, I will update asap

TommyMurphyTM1234, 2024-04-29:
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain
> cd riscv-gnu-toolchain
> git clone https://github.com/gcc-mirror/gcc gcc-master
> ./configure --prefix=`pwd`/installed-tools --with-gcc-src=`pwd`/gcc-master --with-cmodel=medany
> make 2>&1 | tee build.log

As described above, this approach works for me and does not result in any error:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc gcc-master
./configure --prefix=`pwd`/installed-tools --with-gcc-src=`pwd`/gcc-master --with-cmodel=medany
make 2>&1 | tee build.log
```
If you get an error with this approach then it is almost certainly due to some local issue with your host, environment, host toolchain etc.

fwsGonzo, 2024-04-29:
I also got the cosl and sinl errors when building libm.

TommyMurphyTM1234, 2024-04-29:
> I also got the cosl and sinl errors when building libm.

You'll need to provide more details of all of the commands used to build the toolchain including any `git clone`, `configure`, `make` commands used and details of the build host platform used. 

As I said above, these commands successfully build the toolchain without any errors on two different Linux distros (Ubuntu 22.04 and Zorin OS 17.1 Core).
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc gcc-master
./configure --prefix=`pwd`/installed-tools --with-gcc-src=`pwd`/gcc-master --with-cmodel=medany
make 2>&1 | tee build.log
```

fwsGonzo, 2024-04-29:
I solved it by stubbing out the complex number implementation of newlib, and also fixing quite a few missing casts in the system call wrappers in libgloss. I've attached the diff.

https://gist.github.com/fwsGonzo/b9c54e49fef2337896ed7e2e495214fe

```
-- The C compiler identification is GNU 15.0.0
-- The CXX compiler identification is GNU 15.0.0
```
It builds, links and runs fine.

TommyMurphyTM1234, 2024-04-29:
> Just because it works for you doesn't mean it works for me. I am building with Clang 18.x. It might be that Clang is stricter than the Ubuntu 22.04 default system compiler.

That why I asked for the additional clarifying information which you haven't provided.

RohanAhmed1, 2024-05-02:
Thanks to all of you, I tried the both ways to build the toolchain and it 

First, I tried to comment out code of **newlib** files, just like @fwsGonzo, and it build successfully, so, it confirms that, the problem is in **newlib folder**.

Secondly, I start from the scratch from clone to build as told by @TommyMurphyTM1234.

closed 2024-05-02 by RohanAhmed1

TommyMurphyTM1234, 2024-05-02:
Thanks for the update @RohanAhmed1. Glad to hear that you made progress.

> I am building with Clang 18.x. It might be that Clang is stricter than the Ubuntu 22.04 default system compiler.

If there are issues with building Newlib using LLVM/Clang - which should be possible to reproduce by building Newlib "standalone" outside of the context of the `riscv-gnu-toolchain` build process - then it might make sense to report them upstream. 

* https://sourceware.org/newlib/