riscv-collab/riscv-gnu-toolchain #1467: Avoid branch instruction substitutions

bautixter, 2024-04-25:
Hello, I'm trying to debug a Risc-V decoder implementation and I'm using the `riscv32-unknown-linux-gnu-as` assembler to generate the binaries for a test program. I also use `riscv32-unknown-linux-gnu-objdump` to visualize the binaries along with its respective instructions.

Now, my problem is that when I try the following sample code:

```verilog
    beq        x0,   x1,  128
    beq        x1,   x2,  64
    beq        x2,   x3,  32
    beq        x4,   x5,  16
    beq        x8,   x9,  8
    beq       x16,   x2,  4
```

I get the following objdump:

```verilog
   0:	00101463          	bne	zero,ra,8 <_start+0x8>
   4:	07c0006f          	j	80 <_start+0x80>
   8:	00209463          	bne	ra,sp,10 <_start+0x10>
   c:	0340006f          	j	40 <_start+0x40>
  10:	00311463          	bne	sp,gp,18 <_start+0x18>
  14:	00c0006f          	j	20 <_start+0x20>
  18:	00521463          	bne	tp,t0,20 <_start+0x20>
  1c:	ff5ff06f          	        j	10 <_start+0x10>
  20:	00941463          	bne	s0,s1,28 <_start+0x28>
  24:	fe5ff06f          	        j	8 <_start+0x8>
  28:	00281463          	bne	a6,sp,30 <_start+0x30>
  2c:	fd9ff06f          	        j	4 <_start+0x4>
```

It seems the assembler is doing this translation from `beq` to `bne + j` that I really don't need to be performed, as what I want to evaluate is just if  the `beq` data is being correctly decoded. Also this does not happen when we use a tag instead of an immediate value.

I have tried several flag combinations without success, so I was wondering if anyone knew the right flag to avoid this kind of behavior.

Thank you.

TommyMurphyTM1234, 2024-04-25:
This may be relevant? 

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/143#issuecomment-219193419

bautixter, 2024-04-25:
Thank you very much Tommy, in your second link Peter Cordes commented the following:

> As a workaround, you could write the destination in the source relative to the instruction's location, right? Like beq x,x, . +
> 0x48-0x16. Or relative to a label at the top of your code. Like start: ... beq x,x,  start + 0x48. If that code from from absolute 
> address zero, then branch target will be absolute address 0x48

I've tried this solution and it solves my needs. I'm closing the issue, thanks a lot. 👍

closed 2024-04-25 by bautixter

cmuellner, 2024-04-25:
It seems you want to write code that branches to address `128` if `x0` is equal to `x1`.

The `beq` instruction accepts a 12-bit signed immediate offset (relative to the branch instruction) that will be multiplied by 2. Thus, you have a reach of +-4 kiB.

Now, the assembler has the issue that it does not know where your code will be loaded to, because it is not linked yet.
Therefore, it cannot emit the right instruction but emits an instruction with a relocation. Looking into the relocations (`objdump -d -r`) gives the necessary insights:

```
0000000000000000 <.text>:                        
   0:   00101463                bne     zero,ra,8 <.text+0x8>
   4:   0000006f                j       4 <.text+0x4>                                                     
                        4: R_RISCV_JAL  *ABS*+0x80
   8:   00209463                bne     ra,sp,10 <.text+0x10>
   c:   0000006f                j       c <.text+0xc>
                        c: R_RISCV_JAL  *ABS*+0x40
[...]
```

Here, we can see that there are relocation entries for the jump targets.
The linker will resolve these relocations later on.

In case my assumption about your branch target is wrong, and you really want to jump to targets a few bytes away from your branch instruction, you may want to define labels and branch to them:
```
base:
        beq        x0,   x1,  L128

.skip 128
L128:
        nop
```

==>
```
0000000000000000 <base>:
   0:   08100263                beq     zero,ra,84 <L128>
                        0: R_RISCV_BRANCH       L128
        ...

0000000000000084 <L128>:
  84:   0001 
```

bautixter, 2024-04-25:
Answering to both your questions, this is indeed some synthetic test with zero real applications whatsoever. I'm going through all the encoding formats of the riscv specs and giving significant values to the different arguments, in an attempt to validate our decoder behavior and search for bugs.