riscv-collab/riscv-gnu-toolchain #1469: question about adding "mod" instruction in toolchain

1781559198, 2024-04-29:
Hi dear all
I have added the corresponding shell commands to riscv-gdb as well, and followed the four steps at[
https://nitish2112.github.io/post/adding-instruction-riscv/]()

1 add instruction to riscv-opcodes/opcodes
```
mod     rd rs1 rs2 31..25=1  14..12=0 6..2=0x1A 1..0=3
```
2 run the command:
```
cat opcodes-pseudo opcodes opcodes-rvc opcodes-rvc-pseudo opcodes-custom | ./parse-opcodes -c > ~/temp.h
```
3 add instruction to riscv-gnu-toolchain/gdb/opcodes/riscv-opc.h
```
#define MATCH_MOD 0x200006b                                                    
#define MASK_MOD 0xfe00707f
```
4 add instruction to riscv-gnu-toolchain/gdb/opcodes/riscv-opc.c
```
{"mod",       "I",   "d,s,t",  MATCH_MOD, MASK_MOD, match_opcode, 0 }
```
[image]
[image]
[image]

I followed the tutorial step by step.but when i compile the riscv-gnu-toolchain again and instructions "riscv64-unknown-elf-gcc mod.c -o mod" again,i have a mistake.

[image]

TommyMurphyTM1234, 2024-04-29:
That tutorial is quite old and may not be up to date with the latest repo contents. I see at least one part that is definitely wrong (and mentioned as such in the comments by somebody else) albeit not central to the exercise in question:
>```
> $ export RISCV=/path/to/install/riscv/toolchain
> $ ./build.sh
>``
In fact, building the toolchain at all before making the changes to add a new instruction seems pointless and may be undesirable since after doing this you'll probably need to do `make distclean` anyway in order to ensure that the toolchain is built properly so that it integrates all of the custom instruction changes.

So, I would suggest two things to try...

1. After adding the custom instruction changes do a `make distclean` before building the toolchain with support for the `mod` custom instruction to make sure that those changes get built in.
2. If that doesn't help then maybe try this variant of the guide which seems more up to date: https://pcotret.gitlab.io/riscv-custom/

When I get a chance later I'll try to run through these guides myself to see how it goes for me.

TommyMurphyTM1234, 2024-04-29:
I tried to follow the instructions here:

* https://nitish2112.github.io/post/adding-instruction-riscv/

I skipped this as explained in my previous post:
```
export RISCV=/path/to/install/riscv/toolchain
./build.sh
```
But immediately I hit a problem:

* https://nitish2112.github.io/post/adding-instruction-riscv/#:~:text=Open%20the%20file%20riscv%2Dopcodes/opcodes%2C

> Open the file riscv-opcodes/opcodes

There is no `riscv-opcodes/opcodes` in the current repo - both of the following searches return nothing:
```
find . -type f -name opcodes
find . -type d -name riscv-opcodes
```
Did they mean `opcodes.h`? But those files don't look relevant here as they are for the QEMU and Spike simulators and not the toolchain proper:
```
find . -name opcodes.h
./qemu/target/hexagon/opcodes.h
./qemu/roms/u-boot-sam460ex/arch/nios2/include/asm/opcodes.h
./qemu/roms/u-boot/arch/arm/include/asm/opcodes.h
./qemu/roms/u-boot/arch/nios2/include/asm/opcodes.h
./spike/riscv/opcodes.h
```
Or do they mean one of these files?
```
find . -type f -name *opcodes*
./qemu/target/tricore/tricore-opcodes.h
./qemu/target/hexagon/gen_opcodes_def.py
./qemu/target/hexagon/opcodes.h
./qemu/target/hexagon/opcodes.c
./qemu/roms/u-boot-sam460ex/arch/nios2/include/asm/opcodes.h
./qemu/roms/u-boot/arch/arm/include/asm/opcodes-sec.h
./qemu/roms/u-boot/arch/arm/include/asm/opcodes.h
./qemu/roms/u-boot/arch/arm/include/asm/opcodes-virt.h
./qemu/roms/u-boot/arch/nios2/include/asm/opcodes.h
./llvm/lld/test/MachO/bind-opcodes.s
./llvm/lld/test/MachO/rebase-opcodes.s
./llvm/mlir/utils/spirv/define_opcodes.sh
./llvm/llvm/test/Transforms/MetaRenamer/opcodes.ll
./llvm/llvm/test/tools/llvm-reduce/reduce-opcodes.ll
./llvm/llvm/test/tools/llvm-reduce/reduce-opcodes-call.ll
./llvm/llvm/test/tools/llvm-readobj/COFF/arm64-unwind-opcodes.s
./llvm/llvm/test/tools/llvm-readobj/COFF/arm-unwind-opcodes.s
./llvm/llvm/test/CodeGen/AMDGPU/lower-term-opcodes.mir
./llvm/llvm/test/CodeGen/ARM/Windows/wineh-opcodes.ll
./llvm/libc/include/llvm-libc-types/test_rpc_opcodes_t.h
./llvm/libc/include/llvm-libc-types/rpc_opcodes_t.h
./spike/riscv/opcodes.h
./gdb/sim/testsuite/bfin/se_allopcodes.h
./gdb/sim/testsuite/bfin/se_all32bitopcodes.S
./gdb/sim/testsuite/bfin/se_all64bitg1opcodes.S
./gdb/sim/testsuite/bfin/se_all64bitg0opcodes.S
./gdb/sim/testsuite/bfin/se_all64bitg2opcodes.S
./gdb/sim/testsuite/bfin/se_all16bitopcodes.S
./gdb/sim/testsuite/bfin/se_all32bitopcodes.lds
./gdb/opcodes/po/opcodes.pot
./gdb/gas/testsuite/gas/tic4x/opcodes.s
./gdb/gas/testsuite/gas/tic4x/allopcodes.S
./gdb/gas/testsuite/gas/tic4x/opcodes_c3x.d
./gdb/gas/testsuite/gas/tic4x/opcodes_new.d
./gdb/gas/testsuite/gas/tic4x/opcodes_c4x.d
./gdb/gas/testsuite/gas/tic54x/opcodes.s
./gdb/gas/testsuite/gas/tic54x/all-opcodes.s
./gdb/gas/testsuite/gas/tic54x/opcodes.d
[... 102 lines cut ...]
   101b4:       fe842703                lw      a4,-24(s0)
   101b8:       02e7878b                mod     a5,a5,a4
   101bc:       fef42223                sw      a5,-28(s0)
   101c0:       fe442783                lw      a5,-28(s0)
   101c4:       0007871b                sext.w  a4,a5
   101c8:       4785                    li      a5,1
   101ca:       00f70963                beq     a4,a5,101dc <main+0x40>
   101ce:       67c9                    lui     a5,0x12
   101d0:       60878513                addi    a0,a5,1544 # 12608 <__errno+0x8>
   101d4:       392000ef                jal     10566 <puts>
   101d8:       57fd                    li      a5,-1
   101da:       a039                    j       101e8 <main+0x4c>
   101dc:       67c9                    lui     a5,0x12
   101de:       61878513                addi    a0,a5,1560 # 12618 <__errno+0x18>
   101e2:       384000ef                jal     10566 <puts>
   101e6:       4781                    li      a5,0
   101e8:       853e                    mv      a0,a5
   101ea:       60e2                    ld      ra,24(sp)
   101ec:       6442                    ld      s0,16(sp)
   101ee:       6105                    addi    sp,sp,32
   101f0:       8082                    ret

...
```
As can be seen from the disassembly of `main()` the `mov` instruction is present and correct.

1781559198, 2024-04-30:
Thank you very much for the advice you gave me, my problem was solved

closed 2024-04-30 by 1781559198