riscv-collab/riscv-gnu-toolchain #1470: Single Assembly Instruction of Assigning a Variable is Generating Two Instructions

nathan-menhorn, 2024-04-30:
When using the inline C assembly instruction:

`__asm__ volatile("addk  %0, r0, %1" : "=r" (LHS) : "r" (RHS))`

This generates two assembly instructions instead of just assigning the variable to a register. For example, assign -1 into register a5 generates the assembly code:

`li a5, -1`
`mv a5, a5`

The first assembly instruction is expected but the second move of moving the same register into itself isn't expected.

GCC version 12.2.0
This is happening at all optimization levels and various settings of the gcc compiler, i.e. the extra move instruction always occurs and is independent of optimizations and compiler flags.

TommyMurphyTM1234, 2024-04-30:
> `__asm__ volatile("addk %0, r0, %1" : "=r" (LHS) : "r" (RHS))`

I presume that this is a typo and you meant `addi`?

Can you provide a self-contained reproducible test example please?

> GCC version 12.2.0

Have you tried it with GCC 13, 14 or 15?

nathan-menhorn, 2024-05-01:
Hi @TommyMurphyTM1234 this was found using the RISC-V MicroBlaze processor (https://www.xilinx.com/products/design-tools/microblaze-v.html) and it appears that this processor supports both RISC-V and MicroBlaze instructions so hence the addk was used by the developer and not the addi. The extra mv must have something to do with this. Apologies for any confusion but this can be closed.

As a side note, I tested the mv/addi inline assembly instruction surrounded by other code and do not see an extra move.

closed 2024-05-01 by nathan-menhorn