riscv-collab/riscv-gnu-toolchain #1471: any plan to bump GCC to 14.x ?

Du-Chao, 2024-05-07:
It's good to have the GCC 14 new features, and also support more RISC-V extensions.

TommyMurphyTM1234, 2024-05-07:
It was [only released today](https://gcc.gnu.org/gcc-14/) so it's not surprising that `riscv-gnu-toolchain` hasn't been bumped to use it yet.
I'll create a PR for it but if you need to build with it in the meantime then you can do the following:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc gcc-14 -b releases/gcc-14 
./configure ... --with-gcc-src=`pwd`/gcc-14
make
```

TommyMurphyTM1234, 2024-08-12:
New PR:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1531

TommyMurphyTM1234, 2024-08-14:
> New PR:
> 
> * [Bump gcc into 14.2 release version. #1531](https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1531)

I'll going to assume that the answer to the question here:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1531#issuecomment-2284315382

is "yes" and, consequently, I'm closing this issue and the related PR.

closed 2024-08-14 by TommyMurphyTM1234