riscv-collab/riscv-gnu-toolchain #1473: Building toolchain with all extensions support

ashuthosh-mr, 2024-05-08:
Hi, 

I have built the toolchain that can compile an application with RV64GC extensions and uses the maximum set of extensions available. How can I build the toolchain that can compile with more extensions or that support more extensions? Because I see there are so many march options supported as shown in https://gcc.gnu.org/onlinedocs/gcc/RISC-V-Options.html
Please let me know if there are already set of prebuilt toolchain similar to SiFive releases for all extension or a subset of them (more than RV64GC of course). Any more information will help.

Thanks in advance.

TommyMurphyTM1234, 2024-05-08:
Any toolchain can compile code for all ratified/implemented extensions when passed the appropriate `-march=... -mabi...` options. 
Depending on your use case you may also need the toolchain to provide startup code/multilibs for the chosen `arch/abi` in order to avoid link time issues - e.g. see here for some relevant discussion: 

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1315

Note that `riscv-gnu-toolchain` hasn't yet been bumped to use GCC 14 so if you need that and RISC-V support that it introduces then you will need to take steps to build it for now: 

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1471

It might help if you clarify which type of toolchain (e.g. GCC, LLVM/Clang, bare-metal, Linux etc.) and what specific RISC-V extensions you're interested in. E.g. what does your target RISC-V implementation support?

ashuthosh-mr, 2024-05-08:
Thanks for the response. I shall look into these. 

Not looking in terms of specific extensions or type, but need as much extension as possible, to see what extensions are suitable for certain applications and what extensions are not needed. That way, the application specific target RISCV can be tailored.

TommyMurphyTM1234, 2024-05-08:
Well, to put it another way, what `-march=... -mabi=...` combination(s?) are you intending to compile your code with?

Or, if it's the case that you want a toolchain that compiles using "all" extensions by default (in the absence of any explicit `-march=... -mabi=...` options) then you need to specify the relevant "all extensions" architecture at configuration time using `configure ... --with-arch=...`. I'm not sure what that architecture string/specifier would look like offhand though. You'd probably need to study the GCC documentation to see what extensions it supports and assemble the string manually.

ashuthosh-mr, 2024-05-08:
Alright, its the later case. I shall go through the strings to input manually. One more minor doubt, during configuration time, isn't it --with-multilib-generator? or is it --with-arch?

TommyMurphyTM1234, 2024-05-08:
The readme explains those options. 

* https://github.com/riscv-collab/riscv-gnu-toolchain

TommyMurphyTM1234, 2024-05-09:
I presume that this question can be closed as it has been answered and doesn't identify any bug/issue with the `riscv-gnu-toolchain`?

omerguzelelectronicguy, 2024-05-10:
@ashuthosh-mr I am using those combinations maybe it helps you :
```
../configure --prefix=/opt/riscv --with-cmodel=medany --enable-multilib --with-multilib-generator="rv32e-ilp32e--;rv32ea-ilp32e--;rv32em-ilp32e--;rv32eac-ilp32e--;rv32emac-ilp32e--;rv32i-ilp32--;rv32if-ilp32f--;rv32ifd-ilp32d--;rv32ia-ilp32--;rv32iaf-ilp32f--;rv32imaf-ilp32f--;rv32iafd-ilp32d--;rv32im-ilp32--;rv32imf-ilp32f--;rv32imfc-ilp32f--;rv32imfd-ilp32d--;rv32iac-ilp32--;rv32imac-ilp32--;rv32imafc-ilp32f--;rv32imafdc-ilp32d--;rv64i-lp64--;rv64if-lp64f--;rv64ifd-lp64d--;rv64ia-lp64--;rv64iaf-lp64f--;rv64imaf-lp64f--;rv64iafd-lp64d--;rv64im-lp64--;rv64imf-lp64f--;rv64imfc-lp64f--;rv64imfd-lp64d--;rv64iac-lp64--;rv64imac-lp64--;rv64imafc-lp64f--;rv64imafdc-lp64d--;"
```

TommyMurphyTM1234, 2024-05-10:
> @ashuthosh-mr I am using those combinations maybe it helps you :
>
> ```
> ../configure --prefix=/opt/riscv --with-cmodel=medany --enable-multilib [...]

The question doesn't seem to be about multilibs but about building a toolchain that has a default `arch/abi` that encompasses all available extensions rather than the default `rv64gc/lp64d` (or similar - it may actually be `rv64gc_zicsr_zifencei/lp64d` - I can never remember or keep up with the latest state of play in this regard). In that case multilibs are arguably irrelevant since the toolchain default libs will be sufficient.

ashuthosh-mr, 2024-05-11:
Thanks for the help. I shall close this question as it is resolved.

closed 2024-05-11 by ashuthosh-mr