riscv-collab/riscv-gnu-toolchain #1474: Compile arbitrary C code to assembly code

toolchainman, 2024-05-09:
Hello.
I finished installing riscv-gnu-toolchain in order. What I want to do is to randomly convert the C code I wrote into an assembly code converted to the rv32i ISA format. I can't find the explanation anywhere, maybe I'm not good at finding it. How can I convert a randomly written C code into an assembly to the rv32i ISA format? Need help. Thank you.

TommyMurphyTM1234, 2024-05-09:
I'm not sure what you mean "randomly" here: 

> What I want to do is to randomly convert the C code I wrote into an assembly code converted to the rv32i ISA format

But to compile C code to assembly code and not proceed to generating an object file you can use the `-S` option that is obviously documented in the GCC manual:

* https://gcc.gnu.org/onlinedocs/gcc-14.1.0/gcc/
  * https://gcc.gnu.org/onlinedocs/gcc-14.1.0/gcc/Overall-Options.html

To compile C code to RV32I assembly code that will appear in the generated `test.s` file you would do this:
```
riscv64-unknown-elf-gcc -march=rv32i -march=ilp32 -S test.c
```
I'm surprised that you couldn't find this info by doing a quick search - e.g. 

*

TommyMurphyTM1234, 2024-05-09:
There was a typo in my reply which I have fixed now. 
It should've been...
```
riscv64-unknown-elf-gcc -march=rv32i -mabi=ilp32 -S test.c
```

toolchainman, 2024-05-09:
Thank you. It is completed without error. I will refer to the data you gave me. Have a good night.

closed 2024-05-09 by TommyMurphyTM1234