riscv-collab/riscv-gnu-toolchain #1476: I meet some truoble aboult branch jump instruction

Erucaaa, 2024-05-13:
Here are my code:

#include<stdio.h>
#include<string.h>
#include<stdint.h>

void compare_and_swap(int a[], int &loadFrom, int &opposite) {
    __asm__ volatile(
        "c.mv a2, %2 \n"          
        "slt a1, %0, %1 \n"       
        "bnez a1, eqa \n"   
        "c.mv %2, %3 \n"          
        "c.mv %3, a2 \n"
        "eqa: \n"           
        : "=r" (a[loadFrom]), "=r" (a[opposite]), "=r" (loadFrom), "=r" (opposite)  
        : "0" (a[loadFrom]), "1" (a[opposite]), "2" (loadFrom), "3" (opposite)      
        : "cc", "memory", "a1", "a2"                                                
    );
}

int main()
{

  int32_t a[2] = {6,5};
  int loadFrom = 0;
  int opposite = 1;
  compare_and_swap(a,loadFrom,opposite);

  printf("%d \n",a[loadFrom]);

}
I want to create a function for comparing two elements. The loadFrom variable always represents the index of the smaller value.
If I use the -O3 optimization option, the result is error:
[image]
But if I can't use any optimization options, the result is true.
I use volatile to ensure the compiler doesn't optimize this part. I'm not sure if it's related to compiler optimization.

Erucaaa, 2024-05-13:
> Please clarify the full command line used to compile your code in each case. It's not clear what toolchain you're using -.e.g. bare-metal, Linux, other.

g++ main.cpp -o main.exe -O3   and  g++ main.cpp -o main.exe
Thanks

TommyMurphyTM1234, 2024-05-13:
> > Please clarify the full command line used to compile your code in each case. It's not clear what toolchain you're using -.e.g. bare-metal, Linux, other.
> 
> g++ main.cpp -o main.exe -O3 and g++ main.cpp -o main.exe Thanks

`g++` is not a RISC-V compiler unless you've created a link to `riscv64-unknown-elf-gcc` or similar. Please provide full details needed to reproduce your issue if you want assistance here.

Erucaaa, 2024-05-13:
> > > Please clarify the full command line used to compile your code in each case. It's not clear what toolchain you're using -.e.g. bare-metal, Linux, other.
> > 
> > 
> > g++ main.cpp -o main.exe -O3 and g++ main.cpp -o main.exe Thanks
> 
> `g++` is not a RISC-V compiler unless you've created a link to `riscv64-unknown-elf-g++` or similar. Please provide full details needed to reproduce your issue if you want assistance here.

/opt/gcc10.2/native/lib/gcc/riscv64-linux-gnu/10.2.0/specs

TommyMurphyTM1234, 2024-05-13:
> /opt/gcc10.2/native/lib/gcc/riscv64-linux-gnu/10.2.0/specs

I'm not exactly sure what you mean by this?
Are you saying that you're using the RISC-V Linux GCC 10.2.0 compiler?
If so then you should try a more recent version.
10.2 is very old and `riscv-gnu-toolchain` is currently at 14.1.

TommyMurphyTM1234, 2024-05-13:
OK - I can reproduce the issue with the [latest CI build of the toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain/releases):
```
// test.cpp
#include<stdio.h>
#include<string.h>
#include<stdint.h>

void compare_and_swap(int a[], int &loadFrom, int &opposite) {
  asm volatile(
    "c.mv a2, %2 \n"
    "slt a1, %0, %1 \n"
    "bnez a1, eqa \n"
    "c.mv %2, %3 \n"
    "c.mv %3, a2 \n"
    "eqa: \n"
    : "=r" (a[loadFrom]), "=r" (a[opposite]), "=r" (loadFrom), "=r" (opposite)
    : "0" (a[loadFrom]), "1" (a[opposite]), "2" (loadFrom), "3" (opposite)
    : "cc", "memory", "a1", "a2"
  );
}

int main()
{
  int32_t a[2] = {6,5};
  int loadFrom = 0;
  int opposite = 1;
  compare_and_swap(a,loadFrom,opposite);

  printf("%d \n",a[loadFrom]);
}
```
```
wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.04.12/riscv64-glibc-ubuntu-22.04-gcc-nightly-2024.04.12-nightly.tar.gz
tar xzvf riscv64-glibc-ubuntu-22.04-gcc-nightly-2024.04.12-nightly.tar.gz
cd riscv/bin
./riscv64-unknown-linux-gnu-g++ test.cpp -o test

file test
test: ELF 64-bit LSB executable, UCB RISC-V, RVC, double-float ABI, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-riscv64-lp64d.so.1, for GNU/Linux 4.15.0, not stripped

./riscv64-unknown-linux-gnu-g++ -O3 test.cpp -o test
test.cpp: Assembler messages:
test.cpp:11: Error: symbol `eqa' is already defined
```
Looking at the generated assembly code for the `-O3` case...
```
./riscv64-unknown-linux-gnu-g++ -S -O3 test.cpp
```
`test.s` contains:
```
        .file   "test.cpp"
        .option nopic
        .attribute arch, "rv64i2p1_m2p0_a2p1_f2p2_d2p2_c2p0_zicsr2p0_zifencei2p0"
        .attribute unaligned_access, 0
        .attribute stack_align, 16
        .text
        .align  1
        .globl  _Z16compare_and_swapPiRiS0_
        .type   _Z16compare_and_swapPiRiS0_, @function
_Z16compare_and_swapPiRiS0_:
.LFB26:
        .cfi_startproc
        lw      a4,0(a1)
        lw      a5,0(a2)
        mv      t1,a1
        slli    t3,a4,2
        slli    a3,a5,2
        add     t3,a0,t3
        add     a0,a0,a3
        lw      a6,0(t3)
        lw      a3,0(a0)
        mv      a7,a2
#APP
# 6 "test.cpp" 1
        c.mv a2, a4
slt a1, a6, a3
bnez a1, eqa
c.mv a4, a5
c.mv a5, a2
eqa:

# 0 "" 2
#NO_APP
        sw      a6,0(t3)
        sw      a3,0(a0)
        sw      a4,0(t1)
        sw      a5,0(a7)
        ret
        .cfi_endproc
.LFE26:
        .size   _Z16compare_and_swapPiRiS0_, .-_Z16compare_and_swapPiRiS0_
        .section        .rodata.str1.8,"aMS",@progbits,1
        .align  3
.LC0:
        .string "%d \n"
        .section        .text.startup,"ax",@progbits
        .align  1
        .globl  main
        .type   main, @function
main:
.LFB27:
        .cfi_startproc
        li      a5,5
        slli    a5,a5,32
        addi    sp,sp,-32
        .cfi_def_cfa_offset 32
        addi    a5,a5,6
        sd      ra,24(sp)
        .cfi_offset 1, -8
        sd      a5,8(sp)
        li      a3,6
        li      a4,5
        li      a5,0
        li      a0,1
#APP
# 6 "test.cpp" 1
        c.mv a2, a5
slt a1, a3, a4
bnez a1, eqa
c.mv a5, a0
c.mv a0, a2
eqa:

# 0 "" 2
#NO_APP
        sext.w  a5,a5
        slli    a5,a5,2
        addi    a5,a5,16
        sw      a3,8(sp)
        sw      a4,12(sp)
        add     a5,a5,sp
        lw      a1,-8(a5)
        lui     a0,%hi(.LC0)
        addi    a0,a0,%lo(.LC0)
        call    printf
        ld      ra,24(sp)
        .cfi_restore 1
        li      a0,0
        addi    sp,sp,32
        .cfi_def_cfa_offset 0
        jr      ra
        .cfi_endproc
.LFE27:
        .size   main, .-main
        .ident  "GCC: () 13.2.0"
        .section        .note.GNU-stack,"",@progbits
```
For some reason the label `eqa` is defined twice.

But this may be an issue with your inline assembly code rather than the toolchain per se?

Erucaaa, 2024-05-13:
> OK - I can reproduce the issue with the [latest CI build of the toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain/releases):
> 
> ```
> // test.cpp
> #include<stdio.h>
> #include<string.h>
> #include<stdint.h>
> 
> void compare_and_swap(int a[], int &loadFrom, int &opposite) {
>   asm volatile(
>     "c.mv a2, %2 \n"
>     "slt a1, %0, %1 \n"
>     "bnez a1, eqa \n"
>     "c.mv %2, %3 \n"
>     "c.mv %3, a2 \n"
>     "eqa: \n"
>     : "=r" (a[loadFrom]), "=r" (a[opposite]), "=r" (loadFrom), "=r" (opposite)
>     : "0" (a[loadFrom]), "1" (a[opposite]), "2" (loadFrom), "3" (opposite)
>     : "cc", "memory", "a1", "a2"
>   );
> }
> 
> int main()
> {
>   int32_t a[2] = {6,5};
>   int loadFrom = 0;
>   int opposite = 1;
>   compare_and_swap(a,loadFrom,opposite);
> 
>   printf("%d \n",a[loadFrom]);
> }
> ```
> 
> ```
> wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2024.04.12/riscv64-glibc-ubuntu-22.04-gcc-nightly-2024.04.12-nightly.tar.gz
> tar xzvf riscv64-glibc-ubuntu-22.04-gcc-nightly-2024.04.12-nightly.tar.gz
> 
> cd riscv/bin
> 
> ./riscv64-unknown-linux-gnu-g++ -v
> Using built-in specs.
> COLLECT_GCC=./riscv64-unknown-linux-gnu-g++
> COLLECT_LTO_WRAPPER=/home/user/Downloads/riscv/bin/../libexec/gcc/riscv64-unknown-linux-gnu/13.2.0/lto-wrapper
> Target: riscv64-unknown-linux-gnu
> Configured with: /home/runner/work/riscv-gnu-toolchain/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/opt/riscv --with-sysroot=/opt/riscv/sysroot --with-pkgversion= --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=.././gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-O2    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2    -mcmodel=medlow'
> Thread model: posix
> Supported LTO compression algorithms: zlib zstd
> gcc version 13.2.0 ()
> 
> ./riscv64-unknown-linux-gnu-g++ test.cpp -o test
> 
> file test
> test: ELF 64-bit LSB executable, UCB RISC-V, RVC, double-float ABI, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-riscv64-lp64d.so.1, for GNU/Linux 4.15.0, not stripped
> 
> ./riscv64-unknown-linux-gnu-g++ -O3 test.cpp -o test
> test.cpp: Assembler messages:
> test.cpp:11: Error: symbol `eqa' is already defined
> ```
> 
> Looking at the generated assembly code for the `-O3` case...
> 
> ```
> ./riscv64-unknown-linux-gnu-g++ -S -O3 test.cpp
> ```
> 
> `test.s` contains:
> 
> ```
>         .file   "test.cpp"
>         .option nopic
>         .attribute arch, "rv64i2p1_m2p0_a2p1_f2p2_d2p2_c2p0_zicsr2p0_zifencei2p0"
>         .attribute unaligned_access, 0
>         .attribute stack_align, 16
>         .text
>         .align  1
>         .globl  _Z16compare_and_swapPiRiS0_
>         .type   _Z16compare_and_swapPiRiS0_, @function
> _Z16compare_and_swapPiRiS0_:
> .LFB26:
>         .cfi_startproc
>         lw      a4,0(a1)
>         lw      a5,0(a2)
>         mv      t1,a1
>         slli    t3,a4,2
>         slli    a3,a5,2
>         add     t3,a0,t3
>         add     a0,a0,a3
>         lw      a6,0(t3)
>         lw      a3,0(a0)
>         mv      a7,a2
> #APP
> # 6 "test.cpp" 1
>         c.mv a2, a4
> slt a1, a6, a3
> bnez a1, eqa
> c.mv a4, a5
> c.mv a5, a2
> eqa:
> 
> # 0 "" 2
> #NO_APP
>         sw      a6,0(t3)
>         sw      a3,0(a0)
>         sw      a4,0(t1)
>         sw      a5,0(a7)
>         ret
>         .cfi_endproc
> .LFE26:
>         .size   _Z16compare_and_swapPiRiS0_, .-_Z16compare_and_swapPiRiS0_
>         .section        .rodata.str1.8,"aMS",@progbits,1
>         .align  3
> .LC0:
>         .string "%d \n"
>         .section        .text.startup,"ax",@progbits
>         .align  1
>         .globl  main
>         .type   main, @function
> main:
> .LFB27:
>         .cfi_startproc
>         li      a5,5
>         slli    a5,a5,32
>         addi    sp,sp,-32
>         .cfi_def_cfa_offset 32
>         addi    a5,a5,6
>         sd      ra,24(sp)
>         .cfi_offset 1, -8
>         sd      a5,8(sp)
>         li      a3,6
>         li      a4,5
>         li      a5,0
>         li      a0,1
> #APP
> # 6 "test.cpp" 1
>         c.mv a2, a5
> slt a1, a3, a4
> bnez a1, eqa
> c.mv a5, a0
> c.mv a0, a2
> eqa:
> 
> # 0 "" 2
> #NO_APP
>         sext.w  a5,a5
>         slli    a5,a5,2
>         addi    a5,a5,16
>         sw      a3,8(sp)
>         sw      a4,12(sp)
>         add     a5,a5,sp
>         lw      a1,-8(a5)
>         lui     a0,%hi(.LC0)
>         addi    a0,a0,%lo(.LC0)
>         call    printf
>         ld      ra,24(sp)
>         .cfi_restore 1
>         li      a0,0
>         addi    sp,sp,32
>         .cfi_def_cfa_offset 0
>         jr      ra
>         .cfi_endproc
> .LFE27:
>         .size   main, .-main
>         .ident  "GCC: () 13.2.0"
>         .section        .note.GNU-stack,"",@progbits
> ```
> 
> For some reason the label `eqa` is defined twice.
> 
> But this may be an issue with your inline assembly code rather than the toolchain per se?

Thanks. I seem to have found the reason. I used numeric label to replace 'eqa' and then obtained the correct result.
__asm__(
        "c.mv a2, %2 \n"          
        "slt a1, %0, %1 \n"       
        "bnez a1, 1f \n"   
        "c.mv %2, %3 \n"          
        "c.mv %3, a2 \n"
        "1: \n"           
        : "=r" (a[loadFrom]), "=r" (a[opposite]), "=r" (loadFrom), "=r" (opposite)  
        : "0" (a[loadFrom]), "1" (a[opposite]), "2" (loadFrom), "3" (opposite)      
        : "cc", "memory", "a1", "a2"                                                
    );

Erucaaa, 2024-05-13:
> OK - I guess that it's probably some nuance with the inline assembly code so and that this issue can be closed?

Of course. Thanks for your help!

closed 2024-05-13 by TommyMurphyTM1234