riscv-collab/riscv-gnu-toolchain #1479: I cannot download and setup some submodules

Ersanny, 2024-05-17:
Hello, I have some problems. I cannot download some submodules.  I cannot access some folder like binutils.

TommyMurphyTM1234, 2024-05-17:
> Hello, I have some problems. I cannot download some submodules. I cannot access some folder like binutils. ![image](https://private-user-images.githubusercontent.com/101807681/331638903-c717d885-2751-4730-9c99-07bbfa45d178.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU5NTgxMzEsIm5iZiI6MTcxNTk1NzgzMSwicGF0aCI6Ii8xMDE4MDc2ODEvMzMxNjM4OTAzLWM3MTdkODg1LTI3NTEtNDczMC05Yzk5LTA3YmJmYTQ1ZDE3OC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUxN1QxNDU3MTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iN2EyNjA5YTM4NzNlZTdkNzEzNjVjYjE5ZGQyMjhmNjZjYjllM2E5NjE5NzAwNmVmN2VmM2NiZmUyNmQyZDQyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5HBjRzujfOA7v9gX1URetwlMcKQVIBNP3HTBw-B1rUc) When I click here this page comes up.

When you click where?

> When I click here this page comes up.
> "git submodule update --init --recursive" I can't download with this command. I'm getting an error

What error?

What exactly happens when you do this?
```
git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain
```
I did this just now and it worked fine:
```
git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain
Cloning into 'riscv-gnu-toolchain'...
remote: Enumerating objects: 9198, done.
remote: Counting objects: 100% (1022/1022), done.
remote: Compressing objects: 100% (345/345), done.
remote: Total 9198 (delta 758), reused 685 (delta 677), pack-reused 8176
Receiving objects: 100% (9198/9198), 5.37 MiB | 2.94 MiB/s, done.
Resolving deltas: 100% (4957/4957), done.
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Submodule 'llvm' (https://github.com/llvm/llvm-project.git) registered for path 'llvm'
Submodule 'musl' (https://git.musl-libc.org/git/musl) registered for path 'musl'
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Submodule 'pk' (https://github.com/riscv-software-src/riscv-pk.git) registered for path 'pk'
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/binutils'...
remote: Enumerating objects: 1223613, done.
remote: Counting objects: 100% (49867/49867), done.
remote: Compressing objects: 100% (42237/42237), done.
remote: Total 1223613 (delta 12225), reused 10189 (delta 7550), pack-reused 1173746
Receiving objects: 100% (1223613/1223613), 456.76 MiB | 10.07 MiB/s, done.
Resolving deltas: 100% (1016200/1016200), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/dejagnu'...
remote: Counting objects: 9063, done.
remote: Compressing objects: 100% (2045/2045), done.
remote: Total 9063 (delta 6926), reused 9013 (delta 6890)
Receiving objects: 100% (9063/9063), 4.71 MiB | 5.39 MiB/s, done.
Resolving deltas: 100% (6926/6926), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/gcc'...
remote: Enumerating objects: 3040610, done.
remote: Counting objects: 100% (241000/241000), done.
remote: Compressing objects: 100% (21041/21041), done.
remote: Total 3040610 (delta 230621), reused 224817 (delta 219773), pack-reused 2799610
Receiving objects: 100% (3040610/3040610), 1.18 GiB | 11.93 MiB/s, done.
Resolving deltas: 100% (2501087/2501087), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/gdb'...
remote: Enumerating objects: 1223613, done.
remote: Counting objects: 100% (49867/49867), done.
remote: Compressing objects: 100% (42237/42237), done.
remote: Total 1223613 (delta 12225), reused 10189 (delta 7550), pack-reused 1173746
Receiving objects: 100% (1223613/1223613), 456.77 MiB | 10.22 MiB/s, done.
Resolving deltas: 100% (1016200/1016200), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/glibc'...
remote: Enumerating objects: 722356, done.
remote: Counting objects: 100% (10227/10227), done.
remote: Compressing objects: 100% (3277/3277), done.
remote: Total 722356 (delta 8263), reused 7948 (delta 6781), pack-reused 712129
Receiving objects: 100% (722356/722356), 224.76 MiB | 7.56 MiB/s, done.
Resolving deltas: 100% (620613/620613), done.
Cloning into '/home/user/Downloads/riscv-gnu-toolchain/llvm'...
remote: Enumerating objects: 5939815, done.
[... 329 lines cut ...]
Receiving objects: 100% (9580/9580), 6.74 MiB | 2.46 MiB/s, done.
Resolving deltas: 100% (6547/6547), done.
Submodule path 'qemu/roms/edk2/MdePkg/Library/MipiSysTLib/mipisyst/external/googletest': checked out 'a6f06bf2fd3b832822cd4e9e554b7d47f32ec084'
Submodule path 'qemu/roms/edk2/MdePkg/Library/MipiSysTLib/mipisyst/external/pugixml': checked out 'c53fdab93af76106b963216d85897614b996f8b6'
Submodule path 'qemu/roms/edk2/RedfishPkg/Library/JsonLib/jansson': checked out 'e9ebfa7e77a6bee77df44e096b100e7131044059'
[... cut ...]
Submodule path 'qemu/roms/vbootrom': checked out '0c37a43527f0ee2b9584e7fb2fdc805e902635ac'
Submodule path 'qemu/tests/lcitool/libvirt-ci': checked out '77c800186f34b21be7660750577cc5582a914deb'
Submodule path 'spike': checked out '20a2b6d05a562236a32ca3ce1642a2bdbbd907a0'
```
Problems with GitHub repo access are usually down to local host or network configuration issues.
Occasionally GitHub repos exhibit connectivity/responsiveness issues but that's usually rare and transient.
But I can't see that there's any such problem at the moment.

drkome, 2024-05-17:
I have the same problem.  if I did git clone, I cant download. I just waiting console. At the end of the job, I encounter an error It may be a regional problem. or dns also sometimes I cant open sourcewave.  I wish these files were uploaded to a second platform.

TommyMurphyTM1234, 2024-05-17:
> At the end of the job, I encounter an error

It might help if you detailed the error.

drkome, 2024-05-18:
I tried to install it from scratch to get rid of this error. But in the morning it loaded some submodules meaninglessly. Even yesterday it wasn't loading the submodules. But llvm also gave an error this time.

`Cloning into 'riscv-gnu-toolchain'...
remote: Enumerating objects: 9198, done.
remote: Counting objects: 100% (1022/1022), done.
remote: Compressing objects: 100% (345/345), done.
remote: Total 9198 (delta 758), reused 685 (delta 677), pack-reused 8176
Receiving objects: 100% (9198/9198), 5.37 MiB | 1.23 MiB/s, done.
Resolving deltas: 100% (4957/4957), done.
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Submodule 'llvm' (https://github.com/llvm/llvm-project.git) registered for path 'llvm'
Submodule 'musl' (https://git.musl-libc.org/git/musl) registered for path 'musl'
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Submodule 'pk' (https://github.com/riscv-software-src/riscv-pk.git) registered for path 'pk'
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Cloning into '/home/kamer/deneeeee/riscv-gnu-toolchain/binutils'...
remote: Enumerating objects: 1223699, done.
remote: Counting objects: 100% (49953/49953), done.
remote: Compressing objects: 100% (42336/42336), done.
Receiving objects: 100% (1223699/1223699), 457.01 MiB | 590.00 KiB/s, done.
remote: Total 1223699 (delta 12290), reused 10164 (delta 7537), pack-reused 1173746
Resolving deltas: 100% (1016265/1016265), done.
Cloning into '/home/kamer/deneeeee/riscv-gnu-toolchain/dejagnu'...
remote: Counting objects: 9063, done.
remote: Compressing objects: 100% (2045/2045), done.
remote: Total 9063 (delta 6926), reused 9013 (delta 6890)
Receiving objects: 100% (9063/9063), 4.71 MiB | 1.32 MiB/s, done.
Resolving deltas: 100% (6926/6926), done.
Cloning into '/home/kamer/deneeeee/riscv-gnu-toolchain/gcc'...
remote: Enumerating objects: 3040748, done.
remote: Counting objects: 100% (241138/241138), done.
remote: Compressing objects: 100% (21179/21179), done.
Receiving objects: 100% (3040748/3040748), 1.18 GiB | 515.00 KiB/s, done.
remote: Total 3040748 (delta 230725), reused 224811 (delta 219772), pack-reused 2799610
Resolving deltas: 100% (2501191/2501191), done.
Cloning into '/home/kamer/deneeeee/riscv-gnu-toolchain/gdb'...
remote: Enumerating objects: 1223699, done.
remote: Counting objects: 100% (49953/49953), done.
remote: Compressing objects: 100% (42336/42336), done.
Receiving objects: 100% (1223699/1223699), 457.02 MiB | 367.00 KiB/s, done.
remote: Total 1223699 (delta 12289), reused 10164 (delta 7537), pack-reused 1173746
Resolving deltas: 100% (1016264/1016264), done.
Cloning into '/home/kamer/deneeeee/riscv-gnu-toolchain/glibc'...
remote: Enumerating objects: 722364, done.
remote: Counting objects: 100% (10235/10235), done.
remote: Compressing objects: 100% (3285/3285), done.
remote: Total 722364 (delta 8267), reused 7947 (delta 6781), pack-reused 712129
Receiving objects: 100% (722364/722364), 224.77 MiB | 272.00 KiB/s, done.
Resolving deltas: 100% (620617/620617), done.
Cloning into '/home/kamer/deneeeee/riscv-gnu-toolchain/llvm'...
remote: Enumerating objects: 5940866, done.
remote: Counting objects: 100% (1371/1371), done.
remote: Compressing objects: 100% (606/606), done.
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
error: 7772 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
fatal: clone of 'https://github.com/llvm/llvm-project.git' into submodule path '/home/kamer/deneeeee/riscv-gnu-toolchain/llvm' failed
Failed to clone 'llvm'. Retry scheduled
Cloning into '/home/kamer/deneeeee/riscv-gnu-toolchain/musl'...
remote: warning: unable to access '/root/.config/git/attributes': Permission denied
remote: Enumerating objects: 38153, done.
remote: Counting objects: 100% (38153/38153), done.
remote: Compressing objects: 100% (16603/16603), done.
[... 19 lines cut ...]
remote: Counting objects: 100% (5435/5435), done.
remote: Compressing objects: 100% (1877/1877), done.
remote: Total 748480 (delta 3760), reused 5021 (delta 3518), pack-reused 743045 (from 1)
Receiving objects: 100% (748480/748480), 349.87 MiB | 1.32 MiB/s, done.
Resolving deltas: 100% (624315/624315), done.
Cloning into '/home/kamer/deneeeee/riscv-gnu-toolchain/spike'...
remote: Enumerating objects: 22478, done.
remote: Counting objects: 100% (116/116), done.
remote: Compressing objects: 100% (74/74), done.
remote: Total 22478 (delta 55), reused 74 (delta 41), pack-reused 22362
Receiving objects: 100% (22478/22478), 6.10 MiB | 393.00 KiB/s, done.
Resolving deltas: 100% (14684/14684), done.
Cloning into '/home/kamer/deneeeee/riscv-gnu-toolchain/llvm'...
remote: Enumerating objects: 5940866, done.
remote: Counting objects: 100% (1371/1371), done.
remote: Compressing objects: 100% (606/606), done.
[... cut ...]
Failed to clone 'llvm' a second time, aborting`

I think this is a dns problem or regional problem. Because yesterday I couldn't even download the file until here. Even the binultis file was not downloading at all 6 months ago. I shared a post on reddit. I saw another person having the same problem. A few months ago, a German friend of mine uploaded these files and shared them with me. I couldn't download most of them at the time.Internet traffic in the download areas may also cause this. I don't know exactly.

TommyMurphyTM1234, 2024-05-18:
> But in the morning it loaded some submodules meaninglessly. 

What do you mean by "meaninglessly" here?

> (I am sorry for code, When I copied the texts from the Linux terminal, they pasted them in a complicated and irregular manner. That's the only way I could edit it.)

You should use the code Markdown code blocks for such content:

* https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

> ```
> Cloning into '/home/kamer/deneeeee/riscv-gnu-toolchain/llvm'...
> remote: Enumerating objects: 5940866, done.
> remote: Counting objects: 100% (1371/1371), done.
> remote: Compressing objects: 100% [...]

You may find some something useful for diagnosing or addressing the `unexpected disconnect while reading sideband packet` error here:

* https://www.google.com/search?q=github+unexpected+disconnect+while+reading+sideband+packet&oq=github+unexpected+disconnect+while+reading+sideband+packet

Maybe doing a shallow clone to reduce the amount of data transferred may help while these connectivity issues persist?
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain --recursive --depth 1
```
Another workaround might be to download tarballs for some or all of the relevant submodules and configure the build to use these rather than trying to clone them from a repo:

* https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#use-source-tree-other-than-riscv-gnu-toolchain

Please bear in mind that the `riscv-gnu-toolchain` repo uses submodules in order to fetch sources from the relevant upstream official repos and has no direct control over these repos or possible localised connectivity issues that may arise.

TommyMurphyTM1234, 2024-05-18:
> But llvm also gave an error this time. 

FWIW - similar issue reported here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1453

TommyMurphyTM1234, 2024-05-18:
FWIW I did several more clones today as follows and each one worked fine:
```
https://github.com/riscv-collab/riscv-gnu-toolchain --recursive
...

https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git submodule update --init --recursive --progress
...
```

TommyMurphyTM1234, 2024-05-23:
Closing this issue as it doesn't seem to be anything specifically to do with the `riscv-gnu-toolchain` repo.

closed 2024-05-23 by TommyMurphyTM1234