riscv-collab/riscv-gnu-toolchain #1480: Some submodule URLs give 404 errors

TommyMurphyTM1234, 2024-05-18:
Apropos of this issue:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1479

And, in particular, this comment:

> When I click here [presumably the links in the GitHub screenshot below] this [404 error] page comes up.
>
> [image]

it seems that most of the submodule URLs here don't work and give 404 errors - those highlighted in red don't work while those highlighted in green do:

[image]

Specifically - these ones generate 404 errors:

- binutils: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/binutils
- dejagnu: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/dejagnu
- gcc: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/gcc
- gdb: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/gdb
- glibc: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/glibc
- musl: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/musl
- newlib: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/newlib
- qemu: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/qemu

These ones work:

- llvm: https://github.com/llvm/llvm-project/tree/b2417f51dbbd7435eb3aaf203de24de6754da50e
- pk: https://github.com/riscv-software-src/riscv-pk/tree/2efabd3e6604b8a9e8f70baf52f57696680c7855
- spike: https://github.com/riscv-software-src/riscv-isa-sim/tree/20a2b6d05a562236a32ca3ce1642a2bdbbd907a0

I guess that some attribute(s) of the submodules need to be tweaked to make the links work?
If anybody knows how to do this then please post back otherwise I'll investigate and hopefully create a PR.

cmuellner, 2024-05-19:
I think

> Specifically - these ones generate 404 errors:
> 
> * binutils: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/binutils

https://sourceware.org/git/binutils-gdb.git

> * dejagnu: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/dejagnu

https://git.savannah.gnu.org/git/dejagnu.git

> * gcc: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/gcc

https://gcc.gnu.org/git/gcc.git

> * gdb: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/gdb

https://sourceware.org/git/binutils-gdb.git

> * glibc: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/glibc

https://sourceware.org/git/glibc.git

> * musl: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/musl

https://git.musl-libc.org/git/musl

> * newlib: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/newlib

https://sourceware.org/git/newlib-cygwin.git

> * qemu: https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/qemu

https://gitlab.com/qemu-project/qemu.git
 
> These ones work:
> 
> * llvm: https://github.com/llvm/llvm-project/tree/b2417f51dbbd7435eb3aaf203de24de6754da50e

https://github.com/llvm/llvm-project.git

> * pk: https://github.com/riscv-software-src/riscv-pk/tree/2efabd3e6604b8a9e8f70baf52f57696680c7855

https://github.com/riscv-software-src/riscv-pk.git

> * spike: https://github.com/riscv-software-src/riscv-isa-sim/tree/20a2b6d05a562236a32ca3ce1642a2bdbbd907a0

https://github.com/riscv-software-src/riscv-isa-sim.git

So, only the links on the submodules to github.com work. Links to submodules pointing to other locations fail.

In the past, git submodules to repos outside of github.com were rendered as non-links (i.e., you could not click on them). It seems like this behavior has changed.

The behavior that I see when I click on these links is that nothing happens. I get the 404 only if I open the link in a new tab.

TommyMurphyTM1234, 2024-05-19:
Thanks @cmuellner.

> The behavior that I see when I click on these links is that nothing happens. I get the 404 only if I open the link in a new tab.

Sorry, I should have made that clear. Directly clicking or Ctrl-clicking also does nothing for me (in Chrome).
I have to right click and choose _Open link in ..._ to try to open it and then get the 404 error page for the ones that don't work.

> So, only the links on the submodules to github.com work. Links to submodules pointing to other locations fail.

I hadn't noticed that the non-working links are actually for submodules that are hosted in non-GitHub repos:

https://github.com/riscv-collab/riscv-gnu-toolchain/blob/6d7b5b720f6c6a89b4f22c62b22c1d0ebc2dc353/.gitmodules#L1-L42

> In the past, git submodules to repos outside of github.com were rendered as non-links (i.e., you could not click on them). It seems like this behavior has changed.

That sounds plausible alright. I don't know if there's any way to tweak things so make the GitHub generated links to non-GitHub submodules work correctly?

It's not a big deal but just has caused confusion here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1479
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1450#issue-2226857430

TommyMurphyTM1234, 2024-05-20:
Since this seems to be an issue with GitHub itself, that it creates incorrect URLs for non-GitHub submodules, and not an issue with the `riscv-gnu-toolchain` repo I'm going to close it.

closed 2024-05-20 by TommyMurphyTM1234