riscv-collab/riscv-gnu-toolchain #1481: Support for python 3.12

TheRealIndianBoi, 2024-05-20:
Hey i got the Problem, that i am using the latest Ubuntu 24.04 LTS which is using python 3.12 as the standard. Running risv32-unknown-elf-gdb gives me the Error: /riscv32-unknown-elf-gdb: error while loading shared libraries: libpython3.10.so.1.0: cannot open shared object file: No such file or directory
Can you guys help me fix this problem, or do i have to downgrade to 3.10. But i don't know if i brake the system or not.

TommyMurphyTM1234, 2024-05-20:
Please clarify the commands used to build the toolchain and attach (after compressing if necessary) your toolchain build log. E.g. 
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure ... --with-arch=rv32... --with-abi=ilp32... 
make 2>&1 | tee build.log
```

TheRealIndianBoi, 2024-05-20:
I run the programm from the nightly-release. I see the problem, that the nightly has been made for ubuntu 22.04 and not 24.04... Do i have to build the riscv-gnu-toolchain on my own to let it all run on python3.12?

TommyMurphyTM1234, 2024-05-20:
> Do i have to build the riscv-gnu-toolchain on my own to let it all run on python3.12?

If you need GDB with Python 3.12 support then you'll probably have to do that for the moment and until such time as the CI is updated to use 24.04 LTS.

Or maybe you can install or build Python 3.10 on Ubuntu 24.04?

If you're not actually using GDB's Python support then I don't think that there's any way to invoke GDB with Python support disabled.

TheRealIndianBoi, 2024-05-20:
I cant install Python 3.10, but i try to build it local. Sadly its stuck on cloning into '...gcc'

TommyMurphyTM1234, 2024-05-20:
> I cant install Python 3.10, but i try to build it local. Sadly its stuck on cloning into '...gcc'

Could be the same (local connectivity?) issue that some others have been experiencing lately. 

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1479

TheRealIndianBoi, 2024-05-20:
Ah okay thanks gonna look at it.

TheRealIndianBoi, 2024-05-21:
Fixed through git cloning recursive

closed 2024-05-21 by TheRealIndianBoi

TheRealIndianBoi, 2024-05-21:
> Did you manage to successfully clone, configure and build the toolchain to
> meet your needs including GDB with Python 3.12 support on Ubuntu 24.04?

Yes i was able to clone it and build it according to python 3.12. The
problem eas that i should clone it with --recursive else it wont work.