riscv-collab/riscv-gnu-toolchain #1482: M and Zmmul extensions do not emmit mul instructions the same way when multiplying an integer with a constant in C

marklinmax, 2024-05-23:
When multiplying integers with a constant value in C, with Zmmul gcc apparently tries to not emmit mul instructions (it may do it for large constants). Instead it tries to implement the operation with some shift and add immediate instructions (which obviously is less optimized than just a single mul operation).

Considering the following code snippet:
```C:
void doSomething(int *ia, int *ib, int *ic) {
    *ia = *ia * 80;
    *ib = *ib * 33938;
    *ic = *ic * 80;
    *ia = *ib * *ic;
}
```
With M extension this will generate:
```
lw      a5,0(a0)
li      a3,80
li      a4,32768
mul     a5,a5,a3
addi    a4,a4,1170
sw      a5,0(a0)
lw      a5,0(a1)
mul     a5,a5,a4
sw      a5,0(a1)
lw      a5,0(a2)
mul     a5,a5,a3
sw      a5,0(a2)
lw      a4,0(a1)
mul     a5,a4,a5
sw      a5,0(a0)
ret
```
With Zmmul extension instead:
```
lw      a4,0(a0)
slli    a5,a4,2
add     a5,a5,a4
slli    a5,a5,4
sw      a5,0(a0)
lw      a5,0(a1)
li      a4,32768
addi    a4,a4,1170
mul     a5,a5,a4
sw      a5,0(a1)
lw      a4,0(a2)
slli    a5,a4,2
add     a5,a5,a4
slli    a5,a5,4
sw      a5,0(a2)
lw      a4,0(a1)
mul     a5,a4,a5
sw      a5,0(a0)
ret
```

Tested with gcc 13.2.0 on my machine and gcc 14.1.0 and 13.2.0 and 13.1.0 on https://godbolt.org/ as well. Also tested with -O0, -O1, -O2, -O3 and -Os.

My workaround for now is to use M extension and add -mno-div to compilation options (which behaves as we would expect Zmmul to do).

kito-cheng, 2024-05-23:
Could you also file a bug to upstream GCC https://gcc.gnu.org/bugzilla/ ? thanks :)

pz9115, 2024-05-30:
Thanks for your report, fixed on upstream
https://gcc.gnu.org/git/?p=gcc.git;a=commit;h=06bb125521dec5648b725ddee4345b00decfdc77

cmuellner, 2024-05-31:
@kito-cheng and @pz9115, I believe this should be backported.

Closing this issue here, because it is not a bug in this repo and the GCC issue is fixed upstream.

closed 2024-05-31 by cmuellner