riscv-collab/riscv-gnu-toolchain #1483: build error on MacOS Monterey on m1 chip

wuyuedefeng, 2024-05-23:
```bash
../.././gcc/gcc/gengtype.h:508:40: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
../.././gcc/gcc/gengtype.h:508:40: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
../.././gcc/gcc/gengtype.h:508:40: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
20 warnings generated.
g++ -std=gnu++11   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE  -o build/gengtype \
	    build/gengtype.o build/errors.o build/gengtype-lex.o build/gengtype-parse.o build/gengtype-state.o ../build-aarch64-apple-darwin21.6.0/libiberty/libiberty.a
LC_ALL=C ; export LC_ALL ; \
	/opt/homebrew/bin/gawk -f ../.././gcc/gcc/opt-gather.awk ../.././gcc/gcc/ada/gcc-interface/lang.opt ../.././gcc/gcc/d/lang.opt ../.././gcc/gcc/fortran/lang.opt ../.././gcc/gcc/go/lang.opt ../.././gcc/gcc/lto/lang.opt ../.././gcc/gcc/m2/lang.opt ../.././gcc/gcc/rust/lang.opt ../.././gcc/gcc/c-family/c.opt ../.././gcc/gcc/common.opt ../.././gcc/gcc/params.opt ../.././gcc/gcc/analyzer/analyzer.opt ../.././gcc/gcc/config/riscv/riscv.opt > tmp-optionlist
/bin/sh ../.././gcc/gcc/../move-if-change tmp-optionlist optionlist
echo timestamp > s-options
/opt/homebrew/bin/gawk -f ../.././gcc/gcc/opt-functions.awk -f ../.././gcc/gcc/opt-read.awk \
	       -f ../.././gcc/gcc/opth-gen.awk \
	       < optionlist > tmp-options.h
/bin/sh ../.././gcc/gcc/../move-if-change tmp-options.h options.h
echo timestamp > s-options-h
g++ -std=gnu++11 -c   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -I. -Ibuild -I../.././gcc/gcc -I../.././gcc/gcc/build -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include  \
		-o build/genconstants.o ../.././gcc/gcc/genconstants.cc
warning: unknown warning option '-Wconditionally-supported' [-Wunknown-warning-option]
1 warning generated.
g++ -std=gnu++11   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE  -o build/genconstants \
	    build/genconstants.o build/read-md.o build/errors.o ../build-aarch64-apple-darwin21.6.0/libiberty/libiberty.a
build/genconstants ../.././gcc/gcc/common.md ../.././gcc/gcc/config/riscv/riscv.md \
	   > tmp-constants.h
/bin/sh ../.././gcc/gcc/../move-if-change tmp-constants.h insn-constants.h
echo timestamp > s-constants
g++ -std=gnu++11 -c   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-strict-aliasing -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -I. -Ibuild -I../.././gcc/gcc -I../.././gcc/gcc/build -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include  \
		-o build/genrvv-type-indexer.o ../.././gcc/gcc/config/riscv/genrvv-type-indexer.cc
warning: unknown warning option '-Wconditionally-supported' [-Wunknown-warning-option]
../.././gcc/gcc/config/riscv/genrvv-type-indexer.cc:118:30: error: no member named 'log2' in namespace 'std'; did you mean simply 'log2'?
    elmul_log2 = lmul_log2 - std::log2 (sew / eew);
                             ^~~~~~~~~
                             log2
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/math.h:1463:1: note: 'log2' declared here
log2(_A1 __lcpp_x) _NOEXCEPT {return ::log2((double)__lcpp_x);}
^
../.././gcc/gcc/config/riscv/genrvv-type-indexer.cc:120:30: error: no member named 'log2' in namespace 'std'; did you mean simply 'log2'?
    elmul_log2 = lmul_log2 + std::log2 (eew / sew);
                             ^~~~~~~~~
                             log2
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/math.h:1463:1: note: 'log2' declared here
log2(_A1 __lcpp_x) _NOEXCEPT {return ::log2((double)__lcpp_x);}
^
1 warning and 2 errors generated.
make[2]: *** [build/genrvv-type-indexer.o] Error 1
make[1]: *** [all-gcc] Error 2
make: *** [stamps/build-gcc-newlib-stage1] Error 2
```

TommyMurphyTM1234, 2024-05-23:
Anything useful in these previous macOS build related issues? 

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues?q=is%3Aissue+macos

It would be useful if somebody with macOS hardware access could step up and volunteer to investigate and resolve these sort of build issues as many of us don't have such access. 

macOS and its development environment(s?) seem to have its own particular quirks and dependencies that can trip people up when they're trying to build the `riscv-gnu-toolchain` tools.

TommyMurphyTM1234, 2024-05-24:
I'm pretty sure that this is an issue with the setup of the macOS development environment that you're using and not with the `riscv-gnu-toolchain` repo per se. 

You might want to review the macOS build environment dependencies here:

* https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#getting-the-sources
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1332
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues?q=is%3Aissue+macos+dependencies

And maybe check the version of the toolchain and headers that you are using to build the RISC-V toolchain here?
I wonder if this warning is a hint that the toolchain that you're using is not suitable?
> ```
> warning: unknown warning option '-Wconditionally-supported' [-Wunknown-warning-option]
> ```
It might also be worth asking about this on a more macOS specific forum?

px341, 2024-05-24:
[error.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/15435376/error.log)
I also encountered the similar problem, maybe it is more smart for me as a green hand to use ubuntu to avoid the source building.

TommyMurphyTM1234, 2024-05-24:
> [error.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/15435376/error.log) I also encountered the similar problem, maybe it is more smart for me as a green hand to use ubuntu to avoid the source building.

Do you actually need to build the toolchain from sources at all or would a binary build meet your needs? For example:

* the regular `riscv-gnu-toolchain` GitHub Actions CI builds:
  * https://github.com/riscv-collab/riscv-gnu-toolchain/releases
* the xPack builds:
  * https://xpack.github.io/dev-tools/riscv-none-elf-gcc/
    * https://xpack.github.io/dev-tools/riscv-none-elf-gcc/releases/
* the Embecosm builds:
  * https://www.embecosm.com/resources/tool-chain-downloads/
* there may be others too?

px341, 2024-05-25:
> Do you actually need to build the toolchain from sources at all or would a binary build meet your needs? For example:
>
> * the regular `riscv-gnu-toolchain` GitHub Actions CI builds:
>
>   * [...]

I prefer the binary release. Thank you for your suggestion.

ihiasi, 2024-06-24:
FYI, this error is not happening anymore after upgrading to macOS Sonoma and Xcode 15.2 (15.4 has other compilation issues).

TommyMurphyTM1234, 2024-06-24:
Thanks for the update @ihiasi.
Since, as I hypothesised earlier, this seems to be an issue with the macOS tools/development environment, and not with `riscv-gnu-toolchain` itself, I'll close the issue.

closed 2024-06-24 by TommyMurphyTM1234