riscv-collab/riscv-gnu-toolchain #1485: fails build with 'configure: error'

accesSudo, 2024-06-01:
I have searched through a multitude of other similarily listed issues and it seems like none of the solutions have been able to work. I downloaded and was able to succesfully run the toolchain initially, but after I added one new custom instruction it seems as though the toolchain will refuse to recompile as I exit out with this error constantly:

checking for suffix of object files... configure: error: in `/home/ulx3/projects/rv32i/sw/toolchain/build-gcc-linux-stage1/riscv32-unknown-linux-gnu/libgcc':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make[1]: *** [Makefile:12811: configure-target-libgcc] Error 1
make[1]: Leaving directory '/home/ulx3/projects/rv32i/sw/toolchain/build-gcc-linux-stage1'
make: *** [Makefile:455: stamps/build-gcc-linux-stage1] Error 2

Just to make it clear I have done the following solutions:
deleted and reinstalled a new virtual machine (Ubuntu)
deleted and did a fresh install of the riscv-gnu-toolchain
I make sure to delete the previosu prefix dir before each run so each run uses a clean unusued dir
I run make distclean and make clean before every new run
I do not change config arguements, and always call only "make linux" to build

The exact changes I have made are as follows (the functionality i am seeking is to add sin(x)):
in toolchain/gdb/opcodes/riscv-opc.c (same changes in toolchain/binutils/opcodes/riscv-opc.c)
{"fsin.s",     0, INSN_CLASS_F_INX,   "d,s",   	   MATCH_FSIN, MASK_FSIN, match_opcode, 0 }

in toolchain/gdb/include/opcode.riscv-opc.h (same changes in toolchain/binutils/opcodes/riscv-opc.c))
#define MATCH_FSIN   0x30000053
#define MASK_FSIN    0xfff0007f
DECLARE_INSN(fsin_s, MATCH_FSIN, MASK_FSIN)

After cloning the repo into the toolchain directory and creating a empty bin and build dir I run this config path:
./configure --prefix=/home/ulx3/build --with-arch=rv32gc --with-abi=ilp32d
make linux
I am aiming to work with the rv32i emulator so I need to stay with the 32 bit build

CONFIG LOG:

This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by riscv-toolchain configure 1.0, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ./configure --prefix=/home/ulx3/build --with-arch=rv32gc --with-abi=ilp32d --disable-float --disable-atomic

## --------- ##
## Platform. ##
## --------- ##

hostname = xx
uname -m = x86_64
uname -r = 5.15.0-105-generic
uname -s = Linux
uname -v = #115-Ubuntu SMP Mon Apr 15 09:52:04 UTC 2024

/usr/bin/uname -p = x86_64
/bin/uname -X     = unknown

/bin/arch              = x86_64
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /home/ulx3/build/bin
PATH: /home/ulx3/build/bin
PATH: /home/ulx3/bin
PATH: /home/ulx3/bin
PATH: /home/.local/bin
PATH: /opt/xilinx/xrt/bin
PATH: /mnt/ssd0/Xilinx/Vitis_HLS/2021.2/bin
PATH: /mnt/ssd0/Xilinx/Model_Composer/2021.2/bin
PATH: /mnt/ssd0/Xilinx/Vitis/2021.2/bin
[... 289 lines cut ...]
with_linux_headers_src='$(srcdir)/linux-headers/include'
with_llvm_src='$(srcdir)/llvm'
with_musl_src='$(srcdir)/musl'
with_newlib_src='$(srcdir)/newlib'
with_pk_src='$(srcdir)/pk'
with_qemu_src='$(srcdir)/qemu'
with_spike_src='$(srcdir)/spike'
with_system_zlib='--with-system-zlib'

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "riscv-toolchain"
#define PACKAGE_TARNAME "riscv-toolchain"
#define PACKAGE_VERSION "1.0"
#define PACKAGE_STRING "riscv-toolchain 1.0"
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL ""
#define HAVE_LIBGMP 1
#define HAVE_LIBMPFR 1
#define HAVE_LIBMPC 1

configure: exit 0

TommyMurphyTM1234, 2024-06-01:
I have kicked off a build to see if I can reproduce this.

However, one tangential thing to note:

> After cloning the repo into the toolchain directory and creating a empty bin and build dir I run this config path:
> ./configure --prefix=/home/ulx3/build --with-arch=rv32gc --with-abi=ilp32d
> make linux
> I am aiming to work with the rv32i emulator so I need to stay with the 32 bit build

If you're targeting `rv32i/ilp32` then an `rv32gc/ilp32d` toolchain is probably not appropriate as, by default (e.g. without explicitly specifying `-march=rv32i -mabi=ilp32`), it will generate instructions not supported by the former `arch/abi` - e.g. instructions from some or all of the `MAFDC` extensions. And I don't think that running Linux on `rv32i/ilp32` is easy if it's possible at all. I think you really need at least `rv32gc/ilp32d`.

TommyMurphyTM1234, 2024-06-01:
I suspect that **_this_** is incorrect?

> The exact changes I have made are as follows (the functionality i am seeking is to add sin(x)):
> in toolchain/gdb/opcodes/riscv-opc.c (same changes in toolchain/binutils/opcodes/riscv-opc.c) [...]

Ignoring GDB for the moment, isn't it the case that the following changes need to be made:
```
// Add this to binutils/opcodes/riscv-opc.c
{"fsin.s", 0, INSN_CLASS_F_INX, "d,s", MATCH_FSIN, MASK_FSIN, match_opcode, 0 }

// Add this to binutils/include/opcode/riscv-opc.h
#define MATCH_FSIN 0x30000053
#define MASK_FSIN 0xfff0007f
DECLARE_INSN(fsin_s, MATCH_FSIN, MASK_FSIN)
```

TommyMurphyTM1234, 2024-06-01:
OK - I can reproduce the issue but it seems to be down to some problem with the addition of the custom instruction - something wrong with the opcode and/or mask used?
This is what I see in ` build-gcc-linux-stage1/riscv32-unknown-linux-gnu/libgcc/config.log` after the failure:
```
configure:3792: checking for suffix of object files                                                                                                                                         configure:3814:  /home/user/Downloads/issue-1485/build-gcc-linux-stage1/./gcc/xgcc -B/home/user/Downloads/issue-1485/build-gcc-linux-stage1/./gcc/ -B/home/user/Downloads/issue-1485/installed-tools/riscv32-unknown-linux-gnu/bin/ -B/home/user/Downloads/issue-1485/installed-tools/riscv32-unknown-linux-gnu/lib/ -isystem /home/user/Downloads/issue-1485/installed-tools/riscv32-unknown-linux-gnu/include -isystem /home/user/Downloads/issue-1485/installed-tools/riscv32-unknown-linux-gnu/sys-include    -c -O2    -mcmodel=medlow  conftest.c >&5
Assembler messages:
Error: internal: bad RISC-V opcode (bits 0x7000 undefined or invalid): fsin.s d,s
Fatal error: internal: broken assembler.  No assembly attempted
Internal error (Segmentation fault).
Please report this bug.
configure:3818: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "GNU C Runtime Library"
| #define PACKAGE_TARNAME "libgcc"
| #define PACKAGE_VERSION "1.0"
| #define PACKAGE_STRING "GNU C Runtime Library 1.0"
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL "http://www.gnu.org/software/libgcc/"
| /* end confdefs.h.  */
|
| int
| main ()
| {
|
|   ;
|   return 0;
| }
configure:3832: error: in `/home/user/Downloads/issue-1485/build-gcc-linux-stage1/riscv32-unknown-linux-gnu/libgcc':
configure:3834: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
```

TommyMurphyTM1234, 2024-06-01:
> OK - I can reproduce the issue but it seems to be down to some problem with the addition of the custom instruction - something wrong with the opcode and/or mask used perhaps? 

Can you explain the rationale behind your choice of instruction opcode and mask here?
```
#define MATCH_FSIN 0x30000053
#define MASK_FSIN 0xfff0007f
DECLARE_INSN(fsin_s, MATCH_FSIN, MASK_FSIN)
```
And how you arrived at the aforementioned changes to the toolchain source files for adding a custom floating point instruction?

accesSudo, 2024-06-01:
Hello Tommy appreciate your prompt reply.

The way I arrived at the specific match and mask values was by utilizing the riscv-tools repo. 

First I went to riscv-tools/riscv-opcodes/opcodes where I inputted this (using other floating point instructions as a reference):
fsin.s   rd rs1 24..20=0 31..27=0x06 rm       26..25=0 6..2=0x14 1..0=3

From there I called:
cat opcodes | ./parse-opcodes -c > temp.h

Which is what gave me the auto generated mask and match values. Is that wrong? Is there someother way to get match and mask values? I apologize if that is a low level question but I am extremely new to RISCV.

TommyMurphyTM1234, 2024-06-01:
> Is that wrong? Is there someother way to get match and mask values? 

Sorry, I don't know enough about adding custom floating point instructions to the toolchain to advise. But I guess that there's something wrong with your first attempt above given that the toolchain builds fine on a clean clone from the repo but fails when you make your custom `fsin` instruction changes to the sources.

accesSudo, 2024-06-02:
Oh thank you, do you perhaps know what github repo would have better resources in terms of adding custom floating point instructions? like riscv-tools perhaps?

TommyMurphyTM1234, 2024-06-02:
> Oh thank you, do you perhaps know what github repo would have better resources in terms of adding custom floating point instructions? 

Unfortunately I don't. You could maybe ask on a GCC mailing list why your changes trigger a build error. 

* https://gcc.gnu.org/lists.html

> like riscv-tools perhaps?

That repo is archived/deprecated:

* https://github.com/riscv-software-src/riscv-tools

accesSudo, 2024-06-03:
Sure thank you one last question, what file did you find this error message in (and where was it located):

Error: internal: bad RISC-V opcode (bits 0x7000 undefined or invalid): fsin.s d,s
Fatal error: internal: broken assembler.  No assembly attempted

TommyMurphyTM1234, 2024-06-03:
> Sure thank you one last question, what file did you find this error message in (and where was it located):
> 
> Error: internal: bad RISC-V opcode (bits 0x7000 undefined or invalid): fsin.s d,s Fatal error: internal: broken assembler. No assembly attempted

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1485#issuecomment-2143445911

> This is what I see in `build-gcc-linux-stage1/riscv32-unknown-linux-gnu/libgcc/config.log` after the failure:
> ...

This is the log file that the original error is pointing to:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1485#issue-2329062801

> ```
> checking for suffix of object files... configure: error: in /home/ulx3/projects/rv32i/sw/toolchain/build-gcc-linux-stage1/riscv32-unknown-linux-gnu/libgcc': configure: error: cannot compute [...]

accesSudo, 2024-06-03:
Ah, I see I must have overlooked that comment.

Thank you, I was able to figure out how to add a floating point instruction as well!

TommyMurphyTM1234, 2024-06-03:
> Ah, I see I must have overlooked that comment.
> 
> Thank you, I was able to figure out how to add a floating point instruction as well!

Can you clarify what you had to change in your original attempt on order to address the problem as it may help others in the same position in the future?

accesSudo, 2024-06-03:
Oh yeah sure, for potential future readers I do not know if this is 100% the right thing to do because I have yet to add sin to an emulator to fully test it, however the changes I made did let me successfully recompile the gnu-toolchain.

First I went into riscv-tools/riscv-opcodes/opcodes-custom
From there I *assumed* the original creator created basic templates that could be used to create custom instructions, I used:
custom rd rs1 imm12 14..12=0 6..2=0x02 1..0=3 since I was just going to load an immediate value into sin(x) and did not need source registers

Then I ran: cat opcodes-custom | ./parse-opcodes -c > temp.h

From which I got:
#define MATCH_CUSTOM 0xb
#define MASK_CUSTOM  0x707f
DECLARE_INSN(custom, MATCH_CUSTOM, MASK_CUSTOM)

I plugged those above values into:
toolchain/gdb/include/opcode.riscv-opc.h &  toolchain/binutils/opcodes/include/riscv-opc.h

This next part is what I am a little unsure of, to make sure it was a floating point instruction in
toolchain/gdb/opcode.riscv-opc.c &  toolchain/binutils/opcodes/riscv-opc.c
{"custom",        0, INSN_CLASS_D_INX,   "d,a",       MATCH_CUSTOM, MASK_CUSTOM, match_opcode, 0 } 

Rebuilding this gave me no errors (keep in mind I have yet to add it to an emulator so I can not say for certain it works)

TommyMurphyTM1234, 2024-06-04:
> Rebuilding this gave me no errors (keep in mind I have yet to add it to an emulator so I can not say for certain it works)

If the modified toolchain can generate your custom instruction (e.g. using [GCC inline assembly](https://gcc.gnu.org/onlinedocs/gcc/Using-Assembly-Language-with-C.html)) when you compile a program then you have presumably made the correct changes to the toolchain. Have you checked this?

accesSudo, 2024-06-04:
Hi Tommy,

So first by inline assembly do you mean compiling a c program using: riscv32-unknown-linux-gnu-gcc?

If so then yes I was able to do this, coverting a .c file to a .o and a .dump file (using riscv32-unknown-elf-objdump).

This is the c file that I was using is calling this to test custom:
float custom_sin() {
    float fact_result_ptr;

    asm volatile("custom %0, %1\n":"=r"(fact_result_ptr):"i"(30):);

    return fact_result_ptr;
}

As for changing fsin.s -> custom I just did not want to change the name but it should be fine to change custom -> fsin.s

As for changing INSN_CLASS_F_INX -> INSN_CLASS_D_INX I noticed in opcodes/risc-opc.c I noticed that INSN_CLASS_F_INX was used under /* Single-precision floating-point instruction subset.  */    while
INSN_CLASS_D_INX was used under  /* Double-precision floating-point instruction subset.  */ and I wanted to use double precision instead of single

TommyMurphyTM1234, 2024-06-04:
> So first by inline assembly do you mean compiling a c program using: riscv32-unknown-linux-gnu-gcc?
>
> If so then yes I was able to do this, coverting a .c file to a .o and a .dump file (using [...]

That being the case it sounds like your toolchain changes to add your custom instruction were successful.

> As for changing fsin.s -> custom I just did not want to change the name but it should be fine to change custom -> fsin.s
> 
> As for changing INSN_CLASS_F_INX -> INSN_CLASS_D_INX I noticed in opcodes/risc-opc.c that INSN_CLASS_F_INX was used under / * Single-precision floating-point instruction subset. */ while on the other hand INSN_CLASS_D_INX was used under / * Double-precision floating-point instruction subset. */ and I wanted to use double precision instead of single

OK - thanks.

I will close this again since the original issue has been addressed.

closed 2024-06-04 by TommyMurphyTM1234