riscv-collab/riscv-gnu-toolchain #1487: How to add extra newlib flags when compiling

mateoconlechuga, 2024-06-05:
I would like to add the newlib option `--enable-newlib-use-gdtoa=no` when compiling newlib. Is there a way to do this from the top level configure or makefile of this repository?

cmuellner, 2024-06-05:
For GCC, we have `GCC_EXTRA_CONFIGURE_FLAGS`, which can be set to add configure flags.

No such variable exists for other toolchain components.
But PRs that introduce that would be very welcome.

mateoconlechuga, 2024-06-05:
Can I not use one of `BINUTILS_TARGET_FLAGS_EXTRA` or `BINUTILS_NATIVE_FLAGS_EXTRA`?

TommyMurphyTM1234, 2024-06-05:
> Can I not use one of `BINUTILS_TARGET_FLAGS_EXTRA` or `BINUTILS_NATIVE_FLAGS_EXTRA`?

Have you tried? 
Last time I was trying to do this it didn't work. 

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1187#issuecomment-1427037219

cmuellner, 2024-06-05:
> Can I not use one of `BINUTILS_TARGET_FLAGS_EXTRA` or `BINUTILS_NATIVE_FLAGS_EXTRA`?

Indeed. But unfortunately, no variable exists for newlib.

closed 2024-06-06 by mateoconlechuga

cmuellner, 2024-06-06:
@mateoconlechuga, you closed this ticket, so I assumed you are not interested in preparing a PR for the missing feature.

I went ahead and prepared a PR (see #1489). It would be great if you could check if that works for you.

PR #1489 (2024-06-26): Add NEWLIB_TARGET_FLAGS_EXTRA
This adds NEWLIB_TARGET_FLAGS_EXTRA, which works the same like GLIBC_TARGET_FLAGS_EXTRA. The flags are used for configuring newlib or newlib-nano.

Fixes #1487.