riscv-collab/riscv-gnu-toolchain #1488: Doubts on the built toolchain

s0umyajeet, 2024-06-06:
I recently built and new toolchain with GCC source as GCC-14 and binutils source as corev-binutils-gdb. I'm seeing a situation which I'm not sure if I fully understand.  I'm using GCC-14 to test out latest features and I had to change binutils because it doesn't support zc extensions. When I configure using the following flags:

`./configure --prefix=/development/build_zc --with-arch=rv32im_zca_zcb_zcmp --with-abi=ilp32 --enable-multilib --with-gcc-src=/development/riscv-gnu-toolchain/gcc-14 --with-binutils-src=/development/riscv-gnu-toolchain/corev-binutils-gdb`

I see the following being generated:
rv32im, rv32imac, rv32imafc under ```build_zc/lib/gcc/riscv32-unkonwn-elf/14.1.1```
 
My question beings:
1. Why are the other two (rv32imac & rv32imacfc) getting generated when the flag being used here is just rv32im
2. Similar to this, why are rv32i_zcb, rv32i_zca etc. not being generated in the folder
3. does the order of passing the flags in the configure command matter or alter the generation in anyway?

Kindly excuse naive questions, I'm just getting started. Any inputs help. Thanks!

TommyMurphyTM1234, 2024-06-06:
> corev-binutils-gdb

What is this? It's not part of the `riscv-gnu-toolchain` repo.

> I had to change binutils because it doesn't support zc extensions.

Perhaps you need to build with the master Binutils sources (pre-2.43 release)?

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1472#issuecomment-2143520816
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc -b releases/gcc-14 gcc-14
git clone https://github.com/bminor/binutils-gdb binutils-master
./configure ... --with-gcc-src=`pwd`/gcc-14 --with-binutils-src=`pwd`/binutils-master
make
...
```

> Why are the other two (rv32imac & rv32imacfc) getting generated when the flag being used here is just rv32im

Because you have specified `--enable-multilib`

* https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#installation-newliblinux-multilib

which, by default, builds these multilibs:

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/6d7b5b720f6c6a89b4f22c62b22c1d0ebc2dc353/configure.ac#L160-L162

> Similar to this, why are rv32i_zcb, rv32i_zca etc. not being generated in the folder

Because you haven't specified that they need to be built. If you want them then you probably need to use `--with-multilib-generator` to specify your required custom set of multilibs:

* https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#build-with-customized-multi-lib-configure

> does the order of passing the flags in the configure command matter or alter the generation in anyway?

No.

s0umyajeet, 2024-06-06:
Thank you for the awesome answer!!!!

s0umyajeet, 2024-06-12:
I tried building toolchain with the configure command 
`./configure --prefix=/4docu/rv32im_zca_zcb_zcmp --with-arch=rv32im_zca_zcb_zcmp_zicsr --with-abi=ilp32 --with-gcc-src=/4docu/riscv-gnu-toolchain/gcc-14 --with-binutils-src=/4docu/riscv-gnu-toolchain/corev-binutils-gdb --with-multilib-generator="rv32im_zca_zcb_zcmp_zicsr-ilp32--"`
but when I go into the directory `rv32im_zca_zcb_zcmp/lib/gcc/riscv32-unkonwn-elf/14.1.1` I don't see any multilib `rv32im_zca_zcb_zcmp` and actually no multilib.

After this I configured using the command:
`./configure --prefix=/4docu/riscvgcc_gcc14_corevbin_zc_multilibgen --with-arch=rv32im_zca_zcb_zcmp_zicsr --with-abi=ilp32 --with-gcc-src=/4docu/riscv-gnu-toolchain/gcc-14 --with-binutils-src=/4docu/riscv-gnu-toolchain/corev-binutils-gdb --with-multilib-generator="rv32i_zca_zcb_zcmp_zicsr-ilp32--"` 
I can see multilib `rv32i_zicsr_zca_zcb_zcmp`. 

I don't understand what's wrong

TommyMurphyTM1234, 2024-06-12:
> I tried building toolchain with the configure command `--with-arch=rv32im_zca_zcb_zcmp_zicsr --with-abi=ilp32 ... --with-multilib-generator="rv32im_zca_zcb_zcmp_zicsr-ilp32--"` but when I go into the directory `rv32im_zca_zcb_zcmp/lib/gcc/riscv32-unkonwn-elf/14.1.1` I don't see any multilib `rv32im_zca_zcb_zcmp` and actually no multilib.

You've asked it to build a toolchain with the default `arch/abi` of `rv32im_zca_zcb_zcmp_zicsr/ilp32` and also a multilib with the same `arch/abi`. It's pointless building this twice so it just gets built once and installed in `<prefix-dir>/lib` and not also in the multilib lib directory.

> After this I configured using the command: `--with-arch=rv32im_zca_zcb_zcmp_zicsr --with-abi=ilp32 ... --with-multilib-generator="rv32i_zca_zcb_zcmp_zicsr-ilp32--"` I can see multilib `rv32i_zicsr_zca_zcb_zcmp`.

That's correct because it has built the default toolchain (multi)lib of `rv32im_zca_zcb_zcmp_zicsr/ilp32` and installed it in `<prefix-dir>/lib` and also the multilib `rv32i_zca_zcb_zcmp_zicsr.ilp32` and installed that in the multilib directory.

> I don't understand what's wrong

There's nothing wrong. This is all working as expected.

TommyMurphyTM1234, 2024-06-12:
> `riscv32-unknown-elf-gcc -print-multi-lib` will show you the (multi)libs supported by the toolchain with the first `.;` representing the toolchain's default `arch/abi` (multi)lib. `riscv32-unknown-elf-gcc -v` will tell you what default `arch/abi` the toolchain was configured with.

For example...
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
git clone https://github.com/gcc-mirror/gcc -b releases/gcc-14 gcc-14
git clone https://github.com/openhwgroup/corev-binutils-gdb

./configure --prefix=`pwd`/installed-tools --with-arch=rv32im_zca_zcb_zcmp_zicsr --with-abi=ilp32 --with-gcc-src=`pwd`/gcc-14 --with-binutils-src=`pwd`/corev-binutils-gdb --with-multilib-generator="rv32im_zca_zcb_zcmp_zicsr-ilp32--" --disable-gdb
make
...

./installed-tools/bin/riscv32-unknown-elf-gcc -print-multi-lib
.;

./installed-tools/bin/riscv32-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./installed-tools/bin/riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/Downloads/issue-1488/installed-tools/libexec/gcc/riscv32-unknown-elf/14.1.1/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /home/user/Downloads/issue-1488/gcc-14/configure --target=riscv32-unknown-elf --prefix=/home/user/Downloads/issue-1488/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=g9100e78ba28 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/Downloads/issue-1488/installed-tools/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=/home/user/Downloads/issue-1488/gcc-14 --enable-multilib --with-multilib-generator=rv32im_zca_zcb_zcmp_zicsr-ilp32-- --with-abi=ilp32 --with-arch=rv32im_zca_zcb_zcmp_zicsr --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 14.1.1 20240612 (g9100e78ba28)
```
And then...
```
make distclean && rm -rf installed-tools

./configure --prefix=`pwd`/installed-tools --with-arch=rv32im_zca_zcb_zcmp_zicsr --with-abi=ilp32 --with-gcc-src=`pwd`/gcc-14 --with-binutils-src=`pwd`/corev-binutils-gdb --with-multilib-generator="rv32i_zca_zcb_zcmp_zicsr-ilp32--" --disable-gdb
make
...

./installed-tools/bin/riscv32-unknown-elf-gcc -print-multi-lib
.;
rv32i_zicsr_zca_zcb_zcmp/ilp32;@march=rv32i_zicsr_zca_zcb_zcmp@mabi=ilp32

./installed-tools/bin/riscv32-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./installed-tools/bin/riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/Downloads/issue-1488/installed-tools/libexec/gcc/riscv32-unknown-elf/14.1.1/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /home/user/Downloads/issue-1488/gcc-14/configure --target=riscv32-unknown-elf --prefix=/home/user/Downloads/issue-1488/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=g9100e78ba28 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/Downloads/issue-1488/installed-tools/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=/home/user/Downloads/issue-1488/gcc-14 --enable-multilib --with-multilib-generator=rv32i_zca_zcb_zcmp_zicsr-ilp32-- --with-abi=ilp32 --with-arch=rv32im_zca_zcb_zcmp_zicsr --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 14.1.1 20240612 (g9100e78ba28)
```

closed 2024-06-13 by s0umyajeet