riscv-collab/riscv-gnu-toolchain #1490: libcrypt.h missing from `sysroot/usr/include/` after otherwise successful build

gsomlo, 2024-06-08:
I build my own riscv-gnu-toolchain from this repo's sources, like so:

```
   git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
   pushd riscv-gnu-toolchain
   ./configure --prefix=$HOME/RISCV --enable-multilib
   make newlib linux
   popd
```

In the past, I used to get `~/RISCV/sysroot/usr/include/crypt.h` but with the latest build (based on commit 6d7b5b7) I no longer get that file generated during build. It's a problem when trying to build a cross-compiled version of the latest busybox for rv64gc.

Any idea how I might update my build flow, or whether this may be an actual regression ?

Thanks much!

TommyMurphyTM1234, 2024-06-08:
Any use?> 

* https://lists.gnu.org/archive/html/info-gnu/2024-01/msg00017.html

> libcrypt has been removed from the GNU C Library.  The configure
>   options "--enable-crypt" and "--enable-nss-crypt" are no longer
>   available.  <crypt.h>, libcrypt.a, and libcrypt.so.1 will not be
>   installed.  For now <unistd.h> continues to declare the crypt
>   function by default, to avoid introducing vulnerabilities into
>   existing applications due to a missing prototype.  This declaration
>   is deprecated and may be removed in a future glibc release.
> 
>   The replacement for libcrypt is libxcrypt, maintained separately from
>   GNU libc, but available under compatible licensing terms, and providing
>   binary backward compatibility with the former libcrypt.  It is currently
>   distributed from <https://github.com/besser82/libxcrypt/>.

gsomlo, 2024-06-08:
Thanks, that helps explain things.

Is there a known/documented way to get libxcrypt worked into riscv-gnu-toolchain, to still be able to provide the functionality during cross-compilation of programs that expect it?

Alternatively, I'll have to figure out if there's a way I can disable (and live without) the portion of busybox that still relies on libcrypt.h...

Or maybe the "canonical" answer is to stick with an older version of riscv-gnu-toolchain until all the programs I need to build have stopped requiring libcrypt?

Any advice/thoughts/pointers much appreciated!

jrtc27, 2024-06-08:
Cross-compile it with --prefix= like anything else?

TommyMurphyTM1234, 2024-06-08:
> Thanks, that helps explain things.
> 
> Is there a known/documented way to get libxcrypt worked into riscv-gnu-toolchain, to still be able to provide the functionality during cross-compilation of programs that expect it?

Have you tried just building it for RISC-V or, if that doesn't work straight off, checking or posting in the `libxcrypt` repo/forums in case somebody else has tackled this? 

* https://github.com/besser82/libxcrypt

> Or maybe the "canonical" answer is to stick with an older version of riscv-gnu-toolchain until all the programs I need to build have stopped requiring libcrypt?

This isn't really a RISC-V specific question/issue so, again, there may be some guidance in other forums such as `libcrypt`, `libxxrypt`, Glibc, GCC etc.?

cmuellner, 2024-06-08:
> Alternatively, I'll have to figure out if there's a way I can disable (and live without) the portion of busybox that still relies on libcrypt.h...

If upstream Busybox relies on libcrypt, then this needs to be addressed (upstream).

Please check if the Busybox developers are aware of that. If so, a fix is probably already available. If not, please support them in working on one.

gsomlo, 2024-06-08:
Thanks everyone, I think I have enough context and situational awareness to start making useful progress from here on out! (including, if needed, notifying/patching upstream busybox).

closed 2024-06-08 by gsomlo

gsomlo, 2024-06-09:
On Sat, Jun 08, 2024 at 08:42:36AM -0700, Christoph Müllner wrote:
> If upstream Busybox relies on libcrypt, then this needs to be addressed
> (upstream).

For "posterity" (i.e., for anyone happening to find this via search
engine): the answer is to enable `CONFIG_USE_BB_CRYPT=y` in busybox's
.config before building, to prevent it from attempting to use
(the now unavailable) glibc libcrypt.

cmuellner, 2024-06-09:
Thanks for providing that information!