riscv-collab/riscv-gnu-toolchain #1491: Is bfloat16 supported in gcc or clang

fanghuaqi, 2024-06-18:
Hello, there

The RISC-V bfloat16 ISA, available at https://github.com/riscv/riscv-bfloat16, is already in version 1.0. I noticed that the BF16 format has been ratified for both scalar and vector extensions. I would like to know if gcc or clang currently support this format. If not, are there any patches being developed to add support for it?

Thanks

TommyMurphyTM1234, 2024-06-18:
> The RISC-V bfloat16 ISA, available at https://github.com/riscv/riscv-bfloat16, is already in version 1.0.

This seems to suggest that it's ready for ratification but not actually ratified yet?

* https://jira.riscv.org/browse/RVS-704?src=confmacro

It's not listed on the RISC-V International ratified specs page:

* https://wiki.riscv.org/display/HOME/Ratified+Extensions

I don't see it mentioned in any GCC release notes:

* https://gcc.gnu.org/releases.html

There does seem to be mention of it on the GCC mailing lists - maybe it's work in progress on the/a development branch?

* https://gcc.gnu.org/cgi-bin/search.cgi?q=bfloat16+riscv

I don't know much about LLVM/Clang but it's mentioned here:

* https://muxup.com/2023q4/whats-new-for-risc-v-in-llvm-17

But I don't see it mentioned in the LLVM 17 release notes:

* https://releases.llvm.org/17.0.1/docs/ReleaseNotes.html#changes-to-the-risc-v-backend

Might be worth asking on the relevant GCC and LLVM/Clang mailing lists?

TommyMurphyTM1234, 2024-06-25:
Any update @fanghuaqi?
Did my posts/links help to clarify this at all?
Have you posted to ask about this in the upstream projects?

fanghuaqi, 2024-06-25:
> Any update @fanghuaqi? Did my posts/links help to clarify this at all? Have you posted to ask about this in the upstream projects?

Hi @TommyMurphyTM1234 , I didn't posted to ask in upstream project, I thought this repo will be watched by many people, so just asked here, I will search the patches in https://patchwork.sourceware.org/project/gcc/list/?q=risc-v later

TommyMurphyTM1234, 2024-07-06:
For what it's worth, the RISC-V BF16 extensions were ratified in June 2024:

- https://wiki.riscv.org/plugins/servlet/mobile?contentId=16458762#content/view/16458762

And the GCC documentation suggests that it's supported:

- https://gcc.gnu.org/onlinedocs/gcc/RISC-V-Options.html

But, confusingly, I don't see it mentioned in the release notes for GCC 12.4, which is the only GCC release post dating ratification.

- https://gcc.gnu.org/gcc-12/changes.html

I suspect that this means that (work in progress?) support is only available via the GCC master sources for now and you would have to build the toolchain with those using `configure ... --with-gcc-src=...`?

- https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#use-source-tree-other-than-riscv-gnu-toolchain

TommyMurphyTM1234, 2024-07-25:
Any update @fanghuaqi?

fanghuaqi, 2024-07-25:
I see https://gcc.gnu.org/git/?p=gcc.git;a=commit;h=666f167bec09d1234e6496c86b566fe1a71f61f0 RISC-V: Add vector type of BFloat16 format commit is added into gcc master, we are looking to see whether it matched our requirements.

TommyMurphyTM1234, 2024-07-25:
> I see https://gcc.gnu.org/git/?p=gcc.git;a=commit;h=666f167bec09d1234e6496c86b566fe1a71f61f0 RISC-V: Add vector type of BFloat16 format commit is added into gcc master, we are looking to see whether it matched our requirements.

What specific instruction(s) are you trying to use and how (e.g. inline assembler, compiler inference etc.)?
Can you provide a small self-contained code snippet?

fanghuaqi, 2024-07-25:
> What specific instruction(s) are you trying to use and how (e.g. inline assembler, compiler inference etc.)? Can you provide a small self-contained code snippet?

We just want to use compiler intrinsic API for `bfloat16` just reusing the existing RVV instructions, but we have a CSR register to control whether the RVV instruction is doing float16 or bfloat16 calculation.

TommyMurphyTM1234, 2024-08-01:
Can this issue be closed now @fanghuaqi?

fanghuaqi, 2024-08-01:
Ok, I close it.

closed 2024-08-01 by fanghuaqi