riscv-collab/riscv-gnu-toolchain #1495: extension `zicsr' required

ahmdotm, 2024-06-19:
I built the toolchain using --with-arch=rv64imac_zicsr --with-abi=lp64 
but still got this msg:
asm/printf.c:307: Error: unrecognized opcode `csrr a6,mcycleh', extension `zicsr' required

TommyMurphyTM1234, 2024-06-19:
> I built the toolchain using --with-arch=rv64imac_zicsr --with-abi=lp64 but still got this msg: asm/printf.c:307: Error: unrecognized opcode `csrr a6,mcycleh', extension `zicsr' required

Please provide full details of how you built the toolchain - all of the commands `git clone`, `configure`, `make` - the full build log and details of how you are invoking the compiler to compiler your own code please.

TommyMurphyTM1234, 2024-06-19:
I built the toolchain as follows:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --with-arch=rv64imac_zicsr --with-abi=lp64 --disable-gdb
make 2>&1 | tee build.log
```
And then I compiled this code without any problems:
```
#include <stdio.h>

int main(int argc, char *argv[])
{
    printf("Hello World\n");
    asm("csrr t1, mcycleh");
    return 0;
}
```
```
./installed-tools/bin/riscv64-unknown-elf hello.c -o hello
./installed-tools/bin/riscv64-unknown-objdump -dS hello

...

000000000001019c <main>:
   1019c:       1101                    addi    sp,sp,-32
   1019e:       ec06                    sd      ra,24(sp)
   101a0:       e822                    sd      s0,16(sp)
   101a2:       1000                    addi    s0,sp,32
   101a4:       87aa                    mv      a5,a0
   101a6:       feb43023                sd      a1,-32(s0)
   101aa:       fef42623                sw      a5,-20(s0)
   101ae:       67c9                    lui     a5,0x12
   101b0:       5e078513                addi    a0,a5,1504 # 125e0 <__errno+0xa>
   101b4:       388000ef                jal     1053c <puts>
   101b8:       b8002373                csrr    t1,mcycleh
   101bc:       4781                    li      a5,0
   101be:       853e                    mv      a0,a5
   101c0:       60e2                    ld      ra,24(sp)
   101c2:       6442                    ld      s0,16(sp)
   101c4:       6105                    addi    sp,sp,32
   101c6:       8082                    ret

...
```

TommyMurphyTM1234, 2024-06-20:
Any update @ahmdotm?
There's no reproducible problem here so far.

TommyMurphyTM1234, 2024-06-25:
No update and no reproducible scenario so I am closing this. 
If the issue can be reproduced then please post details and reopen the issue.

closed 2024-06-25 by TommyMurphyTM1234