riscv-collab/riscv-gnu-toolchain #1497: macOS uses .dylib instead of .so

ihiasi, 2024-06-24:
LLVMGold fails to install on macOS : 
```
cp build-llvm-newlib/lib/LLVMgold.so  /Volumes/case_sensitive/install/riscv/lib
cp: build-llvm-newlib/lib/LLVMgold.so: No such file or directory
```
Where is this generated, I could fix it myself ? Thank you.

ihiasi, 2024-06-24:
I workarounded the issue by changing the Makefile directly - gonna check how it is generated and switch the extension based on OS.

TommyMurphyTM1234, 2024-06-24:
What if you configure and build the relevant component(s?) from upstream sources/repos and not via the `riscv-gnu-toolchain` `configure/make` approach? Because the issue may not be with these but with the upstream scripts and, as such, may need to be reported and dealt with there rather than here.

TommyMurphyTM1234, 2024-06-25:
Oh, I guess that the issue is with the `riscv-gnu-toolchain` `Makefile[.in]` after all because this line is hardcoded to assume a `.so` prefix and doesn't parameterise the filename for the macOS scenario?

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/3fb1523f2df35da6b1008f13a214e4b4370f7b5a/Makefile.in#L998

The same issue presumably applies to the Linux toolchain build:

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/3fb1523f2df35da6b1008f13a214e4b4370f7b5a/Makefile.in#L978

I'm not sure what's the best/most appropriate way to do the equivalent of this pseudocode logic in the `Makefile`?
```
if macOS then
    SHARED_LIB_SUFFIX=dylib
else
    SHARED_LIB_SUFFIX=so
endif

...

    cp $(notdir $@)/lib/LLVMgold.$(SHARED_LIB_SUFFIX) $(INSTALL_DIR)/lib
```

TommyMurphyTM1234, 2024-07-06:
Anybody (in particular anybody with access to the relevant macOS development environment(s?) know how to effect the change that I outlined in my previous post, could test it, and could maybe propose a patch/pull request?

rodolphe-c, 2024-08-07:
Hi folks! I have made a PR to fix this issue: https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1528

TommyMurphyTM1234, 2024-08-08:
Addressed by https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1528

closed 2024-08-08 by TommyMurphyTM1234