riscv-collab/riscv-gnu-toolchain #1498: How can I build the toolchain w/o division instructions

im-Mr-Black, 2024-06-25:
I want to build the toolchain w/o any division instructions.

TommyMurphyTM1234, 2024-06-25:
Just build the toolchain as normal but when compiling your own code use the `-mno-div/-mno-fdiv` options to prevent any hardware division instructions from being generated?

- https://gcc.gnu.org/onlinedocs/gcc/RISC-V-Options.html

TommyMurphyTM1234, 2024-06-25:
> Just build the toolchain as normal but when compiling your own code use the `-mno-div/-mno-fdiv` options to prevent any hardware division instructions from being generated?
> 
> * https://gcc.gnu.org/onlinedocs/gcc/RISC-V-Options.html

Actually, if you're linking standard C/C++ libraries/startup code then you'll need to ensure that they also don't contain hardware division instructions. If this isn't the case by default then you may need to do something at toolchain configuration/build time to make this so. E.g. specify `-mno-div/-mno-fdiv` when building Newlib perhaps (I'm assuming that you're using the bare metal and not the Linux toolchain):

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1489

TommyMurphyTM1234, 2024-06-27:
> > Just build the toolchain as normal but when compiling your own code use the `-mno-div/-mno-fdiv` options to prevent any hardware division instructions from being generated?
> > 
> > * https://gcc.gnu.org/onlinedocs/gcc/RISC-V-Options.html
> 
> Actually, if you're linking standard C/C++ libraries/startup code then you'll need to ensure that they also don't contain hardware division instructions. If this isn't the case by default then you may need to do something at toolchain configuration/build time to make this so. E.g. specify `-mno-div/-mno-fdiv` when building Newlib perhaps (I'm assuming that you're using the bare metal and not the Linux toolchain):
> 
> * [Add NEWLIB_TARGET_FLAGS_EXTRA #1489](https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1489)

Sorry - ignore that - `NEWLIB_TARGET_FLAGS_EXTRA` is for passing flags to Newlib's `configure` command and not to the compiler when actually building Newlib. I think what's needed is this:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools
make CFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv" CXXFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv" ASFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv" 2>&1 | tee build.log
```
I checked a "normal" build where no such extra flags are passed and the Newlib libraries such as `libc.a` and `libm.a` do contain hardware integer and floating point division instructions. So, if you're linking Newlib then you need to take the steps above to ensure that no division instructions are used.

TommyMurphyTM1234, 2024-06-27:
> I think what's needed is this:
>
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain
> cd riscv-gnu-toolchain
> ./configure --prefix=`pwd`/installed-tools
> make [...]
I did that and then checked the Newlib libraries for hardware division instructions and they are no longer present:
```
cd installed-tools/riscv64-unknown-elf/lib

../../bin/riscv64-unknown-elf-objdump -d libc.a | grep div
libc_a-div.o:     file format elf64-littleriscv
Disassembly of section .text.div:
0000000000000000 <div>:
  12:   000080e7                jalr    ra # e <div+0xe>
  20:   000080e7                jalr    ra # 1c <div+0x1c>
libc_a-imaxdiv.o:     file format elf64-littleriscv
Disassembly of section .text.imaxdiv:
0000000000000000 <imaxdiv>:
  12:   000080e7                jalr    ra # e <imaxdiv+0xe>
  20:   000080e7                jalr    ra # 1c <imaxdiv+0x1c>
libc_a-ldiv.o:     file format elf64-littleriscv
Disassembly of section .text.ldiv:
0000000000000000 <ldiv>:
  12:   000080e7                jalr    ra # e <ldiv+0xe>
  20:   000080e7                jalr    ra # 1c <ldiv+0x1c>
libc_a-lldiv.o:     file format elf64-littleriscv
Disassembly of section .text.lldiv:
0000000000000000 <lldiv>:
  12:   000080e7                jalr    ra # e <lldiv+0xe>
  20:   000080e7                jalr    ra # 1c <lldiv+0x1c>

../../bin/riscv64-unknown-elf-objdump -d libm.a | grep div
Disassembly of section .text.__math_divzero:
0000000000000000 <__math_divzero>:
   a:   0007b787                fld     fa5,0(a5) # 0 <__math_divzero>
  32:   0007b787                fld     fa5,0(a5) # 0 <__math_divzero>
Disassembly of section .text.__math_divzerof:
0000000000000000 <__math_divzerof>:
   a:   0007a507                flw     fa0,0(a5) # 0 <__math_divzerof>
  16:   000080e7                jalr    ra # 12 <__math_divzerof+0x12>
  26:   00030067                jr      t1 # 22 <__math_divzerof+0x22>
  2e:   0007a507                flw     fa0,0(a5) # 0 <__math_divzerof>

etc.
```
Whereas, without the additional flags we get...
```
../../bin/riscv64-unknown-elf-objdump -d libc.a | grep div
  90:   1aa47453                fdiv.d  fs0,fs0,fa0
libc_a-div.o:     file format elf64-littleriscv
Disassembly of section .text.div:
0000000000000000 <div>:
   0:   02b547bb                divw    a5,a0,a1
  38:   02e7db3b                divuw   s6,a5,a4
 278:   1af6f7d3                fdiv.d  fa5,fa3,fa5
 29a:   1ae7f7d3                fdiv.d  fa5,fa5,fa4
 320:   1ad77753                fdiv.d  fa4,fa4,fa3
 4ee:   1ae7f7d3                fdiv.d  fa5,fa5,fa4
 528:   1ae7f6d3                fdiv.d  fa3,fa5,fa4
libc_a-imaxdiv.o:     file format elf64-littleriscv
Disassembly of section .text.imaxdiv:
0000000000000000 <imaxdiv>:
   4:   02b54533                div     a0,a0,a1
...

 ../../bin/riscv64-unknown-elf-objdump -d libm.a | grep div
  50:   1aa57553                fdiv.d  fa0,fa0,fa0
 12c:   1af57553                fdiv.d  fa0,fa0,fa5
  82:   02f4c4bb                divw    s1,s1,a5
  5a:   1ae7f7d3                fdiv.d  fa5,fa5,fa4
 1ac:   1ae7f7d3                fdiv.d  fa5,fa5,fa4
 20e:   1af777d3                fdiv.d  fa5,fa4,fa5
 25c:   1aa7f7d3                fdiv.d  fa5,fa5,fa0
...

etc.
```
Given that this addresses the original question I am closing the issue.

closed 2024-06-27 by TommyMurphyTM1234