riscv-collab/riscv-gnu-toolchain #1503: multilib-generator flag in CentOS ends up with python issue

RRArinc, 2024-07-04:
Hi,
When enabled the RISC-V toolchain with multilib-generator say for e.g. --with-multilib-generator="rv32i-ilp32--;rv32im-ilp32--;rv32imc-ilp32--;" is provided as part of ./config command, build terminates with error no Python and invalid multilib-generator.

I had same issue with Ubuntu 22. However the same issue was solved using **sudo apt install python-is-python3** package. 
However, similar package is not available with CentOS. How to resolve the issue in CentOS

Thanks

TommyMurphyTM1234, 2024-07-04:
Any use?

- https://stackoverflow.com/questions/45542690/how-to-set-python3-5-2-as-default-python-version-on-centos

RRArinc, 2024-07-04:
Yes, when executed command which python it echos back with below result 
alias python='python3'
        /usr/bin/python3  

But still ends with same error [no python and invalid multilib-generation combination]

Thanks

TommyMurphyTM1234, 2024-07-04:
What version of CentOS?

RRArinc, 2024-07-05:
CentOS8

TommyMurphyTM1234, 2024-07-05:
Perhaps this might work and is the effective equivalent on CentOS 8 of `python-is-python3`?

- https://github.com/wazuh/wazuh/issues/4879#issuecomment-635949993

> **RHEL/CentOS 8**
> 
> Set an alternative command for python to Python2 or Python3:
> ```
> alternatives --set python /usr/bin/python3
> ```

TommyMurphyTM1234, 2024-07-05:
> Perhaps this might work and is the effective equivalent on CentOS 8 of `python-is-python3`?
> 
> * [Various Wazuh Python scripts don't work on RHEL/CentOS 8 (including oscap.py) wazuh/wazuh#4879 (comment)](https://github.com/wazuh/wazuh/issues/4879#issuecomment-635949993)
> 
> > **RHEL/CentOS 8**
> > Set an alternative command for python to Python2 or Python3:
> > ```
> > alternatives --set python /usr/bin/python3
> > ```

I used that command and a build of the toolchain using `--with-multilib-generator=...` completed successfully on CentOS 8.5.
I guess that that command is the CentOS equivalent of `python-is-python3`.

TommyMurphyTM1234, 2024-07-06:
Did `alternatives --set python /usr/bin/python3` also work for you @RRArinc?

TommyMurphyTM1234, 2024-07-07:
OK - I did a clean Centos 8.5 install in a VirtualBox VM, applied all package updates, installed the necessary prerequisite packages for building the `riscv-gnu-toolchain`, and then cloned, configured and built the toolchain. This failed on the multilib stuff.

I then did this:
```
su
alternatives --set python /usr/bin/python3
exit

make distlcean && rm -rf installed-tools
make 2>&1 | tee build.log
```
and this time the build succeeded.

As such I conclude that the original problem has been addressed and so I am closing the issue.

closed 2024-07-07 by TommyMurphyTM1234

RRArinc, 2024-07-08:
Hi,
Yes please close the ticket and it is working as expected.