riscv-collab/riscv-gnu-toolchain #1504: With -mno-div -mno-fdiv enabled still "divu" instruction is seen is objdump file

RRArinc, 2024-07-04:
Hi,
As mentioned in title even after enabling  **`-mno-div and -mno-fdiv`** the libc.a still comprises of "div/divu" instruction is seen.
How to avoid "div" instruction genertaion 

Thanks

TommyMurphyTM1234, 2024-07-04:
I presume that you're using the bare-metal/Newlib toolchain?
In that case you need to ensure that Newlib is built with `-mno-div/-mno-fdiv` and also pass `-mno-div/-mnofdiv` when compiling your own code.

See here:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1498

In particular my post here:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1498#issuecomment-2195530810

RRArinc, 2024-07-04:
Hi, Please find my comments below:
I presume that you're using the bare-metal/Newlib toolchain?
[RESPONSE] Yes, it is bare metal not Linux version. 

In that case you need to ensure that Newlib is built with -mno-div/-mno-fdiv and also pass -mno-div/-mnofdiv when compiling your own code.
[RESPONSE] Yes, indeed i ma passing -mno-div and -mno-fdiv flags while compiling code.

You mentioned to execute make with command 
"make CFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv" CXXFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv" ASFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv". 

However, above command fails with invalid libgloss error. I passed the same to ./configure command build passed but still when application uses <math.h> library to execute sqrt() function i see "divu" instruction.

Thanks.

TommyMurphyTM1234, 2024-07-04:
> Hi, Please find my comments below: I presume that you're using the bare-metal/Newlib toolchain? [RESPONSE] Yes, it is bare metal not Linux version.
>
> In that case you need to ensure that Newlib is [...]

Please follow the **exact** instructions that I posted in the other thread. They work. You may need to do a clean build by running `make distclean` and removing the previously built toolchain directory before running the new build with the additional flags.

If you still have problems then you need to clarify exactly all steps that you are taking, provide logs etc. Otherwise those reading don't have sufficient info to comment meaningfully.

RRArinc, 2024-07-04:
Yes, i did make clean before every build command 

I executed the command 
make CFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv" CXXFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv" ASFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv" and it failed with libgloss error.

RRArinc, 2024-07-04:
With the below command, build terminated with error mentioned below

// Steps:
make clean
./configure <command>
// make command
make CFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv" CXXFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv" ASFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv"

// error termination 
checking whether we are cross compiling... yes
checking for suffix of object files... checking whether we are cross compiling... yes
checking for suffix of object files... configure: error: in `/home/Arr/mnodiv/riscv-gnu-toolchain/build-newlib/riscv32-unknown-elf/newlib':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
configure: error: in `/home/Arr/mnodiv/riscv-gnu-toolchain/build-newlib/riscv32-unknown-elf/libgloss':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
checking whether we are cross compiling... yes
checking for suffix of object files... checking whether we are cross compiling... yes
checking for suffix of object files... configure: error: in `/home/Arr/mnodiv/riscv-gnu-toolchain/build-newlib-nano/riscv32-unknown-elf/libgloss':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make[2]: *** [Makefile:8448: configure-target-newlib] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:9484: configure-target-libgloss] Error 1
configure: error: in `/home/Arr/mnodiv/riscv-gnu-toolchain/build-newlib-nano/riscv32-unknown-elf/newlib':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make[2]: *** [Makefile:9484: configure-target-libgloss] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:8448: configure-target-newlib] Error 1
done

cmuellner, 2024-07-04:
> // error message and termination checking whether we are cross compiling... yes checking for suffix of object files... checking whether we are cross compiling... yes checking for suffix of object files... configure: error: in `/home/Arr/mnodiv/riscv-gnu-toolchain/build-newlib-nano/riscv32-unknown-elf/libgloss': configure: error: cannot compute suffix of object files: cannot compile See `config.log' for more details make[2]: *** [Makefile:8448: configure-target-newlib] Error 1 make[2]: *** Waiting for unfinished jobs.... make[2]: *** [Makefile:9484: configure-target-libgloss] Error 1 configure: error: in `/home/Arr/mnodiv/riscv-gnu-toolchain/build-newlib-nano/riscv32-unknown-elf/newlib': configure: error: cannot compute suffix of object files: cannot compile See `config.log' for more details make[2]: *** [Makefile:9484: configure-target-libgloss] Error 1 make[2]: *** Waiting for unfinished jobs.... make[2]: *** [Makefile:8448: configure-target-newlib] Error 1 done

And what error can be found in the mentioned `config.log`?

cmuellner, 2024-07-04:
I just reproduced here and the exact error message is:
> riscv32-unknown-elf-gcc: error: unrecognized command-line option '-mno=div'; did you mean '-mno-div'?

You set the following variables:
> CFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv" CXXFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv" ASFLAGS_FOR_TARGET_EXTRA="-mno=div -mno-fdiv"

And indeed, there is a typo here (but you copy-pasted it correctly from the referenced source).

So instead, try replacing the `=` by a `-` following:
> CFLAGS_FOR_TARGET_EXTRA="-mno-div -mno-fdiv" CXXFLAGS_FOR_TARGET_EXTRA="-mno-div -mno-fdiv" ASFLAGS_FOR_TARGET_EXTRA="-mno-div -mno-fdiv"

TommyMurphyTM1234, 2024-07-04:
> I just reproduced here and the exact error message is:
>
>
> You set the following variables:
>
>
> And indeed, there is a typo here (but you copy-pasted it correctly from the referenced source).
>
> So [...]

Apologies! I see my typos in the other thread and have corrected them now.

RRArinc, 2024-07-05:
Yes, It is typo. It is working as expected. We can close this ticket. 
Thanks for quick reply.

closed 2024-07-05 by TommyMurphyTM1234