riscv-collab/riscv-gnu-toolchain #1505: riscv64-unknown-linux-gnu-g++: error: '-march=native': ISA string must begin with rv32 or rv64

grigohas, 2024-07-05:
Hello, i am executing this command on ubuntu

TommyMurphyTM1234, 2024-07-05:
Not enough information provided. 
I can only guess that your `Makefile` is passing `-march=native` to the RISC-V compiler and this is not a valid option for this toolchain.
A build log and/or more details about the `Makefile` might help clarify matters.
Almost certainly this is not a `riscv-gnu-tpplchain` problem but something local to your setup.

grigohas, 2024-07-05:
i am trying to run llama scalar on risc v with qemu emulator following these steps

[https://cloud-v.co/blog/risc-v-1/accelerating-llama-cpp-with-risc-v-vector-extension-3]

TommyMurphyTM1234, 2024-07-05:
> i am trying to run llama scalar on risc v with qemu emulator following these steps
> 
> []https://cloud-v.co/blog/risc-v-1/accelerating-llama-cpp-with-risc-v-vector-extension-3

You haven't provided a build log.

TommyMurphyTM1234, 2024-07-05:
If you don't define `RISCV` then the `Makefile` code after this test will use x86 options such as `-march=native`. I suspect that you're not using the `Makefile` correctly for RISC-V cross compilation. A build log would clarify matters quicker than having to guess.

grigohas, 2024-07-05:
yeah i am so sorry but i am new to this. i added riscv=1 and it was compiled correctly thank you.

TommyMurphyTM1234, 2024-07-05:
> yeah i am so sorry but i am new to this. i added riscv=1 and it was compiled correctly thank you.

I presume you mean `RISCV=1`?

This also doesn't seem quite right:
> ```
> make llama-cli CC="riscv64-unknown-linux-gnu-gcc -march=rv64gc -mabi=lp64d" CXX="riscv64-unknown-linux-gnu-g++ -march=rv64gc -mabi=lp64d"
> ```

Seems to me that something like this maye be more appropriate:

> ```
> make llama-cli RISCV=1 RISCV_CROSS_COMPILE=1
> ```
But you'd need to clarify such issues on that site/forum rather than here.

closed 2024-07-05 by TommyMurphyTM1234