riscv-collab/riscv-gnu-toolchain #1506: Question: how to run test suite a second time?

TommyMurphyTM1234, 2024-07-12:
Maybe somebody can help me here.

I have run the test suite once but want to run it again.
However when I do this I just get this:
```
make report-newlib
/home/user/Downloads/riscv-gnu-toolchain-bump-spike/scripts/testsuite-filter gcc newlib /home/user/Downloads/riscv-gnu-toolchain-bump-spike/test/allowlist `find build-gcc-newlib-stage2/gcc/testsuite/ -name *.sum |paste -sd "," -`
                === gcc: Unexpected fails for rv64imafdc lp64d medlow  ===
FAIL: gcc.c-torture/execute/restrict-1.c   -O1  execution test

               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
 rv64imafdc/  lp64d/ medlow |    1 /     1 |    0 /     0 |      - |
make: *** [Makefile:1099: report-gcc-newlib] Error 1
```
How can I do the equivalent of a `make clean` for the test suite so that I can run it afresh a second time?
Are there specific files/directories that I need to delete manually?

Thanks.

cmuellner, 2024-07-12:
Like most other targets, the report target uses stamp files.
In your case, the file is `stamps/check-gcc-newlib`. If you remove it, the tests should be executed again.

TommyMurphyTM1234, 2024-07-12:
Thanks @cmuellner - I had checked the `stamps` directory but when I didn't see any `report-newlib` or similar file I assumed that this mechanism wasn't relevant for the test suite. Thanks for clariifying. 👍

closed 2024-07-12 by TommyMurphyTM1234