riscv-collab/riscv-gnu-toolchain #1507: Bump Spike to latest commit

TommyMurphyTM1234, 2024-07-12:
I'd like to prepare a PR to bump Spike from https://github.com/riscv-software-src/riscv-isa-sim/commit/20a2b6d05a562236a32ca3ce1642a2bdbbd907a0 to the latest commit https://github.com/riscv-software-src/riscv-isa-sim/commit/f7d0dba6012d49cb75b2f175a31a85cf4b2db3b3.

One of the reasons being this:

- https://github.com/riscv-collab/riscv-openocd/issues/1098#issuecomment-2219750314

I bumped Spike done this locally in preparation for creating a PR but I am not clear on how to sanity check that the change is safe.
I have run the test suite for the bare-metal/Newlib toolchain and the results before and after this change are identical:
```
make report-newlib SIM=spike

...

               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
 rv64imafdc/  lp64d/ medlow |    0 /     0 |    0 /     0 |      - |
```
However I'm not clear on how to run the test suite to get results similar to those in the previous "bump Spike" PR which has results for a different set of `arch/abi` combinations:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1466
```
               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
     rv64gc/  lp64d/ medlow |    0 /     0 |    0 /     0 |      - |
  rv32imafc/ ilp32f/ medlow |    0 /     0 |    0 /     0 |      - |
```
And I'm not sure if I should also be running the `make report-linux` tests?

Can anybody give me some guidance on this please?
Thanks.

TommyMurphyTM1234, 2024-07-13:
> And I'm not sure if I should also be running the `make report-linux` tests?

Perhaps not? When I do so I get this:
```
make report-linux SIM=spike 2>&1 | tee report-linux.log

...

FAIL: g++.target/riscv/call-with-empty-struct-int.C execution test

               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
 rv64imafdc/  lp64d/ medlow |20226 /  3893 |10182 /  2582 |18352 /  3093 |
make: *** [Makefile:1106: report-gcc-linux] Error 1
```

TommyMurphyTM1234, 2024-07-14:
> However I'm not clear on how to run the test suite to get results similar to those in the previous "bump Spike" PR which has results for a different set of `arch/abi` combinations:
> 
> * [spike: Bump version from d1efcdf to 20a2b6d #1466](https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1466)
> 
> ```
>                ========= Summary of gcc testsuite =========
>                             | # of unexpected case / # of unique unexpected case
>                             |          gcc |          g++ |     gfortran |
>      rv64gc/  lp64d/ medlow |    0 /     0 |    0 /     0 |      - |
>   rv32imafc/ ilp32f/ medlow |    0 /     0 |    0 /     0 |      - |
> ```
I thought that this might work and perform the tests on the two `arch/abi`s listed above:
```
make distclean && rm -rf installed-tools
./configure --prefix=`pwd`/installed-tools --with-multilib-generator="rv32imafc-ilp32f--" --with-sim=spike
make && make build-sim
make report-newlib SIM=spike 2>&1 | tee report-newlib.log
```
But it didn't and I got this:
```
make report-newlib SIM=spike 2>&1 | tee report-newlib.log
Traceback (most recent call last):
  File "/home/user/Downloads/riscv-gnu-toolchain/scripts/generate_target_board", line 82, in <module>
    sys.exit(main(sys.argv))
  File "/home/user/Downloads/riscv-gnu-toolchain/scripts/generate_target_board", line 59, in main
    target_board = generate_one_target_board(one_arch_abi, "", options)
  File "/home/user/Downloads/riscv-gnu-toolchain/scripts/generate_target_board", line 35, in generate_one_target_board
    abi = arch_and_abi[1]
IndexError: list index out of range
PATH="/home/user/Downloads/riscv-gnu-toolchain/scripts/wrapper/spike:/home/user/Downloads/riscv-gnu-toolchain/scripts:/home/user/Downloads/riscv-gnu-toolchain/installed-tools/bin:/home/user/Downloads/riscv-gnu-toolchain/scripts/wrapper/sed:/home/user/Downloads/riscv-gnu-toolchain/installed-tools/bin:/home/user/.local/bin:/usr/lib/ccache:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/wsl/lib:/mnt/c/Windows/system32:/mnt/c/Windows:/mnt/c/Windows/System32/Wbem:/mnt/c/Windows/System32/WindowsPowerShell/v1.0/:/mnt/c/Windows/System32/OpenSSH/:/mnt/c/Program Files/dotnet/:/mnt/c/Program Files/Meld/:/mnt/c/Program Files/nodejs/:/mnt/c/Program Files/Crucial/Crucial Storage Executive:/mnt/c/Program Files/PuTTY/:/mnt/c/Program Files/usbipd-win/:/mnt/c/Program Files/Git/cmd:/mnt/c/Users/Tommy Murphy/scoop/shims:/mnt/c/Users/Tommy Murphy/AppData/Local/Microsoft/WindowsApps:/mnt/c/Users/Tommy Murphy/AppData/Roaming/npm:/mnt/c/Users/Tommy Murphy/AppData/Local/Programs/Microsoft VS Code/bin:/snap/bin" PK_PATH="/home/user/Downloads/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf/bin/" ARCH_STR="--with-arch=rv64imafdc" make -C build-gcc-newlib-stage2 check-gcc "RUNTESTFLAGS= --target_board=''"
make[1]: Entering directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2'
make[2]: Entering directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc'
rm -rf testsuite/gcc-parallel
make[3]: Entering directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc'
(rootme=`${PWDCMD-pwd}`; export rootme; \
srcdir=`cd ../.././gcc/gcc; ${PWDCMD-pwd}` ; export srcdir ; \
if [ -n "" ] \
   && [ -n "$GCC_RUNTEST_PARALLELIZE_DIR" ] \
   && [ -f testsuite/gcc-parallel/finished ]; then \
  rm -rf testsuite/gcc; \
else \
  cd testsuite/gcc; \
  rm -f tmp-site.exp; \
  sed '/set tmpdir/ s|testsuite$|testsuite/gcc|' \
        < ../../site.exp > tmp-site.exp; \
  /bin/bash ${srcdir}/../move-if-change tmp-site.exp site.exp; \
  EXPECT=`if [ -f ${rootme}/../expect/expect ] ; then echo ${rootme}/../expect/expect ; else echo expect ; fi` ; export EXPECT ; \
  if [ -f ${rootme}/../expect/expect ] ; then  \
    TCL_LIBRARY=`cd .. ; cd ${srcdir}/../tcl/library ; ${PWDCMD-pwd}` ; \
    export TCL_LIBRARY ; \
  fi ; \
  `if [ -f ${srcdir}/../dejagnu/runtest ] ; then echo ${srcdir}/../dejagnu/runtest ; else echo runtest; fi` --tool gcc --target_board=''; \
  if [ -n "$GCC_RUNTEST_PARALLELIZE_DIR" ] ; then \
    touch ${rootme}/testsuite/gcc-parallel/finished; \
  fi ; \
fi )
WARNING: Couldn't find the global config file.
Using /home/user/Downloads/riscv-gnu-toolchain/gcc/gcc/testsuite/lib/gcc.exp as tool init file.
Test run by user on Sun Jul 14 12:29:42 2024
Target is riscv64-unknown-elf
Host   is x86_64-pc-linux-gnu

                === gcc tests ===

Schedule of variations:

                === gcc Summary ===

/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc/xgcc  version 13.2.0 (gc891d8dc23e)

[... 26 lines cut ...]
Using /home/user/Downloads/riscv-gnu-toolchain/gcc/gcc/testsuite/lib/g++.exp as tool init file.
Test run by user on Sun Jul 14 12:29:42 2024
Target is riscv64-unknown-elf
Host   is x86_64-pc-linux-gnu

                === g++ tests ===

Schedule of variations:

                === g++ Summary ===

/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc/xg++  version 13.2.0 (gc891d8dc23e)

make[3]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc'
make[2]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc'
make[1]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2'
mkdir -p stamps/
date > stamps/check-gcc-newlib
/home/user/Downloads/riscv-gnu-toolchain/scripts/testsuite-filter gcc newlib /home/user/Downloads/riscv-gnu-toolchain/test/allowlist `find build-gcc-newlib-stage2/gcc/testsuite/ -name *.sum |paste -sd "," -`

               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
make: *** [Makefile:1099: report-gcc-newlib] Error 1
```

cmuellner, 2024-07-14:
When bumping Spike, I built two newlib toolchains (`--with-arch=rv64gc --with-abi=lp64d` and `--with-arch=rv32imac`).

Steps:
```
./configure --prefix=$PATH_TO_INSTALL --with-sim=spike $CONFIGURE_FLAGS_ARCH
make -j$(nproc)
make -j$(nproc) build-sim
make -j$(nproc) report SIM=spike
```

I then just copied the two results into one test result block in the commit message.

TommyMurphyTM1234, 2024-07-14:
> When bumping Spike, I built two newlib toolchains (`--with-arch=rv64gc --with-abi=lp64d` and `--with-arch=rv32imac`).
>
> Steps:
>
> ```
> ./configure --prefix=$PATH_TO_INSTALL --with-sim=spike [...]

Ah - thanks a lot for clarifying @cmuellner. 
And thanks for dealing with the PR. 
👍

TommyMurphyTM1234, 2024-07-14:
Addressed by this PR:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1509

closed 2024-07-14 by TommyMurphyTM1234