riscv-collab/riscv-gnu-toolchain #1508: Why I can't find compressed instructions in my assembly file with -march=rv32imc -mabi=ilp32

wuliqunyy, 2024-07-13:
Dear, 

I used the following cmd to compile:

>riscv32-unknown-elf-gcc -march=rv32imc -mabi=ilp32 -mcmodel=medlow  -S hello_world.c -o hello_world.s

with c source:

int main() {
    int a = 10;
    int b = 20;
    int c = a + b;
   return 0;
}

And I got assembly:

	.file	"hello_world.c"
	.option nopic
	.attribute arch, "rv32i2p1_m2p0_c2p0"
	.attribute unaligned_access, 0
	.attribute stack_align, 16
	.text
	.align	1
	.globl	main
	.type	main, @function
main:
	addi	sp,sp,-32
	sw	s0,28(sp)
	addi	s0,sp,32
	li	a5,10
	sw	a5,-20(s0)
	li	a5,20
	sw	a5,-24(s0)
	lw	a4,-20(s0)
	lw	a5,-24(s0)
	add	a5,a4,a5
	sw	a5,-28(s0)
	li	a5,0
	mv	a0,a5
	lw	s0,28(sp)
	addi	sp,sp,32
	jr	ra
	.size	main, .-main
	.ident	"GCC: (gc891d8dc23e) 13.2.0"

I should expect instruction like c.addi, right?

And hint where should I look into?

Many thanks,
Liqun

TommyMurphyTM1234, 2024-07-13:
Try this and you should see a mix of "normal" 32-bit and 16-bit compressed instructions:
```
riscv32-unknown-elf-gcc -march=rv32imc -mabi=ilp32 -mcmodel=medlow -c hello_world.c
riscv32-unknown-elf-gcc -d hello_world.o
hello_world.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <main>:
   0:   1101                    addi    sp,sp,-32
   2:   ec22                    sd      s0,24(sp)
   4:   1000                    addi    s0,sp,32
   6:   47a9                    li      a5,10
   8:   fef42623                sw      a5,-20(s0)
   c:   47d1                    li      a5,20
   e:   fef42423                sw      a5,-24(s0)
  12:   fec42783                lw      a5,-20(s0)
  16:   873e                    mv      a4,a5
  18:   fe842783                lw      a5,-24(s0)
  1c:   9fb9                    addw    a5,a5,a4
  1e:   fef42223                sw      a5,-28(s0)
  22:   4781                    li      a5,0
  24:   853e                    mv      a0,a5
  26:   6462                    ld      s0,24(sp)
  28:   6105                    addi    sp,sp,32
  2a:   8082                    ret
```
I think when you use `riscv32-unknown-elf-gcc -march=rv32imc -mabi=ilp32 -mcmodel=medlow -S hello_world.c -o hello_world.s` you're just translating it into high level symbolic assembler and the choice of "normal" or compressed instruction has not yet been made?

wuliqunyy, 2024-07-13:
HI Tommy, 

Thanks for the clarification.
Indeed I see 16 bits instructions in .lss file, and the notation of c.__ was not there either. 

Liqun

closed 2024-07-13 by wuliqunyy

TommyMurphyTM1234, 2024-07-13:
Yes, if the disassembler ever did use `c.*` notation for compressed instructions, I think that that was changed at some stage so that only the "regular" mnemonics were used regardless of whether the instruction is a regular 32-bit instruction or a 16-bit compressed instruction.

aswaterman, 2024-07-16:
Note also that you can use the `-Mno-aliases` command-line argument to `objdump` to force it to print the actual instruction mnemonics, rather than their preferred assembler mnemonics.  (The reason this isn't the default is that it makes the output harder to read, and typically makes it less consistent with the preferred assembly coding style.)

TommyMurphyTM1234, 2024-07-16:
Good point @aswaterman - thanks.
```
riscv32-unknown-elf-objdump -d hello_world.o

hello_world.o:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <main>:
   0:   1101                    addi    sp,sp,-32
   2:   ce22                    sw      s0,28(sp)
   4:   1000                    addi    s0,sp,32
   6:   47a9                    li      a5,10
   8:   fef42623                sw      a5,-20(s0)
   c:   47d1                    li      a5,20
   e:   fef42423                sw      a5,-24(s0)
  12:   fec42703                lw      a4,-20(s0)
  16:   fe842783                lw      a5,-24(s0)
  1a:   97ba                    add     a5,a5,a4
  1c:   fef42223                sw      a5,-28(s0)
  20:   4781                    li      a5,0
  22:   853e                    mv      a0,a5
  24:   4472                    lw      s0,28(sp)
  26:   6105                    addi    sp,sp,32
  28:   8082                    ret

riscv32-unknown-elf-objdump -Mno-aliases -d hello_world.o

hello_world.o:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <main>:
   0:   1101                    c.addi  sp,-32
   2:   ce22                    c.swsp  s0,28(sp)
   4:   1000                    c.addi4spn      s0,sp,32
   6:   47a9                    c.li    a5,10
   8:   fef42623                sw      a5,-20(s0)
   c:   47d1                    c.li    a5,20
   e:   fef42423                sw      a5,-24(s0)
  12:   fec42703                lw      a4,-20(s0)
  16:   fe842783                lw      a5,-24(s0)
  1a:   97ba                    c.add   a5,a4
  1c:   fef42223                sw      a5,-28(s0)
  20:   4781                    c.li    a5,0
  22:   853e                    c.mv    a0,a5
  24:   4472                    c.lwsp  s0,28(sp)
  26:   6105                    c.addi16sp      sp,32
  28:   8082                    c.jr    ra
```