riscv-collab/riscv-gnu-toolchain #1510: Question: how to run the test suite to test more than one `arch/abi`?

TommyMurphyTM1234, 2024-07-14:
See here:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1507

As per that issue this PR suggests that the test suite can be run once to test more than one `arch/abi`:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1466
```
               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
     rv64gc/  lp64d/ medlow |    0 /     0 |    0 /     0 |      - |
  rv32imafc/ ilp32f/ medlow |    0 /     0 |    0 /     0 |      - |
```
But when I run it it only tests one - the toolchain default `arch/abi`:
```
make report-newlib SIM=spike

...

               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
 rv64imafdc/  lp64d/ medlow |    0 /     0 |    0 /     0 |      - |
```
The only way that I have found to test both `rv64imafdc/lp64d` and `rv32imafc/lp32f` is to run the test suite twice:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain

./configure --prefix=`pwd`/installed-tools --with-sim=spike
make 
make build-sim
make report-newlib SIM=spike

...

               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
 rv64imafdc/  lp64d/ medlow |    0 /     0 |    0 /     0 |      - |

make distclean 
rm -rf installed-tools

./configure --prefix=`pwd`/installed-tools --with-arch=rv32imafc --with-abi=ilp32d --with-sim=spike
make
make build-sim
make report-newlib SIM=spike

...

               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
 rv32imafc/  ilp32f/ medlow |    0 /     0 |    0 /     0 |      - |
```
I also tried running the test suite on a multilib toolchain but that fails:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1507#issuecomment-2227312656

How can I run the test suite once to test the two `arch/abi` combinations in question?

cmuellner, 2024-07-14:
Running the test suite for multiple arch/abi combinations works at least for the default multilib Linux toolchain (`--enable-multilib --enable-linux`).

Afaik newlib does not support multilib (for sure, this repo does not support that). So, the only way left to test multiple arch/abi combinations is (as you have already found out) to build multiple toolchains and test them.

TommyMurphyTM1234, 2024-07-14:
> Running the test suite for multiple arch/abi combinations works at least for the default multilib Linux toolchain (`--enable-multilib --enable-linux`).

But, as far as I can see, running `make report-linux` against Spike doesn't really make sense?
At least, it doesn't work:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1507#issuecomment-2226849389

> Afaik newlib does not support multilib (for sure, this repo does not support that). 

What? Of course it does - either via `--enable-multilib` for the default set of multilibs defined by the generated `Makefile`:

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/0b4a59812061728bc7a612986b00a8c9ef7b97a4/configure.ac#L160-L162

or via `--with-multilib-generator` for a custom set:

- https://github.com/riscv-collab/riscv-gnu-toolchain/tree/master?tab=readme-ov-file#build-with-customized-multi-lib-configure

> So, the only way left to test multiple arch/abi combinations is (as you have already found out) to build multiple toolchains and test them.

Yes - by trial and error that's what I discovered. Thanks.

closed 2024-07-14 by TommyMurphyTM1234

cmuellner, 2024-07-15:
> What? Of course it does - either via `--enable-multilib` for the default set of multilibs defined by the generated `Makefile`:
>
> * [...]

Ah, you are right:

```
               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
      rv32i/  ilp32/ medlow |    0 /     0 |    0 /     0 |      - |
    rv32iac/  ilp32/ medlow |    0 /     0 |    0 /     0 |      - |
     rv32im/  ilp32/ medlow |    0 /     0 |    0 /     0 |      - |
   rv32imac/  ilp32/ medlow |    0 /     0 |    0 /     0 |      - |
  rv32imafc/ ilp32f/ medlow |    0 /     0 |    0 /     0 |      - |
   rv64imac/   lp64/ medlow |    0 /     0 |    0 /     0 |      - |
 rv64imafdc/  lp64d/ medlow |    0 /     0 |    0 /     0 |      - |
```

I never looked into the details here, but I knew that we don't build newlib the same way we build glibc (by comparing `stamps/build-newlib` with `stamps/build-glibc-linux-%`).

Also the configure flags for multilib newlib don't show any hints about multilib: `$ /home/cm/src/riscv-gnu-toolchain/riscv-rvi/newlib/configure --target=riscv64-unknown-elf --prefix=/opt/riscv-rvi --enable-newlib-io-long-double --enable-newlib-io-long-long --enable-newlib-io-c99-formats --enable-newlib-register-fini CFLAGS_FOR_TARGET=-O2 -D_POSIX_MODE -ffunction-sections -fdata-sections    -mcmodel=medlow CXXFLAGS_FOR_TARGET=-O2 -D_POSIX_MODE -ffunction-sections -fdata-sections    -mcmodel=medlow`.

However, `$installdir/riscv64-unknown-elf/lib/*` shows all the necessary files.

So, sorry for my misleading sentence.

TommyMurphyTM1234, 2024-07-15:
Sorry, I thought that you were referring to the bare-metal/Newlib toolchain not supporting multilibs which it obviously does. But it seems that you were referring to the test suite not testing all Newlib multilib `arch/abi` combinations? But it seems that this works too? The odd thing is that when I tried it it failed. 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1507#issuecomment-2227312656

Perhaps it only works when the toolchain is configured/built with `--enable-multilib` but not when `--with-multilib-generator` is used? I must try this later...

TommyMurphyTM1234, 2024-07-15:
> Perhaps it only works when the toolchain is configured/built with `--enable-multilib` but not when `--with-multilib-generator` is used? 

That seems to be the case alright.

Using `--enable-multilib`:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --enable-multilib --with-sim=spike
make 
make build-sim

make report-newlib SIM=spike

...

                === gcc tests ===

Schedule of variations:
    riscv-sim/-march=rv32i/-mabi=ilp32/-mcmodel=medlow
    riscv-sim/-march=rv32iac/-mabi=ilp32/-mcmodel=medlow
    riscv-sim/-march=rv32im/-mabi=ilp32/-mcmodel=medlow
    riscv-sim/-march=rv32imac/-mabi=ilp32/-mcmodel=medlow
    riscv-sim/-march=rv32imafc/-mabi=ilp32f/-mcmodel=medlow
    riscv-sim/-march=rv64imac/-mabi=lp64/-mcmodel=medlow
    riscv-sim/-march=rv64imafdc/-mabi=lp64d/-mcmodel=medlow

Running target riscv-sim/-march=rv32i/-mabi=ilp32/-mcmodel=medlow
Using /home/user/Downloads/riscv-gnu-toolchain/installed-tools/share/dejagnu/baseboards/riscv-sim.exp as board description file for target.
Using /home/user/Downloads/riscv-gnu-toolchain/installed-tools/share/dejagnu/config/sim.exp as generic interface file for target.
Using /home/user/Downloads/riscv-gnu-toolchain/installed-tools/share/dejagnu/baseboards/basic-sim.exp as board description file for target.
Using /home/user/Downloads/riscv-gnu-toolchain/gcc/gcc/testsuite/config/default.exp as tool-and-target-specific interface file.
Running /home/user/Downloads/riscv-gnu-toolchain/gcc/gcc/testsuite/gcc.c-torture/compile/compile.exp ...

...
```
Using `--with-multilib-generator`:
```
make distclean 
rm -rf installed-tools
./configure --prefix=`pwd`/installed-tools --with-multilib-generator="rv32imafc-ilp32f--" --with-sim=spike
make
make build-sim

make report-newlib SIM=spike
rm -rf stamps/build-dejagnu build-dejagnu
mkdir build-dejagnu
cd build-dejagnu && /home/user/Downloads/riscv-gnu-toolchain/dejagnu/configure \
        --prefix=/home/user/Downloads/riscv-gnu-toolchain/installed-tools
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... /usr/bin/gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking whether make sets $(MAKE)... (cached) yes
checking for gawk... (cached) /usr/bin/gawk
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
[... 118 lines cut ...]
Test run by user on Mon Jul 15 13:00:51 2024
Target is riscv64-unknown-elf
Host   is x86_64-pc-linux-gnu

                === g++ tests ===

Schedule of variations:

                === g++ Summary ===

/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc/xg++  version 13.2.0 (gc891d8dc23e)

make[3]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc'
make[2]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2/gcc'
make[1]: Leaving directory '/home/user/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage2'
mkdir -p stamps/
date > stamps/check-gcc-newlib
/home/user/Downloads/riscv-gnu-toolchain/scripts/testsuite-filter gcc newlib /home/user/Downloads/riscv-gnu-toolchain/test/allowlist `find build-gcc-newlib-stage2/gcc/testsuite/ -name *.sum |paste -sd "," -`

               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
make: *** [Makefile:1099: report-gcc-newlib] Error 1
```
I don't know why the latter fails?