riscv-collab/riscv-gnu-toolchain #1516: Fails to build after adding new trigonometric instructions On spike-riscv

baul-iisc, 2024-07-19:
Hello Team,
Before adding these fsin.s and fcos.s instructions, I had added those mod, fact, gcd examples given in different pointers. And when I am commenting these lines for fsin and fcos instruction, it is going through smoothly. Can you please help me to figure out where I am missing for fsin and fcos instructions?
I have modified the rs2==0 and changed the following: 
1. I have added he riscv_opcodes array in the binutils/opcodes/riscv-opc.c as follows:
{"fsin.s", 0, INSN_CLASS_F_INX, "d,s", MATCH_FSIN, MASK_FSIN, match_opcode, 0},
{"fcos.s", 0, INSN_CLASS_F_INX, "d,s", MATCH_FCOS, MASK_FCOS, match_opcode, 0},

2. And in the binutils/include/opcode/riscv-opc.h I have added the definitions for the MATCH/MASK as follows:

#define MATCH_FSIN 0x30000053
#define MASK_FSIN 0xfff0007f
#define MATCH_FCOS 0x38000053
#define MASK_FCOS 0xfff0007f
And declared these instructions as well:

DECLARE_INSN(fsin_s, MATCH_FSIN, MASK_FSIN)
DECLARE_INSN(fcos_s, MATCH_FCOS, MASK_FCOS)

Rationale behind FSIN Instruction MATCH_FSIN 0x30000053

For the fsin.s instruction:

    funct7 = 0x06 (bits 25-31)
    rs2 = 0 (bits 20-24)
    funct3 = 0x0 (bits 12-14)
    opcode = 0x53 (bits 0-6)

with these also when I am trying to build and install the riscv-gnu-toolchain again with the command as follows:
./configure --prefix=$RISCV --host=riscv64-unknown-elf --with-arch=rv64gcv --with-abi=lp64d --with-sim=spike --enable-multilib
it is not running throwing same errors shown previously as follows:
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc
checking whether the C compiler works... no
configure: error: in `/data/home/chandraboul/development/riscv-gnu-toolchain':
configure: error: C compiler cannot create executables
See `config.log' for more details
And in the config.log the identified error is showing as
cat config.log
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by riscv-toolchain configure 1.0, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ./configure --prefix=/data/home/chandraboul/development/riscv-spike --host=riscv64-unknown-elf --with-arch=rv64gcv --with-abi=lp64d --with-sim=spike --enable-multilib

## --------- ##
## Platform. ##
## --------- ##

hostname = resiliente-1949
uname -m = x86_64
uname -r = 6.2.0-39-generic
uname -s = Linux
uname -v = #40~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 16 10:53:04 UTC 2

/usr/bin/uname -p = x86_64
/bin/uname -X     = unknown

/bin/arch              = x86_64
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /data/home/chandraboul/development/riscv-spike/bin
PATH: /data/home/chandraboul/development/riscv-spike/riscv64-unknown-elf/bin
PATH: /data/home/chandraboul/development/riscv-spike/bin
PATH: /usr/local/sbin
[... 330 lines cut ...]
with_gdb_src='$(srcdir)/gdb'
with_glibc_src='$(srcdir)/glibc'
with_guile=''
with_linux_headers_src='$(srcdir)/linux-headers/include'
with_llvm_src='$(srcdir)/llvm'
with_musl_src='$(srcdir)/musl'
with_newlib_src='$(srcdir)/newlib'
with_pk_src='$(srcdir)/pk'
with_qemu_src='$(srcdir)/qemu'
with_spike_src='$(srcdir)/spike'
with_system_zlib='--with-system-zlib'

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "riscv-toolchain"
#define PACKAGE_TARNAME "riscv-toolchain"
#define PACKAGE_VERSION "1.0"
#define PACKAGE_STRING "riscv-toolchain 1.0"
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL ""

configure: exit 0

TommyMurphyTM1234, 2024-07-19:
> Hello Team, Before adding these fsin.s and fcos.s instructions

This issue deals with adding a custom `fsin.s` instruction do might be of relevance here?

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1485

baul-iisc, 2024-07-20:
From some other place I got to know that I need to change the opcodes as "d,s,t" or "d,a". Even if the "t" is unused, that parameter is telling the RISC-V system how to parse the instruction. If we use the wrong letters, the instruction cannot be parsed.  So I had changed it to "d,s,t" and with that the build and installation process was successful, but it is not recognizing the fsin.s instructions when I am trying the following:
$ riscv64-unknown-elf-as -o test.o -a <<< "fsin.s fa5, fa5, x0"

{standard input}: Assembler messages:

{standard input}:1: Error: unrecognized opcode `fsin.s fa5,fa5,x0'

GAS LISTING  			page 1

   1              	fsin.s fa5,fa5,x0

GAS LISTING  			page 2

NO DEFINED SYMBOLS

NO UNDEFINED SYMBOLS

 But if I try with gcd as follows: it is showing properly: 
$ riscv64-unknown-elf-as -o test.o -a <<< "gcd a5, a5, x0"

GAS LISTING  			page 1

   1 0000 A7E70700 	gcd a5,a5,x0

GAS LISTING  			page 2

DEFINED SYMBOLS

    {standard input}:1      .text:0000000000000000 $xrv64i2p1_m2p0_a2p1_f2p2_d2p2_zicsr2p0_zifencei2p0_zmmul1p0

NO UNDEFINED SYMBOLS

CAn you please guide me if it should be "g,s,t" or "g,a" or there are some other errors also in my instructions definition?

TommyMurphyTM1234, 2024-07-20:
You don't explain exactly what changes to the toolchain were made that resulted in the behaviour described above for the failing `fsin.s` and working `gcd` tests.

baul-iisc, 2024-07-20:
I have added six new instructions as follows modulus(mod), gcd, fact, fsin.s, fcos.s and ftan.s. Once the build process over I am giving the following command to check these newly added instructions and I could see that the three instructon mod, gcd and fact is showing output as expected, but fsin.s, fcos.s and ftan.s instruction is throwing Error: illegal operands. I had defined the fsin.s, fcos.s and ftan.s similar to single precision fsqrt.s instructions. Please suggest/guide me if I am making any mistake in these process so that these trigonometric instructions work properly:

(i) In the first step i have open up binutils/opcodes/riscv-opc.c and added these six instructions to the riscv_opcodes array. 

{"mod",      0, INSN_CLASS_I,       "d,s,t",         MATCH_MOD, MASK_MOD, match_opcode, 0 },
{"gcd",      0, INSN_CLASS_I,       "d,s,t",         MATCH_GCD, MASK_GCD, match_opcode, 0 },
{"fact",      0, INSN_CLASS_I,       "d,a",         MATCH_FACT, MASK_FACT, match_opcode, 0 },
// Single-precision floating-point instruction subset similar to fsqrt.s
{"fsin.s",    0, INSN_CLASS_F_INX,   "d,s",       MATCH_FSIN_S|MASK_RM, MASK_FSIN_S|MASK_RM, match_opcode, 0 },
{"fsin.s",    0, INSN_CLASS_F_INX,   "d,s,m",     MATCH_FSIN_S, MASK_FSIN_S, match_opcode, 0 },
{"fcos.s",    0, INSN_CLASS_F_INX,   "d,s",       MATCH_FCOS_S|MASK_RM, MASK_FCOS_S|MASK_RM, match_opcode, 0 },
{"fcos.s",    0, INSN_CLASS_F_INX,   "d,s,m",     MATCH_FCOS_S, MASK_FCOS_S, match_opcode, 0 },
{"ftan.s",    0, INSN_CLASS_F_INX,   "d,s",       MATCH_FTAN_S|MASK_RM, MASK_FTAN_S|MASK_RM, match_opcode, 0 },
{"ftan.s",    0, INSN_CLASS_F_INX,   "d,s,m",     MATCH_FTAN_S, MASK_FTAN_S, match_opcode, 0 },

(ii) Next, I have open up binutils/include/opcode/riscv-opc.h and added in definitions for the MATCH/MASK:
#define MATCH_GCD 0x6027
#define MASK_GCD 0xfe00707f
#define MATCH_FACT 0x2b
#define MASK_FACT 0x7f
#define MATCH_MOD 0x200006b
#define MASK_MOD 0xfe00707f
#define MATCH_FSIN_S 0x59000053
#define MASK_FSIN_S  0xfff0007f
#define MATCH_FCOS_S 0x5b000053
#define MASK_FCOS_S  0xfff0007f
#define MATCH_FTAN_S 0x5d000053
#define MASK_FTAN_S  0xfff0007f

Further down, where we are declaring our instructions, I have added each of them:

DECLARE_INSN(mod, MATCH_MOD, MASK_MOD)
DECLARE_INSN(gcd, MATCH_GCD, MASK_GCD)
DECLARE_INSN(fact, MATCH_FACT, MASK_FACT)
DECLARE_INSN(fsin_s, MATCH_FSIN_S, MASK_FSIN_S)
DECLARE_INSN(fcos_s, MATCH_FCOS_S, MASK_FCOS_S)
DECLARE_INSN(ftan_s, MATCH_FTAN_S, MASK_FTAN_S)

(iii) Then after executing the following configuration script built and build and installed the riscv-gnu-toolchain as per the commands you suggested earlier:

# then run the config command as follows: 
./configure --prefix=$RISCV --host=riscv64-unknown-elf --with-arch=rv64gcv --with-abi=lp64d --with-sim=spike --enable-multilib
# next execute the build and installed command
sudo make -j$(nproc) && sudo make build-sim

(iv)Once the build process over I am giving the following command to check these newly added instructions and I could see that the three instructon mod, gcd and fact is showing output as expected, but fsin.s, fcos.s and ftan.s instruction is throwing output as illegal operands. I had defined the fsin.s, fcos.s and ftan.s similar to single precision fsqrt.s instructions:
$ echo "mod a5, a5, a3" | riscv64-unknown-elf-as -o test.o -a
GAS LISTING  			page 1

   1 0000 EB87D702 	mod a5,a5,a3

GAS LISTING  			page 2

DEFINED SYMBOLS
    {standard input}:1      .text:0000000000000000 $xrv64i2p1_m2p0_a2p1_f2p2_d2p2_zicsr2p0_zifencei2p0_zmmul1p0

NO UNDEFINED SYMBOLS
$ echo "gcd a5, a5, a3" | riscv64-unknown-elf-as -o test.o -a
GAS LISTING  			page 1

   1 0000 A7E7D700 	gcd a5,a5,a3

GAS LISTING  			page 2

DEFINED SYMBOLS
    {standard input}:1      .text:0000000000000000 $xrv64i2p1_m2p0_a2p1_f2p2_d2p2_zicsr2p0_zifencei2p0_zmmul1p0

NO UNDEFINED SYMBOLS

$ echo "fact a5, 0x0" | riscv64-unknown-elf-as -o test.o -a
GAS LISTING  			page 1

   1 0000 AB070000 	fact a5,0x0

GAS LISTING  			page 2

DEFINED SYMBOLS
    {standard input}:1      .text:0000000000000000 $xrv64i2p1_m2p0_a2p1_f2p2_d2p2_zicsr2p0_zifencei2p0_zmmul1p0

NO UNDEFINED SYMBOLS

$ echo "fsqrt.s fa5, fa5" | riscv64-unknown-elf-as -o test.o -a
GAS LISTING  			page 1

   1 0000 D3F70758 	fsqrt.s fa5,fa5

GAS LISTING  			page 2

DEFINED SYMBOLS
    {standard input}:1      .text:0000000000000000 $xrv64i2p1_m2p0_a2p1_f2p2_d2p2_zicsr2p0_zifencei2p0_zmmul1p0

NO UNDEFINED SYMBOLS
$ echo "fsin.s fa5, fa5" | riscv64-unknown-elf-as -o test.o -a
{standard input}: Assembler messages:
{standard input}:1: Error: illegal operands `fsin.s fa5,fa5'
GAS LISTING  			page 1

   1              	fsin.s fa5,fa5

GAS LISTING  			page 2

NO DEFINED SYMBOLS

NO UNDEFINED SYMBOLS

$ echo "fcos.s fa5, fa5" | riscv64-unknown-elf-as -o test.o -a
{standard input}: Assembler messages:
{standard input}:1: Error: illegal operands `fcos.s fa5,fa5'
GAS LISTING  			page 1

   1              	fcos.s fa5,fa5

GAS LISTING  			page 2

NO DEFINED SYMBOLS

NO UNDEFINED SYMBOLS
$ echo "ftan.s fa5, fa5" | riscv64-unknown-elf-as -o test.o -a
{standard input}: Assembler messages:
{standard input}:1: Error: illegal operands `ftan.s fa5,fa5'
GAS LISTING  			page 1

   1              	ftan.s fa5,fa5

GAS LISTING  			page 2

NO DEFINED SYMBOLS

NO UNDEFINED SYMBOLS

pz9115, 2024-07-20:
> I have added six new instructions as follows modulus(mod), gcd, fact, fsin.s, fcos.s and ftan.s. Once the build process over I am giving the following command to check these newly added instructions [...]

When you try to use floating register in your instruction, you should change the operands into upcast letters to distinguish with use int registers. So it means you need to change your instructions define in binutils/opcodes/riscv-opc.c, it has the same define as fsqrt.s:

                                                              //here use upcast operands
{"fsqrt.s",    0, INSN_CLASS_ZFH_INX,   "D,S",       MATCH_FSQRT_H|MASK_RM, MASK_FSQRT_H|MASK_RM, match_opcode, 0 },
{"fsqrt.s",    0, INSN_CLASS_ZFH_INX,   "D,S,m",     MATCH_FSQRT_H, MASK_FSQRT_H, match_opcode, 0 },

The 'D' means termination register, and 'S' means source register, 'm' means to have a mask RM, all these letters defined in binutils/gas/config/tc-riscv.c, you can check them in function 'validate_riscv_insn'. Hope this help you.

```
 // Single-precision floating-point instruction subset similar to fsqrt.s 
{"fsin.s", 0, INSN_CLASS_F_INX, "D,S", MATCH_FSIN_S|MASK_RM, MASK_FSIN_S|MASK_RM, match_opcode, 0 }, 
{"fsin.s", 0, INSN_CLASS_F_INX, "D,S,m", MATCH_FSIN_S, MASK_FSIN_S, match_opcode, 0 }, 
{"fcos.s", 0, INSN_CLASS_F_INX, "D,S", MATCH_FCOS_S|MASK_RM, MASK_FCOS_S|MASK_RM, match_opcode, 0 }, 
{"fcos.s", 0, INSN_CLASS_F_INX, "D,S,m", MATCH_FCOS_S, MASK_FCOS_S, match_opcode, 0 }, 
{"ftan.s", 0, INSN_CLASS_F_INX, "D,S", MATCH_FTAN_S|MASK_RM, MASK_FTAN_S|MASK_RM, match_opcode, 0 }, 
{"ftan.s", 0, INSN_CLASS_F_INX, "D,S,m", MATCH_FTAN_S, MASK_FTAN_S, match_opcode, 0 },
```

> 1 fsin.s fa5,fa5
>
> GAS LISTING page 2
>
> NO DEFINED SYMBOLS
>
> NO UNDEFINED SYMBOLS
>
> $ echo "fcos.s fa5, fa5" | riscv64-unknown-elf-as -o test.o -a {standard input}: Assembler messages: {standard [...]

baul-iisc, 2024-07-21:
Thank you for pin pointing my implementation issue. After these changes in the riscv-opc.c and riscv-opc.h its working now.
$ echo "fsin.d fa5, fa5" | riscv64-unknown-elf-as -o test.o -a
GAS LISTING  			page 1

   1 0000 D3F70753 	fsin.d fa5,fa5

GAS LISTING  			page 2

DEFINED SYMBOLS
    {standard input}:1      .text:0000000000000000 $xrv64i2p1_m2p0_a2p1_f2p2_d2p2_zicsr2p0_zifencei2p0_zmmul1p0

NO UNDEFINED SYMBOLS
$ echo "fcos.d fa5, fa5" | riscv64-unknown-elf-as -o test.o -a
GAS LISTING  			page 1

   1 0000 D3F70755 	fcos.d fa5,fa5

GAS LISTING  			page 2

DEFINED SYMBOLS
    {standard input}:1      .text:0000000000000000 $xrv64i2p1_m2p0_a2p1_f2p2_d2p2_zicsr2p0_zifencei2p0_zmmul1p0

NO UNDEFINED SYMBOLS
$ echo "ftan.d fa5, fa5" | riscv64-unknown-elf-as -o test.o -a
GAS LISTING  			page 1

   1 0000 D3F70757 	ftan.d fa5,fa5

GAS LISTING  			page 2

DEFINED SYMBOLS
    {standard input}:1      .text:0000000000000000 $xrv64i2p1_m2p0_a2p1_f2p2_d2p2_zicsr2p0_zifencei2p0_zmmul1p0

NO UNDEFINED SYMBOLS
$ echo "fsin.s a5, a5" | riscv64-unknown-elf-as -o test.o -a
GAS LISTING  			page 1

   1 0000 D3F70759 	fsin.s a5,a5

GAS LISTING  			page 2

DEFINED SYMBOLS
    {standard input}:1      .text:0000000000000000 $xrv64i2p1_m2p0_a2p1_f2p2_d2p2_zicsr2p0_zifencei2p0_zmmul1p0

NO UNDEFINED SYMBOLS
$ echo "fcos.s a5, a5" | riscv64-unknown-elf-as -o test.o -a
GAS LISTING  			page 1

   1 0000 D3F7075B 	fcos.s a5,a5

GAS LISTING  			page 2

DEFINED SYMBOLS
    {standard input}:1      .text:0000000000000000 $xrv64i2p1_m2p0_a2p1_f2p2_d2p2_zicsr2p0_zifencei2p0_zmmul1p0

NO UNDEFINED SYMBOLS
$ echo "ftan.s a5, a5" | riscv64-unknown-elf-as -o test.o -a
GAS LISTING  			page 1

   1 0000 D3F7075D 	ftan.s a5,a5

GAS LISTING  			page 2

DEFINED SYMBOLS
    {standard input}:1      .text:0000000000000000 $xrv64i2p1_m2p0_a2p1_f2p2_d2p2_zicsr2p0_zifencei2p0_zmmul1p0

NO UNDEFINED SYMBOLS

TommyMurphyTM1234, 2024-07-21:
> Thank you for pin pointing my implementation issue. After these changes in the riscv-opc.c and riscv-opc.h its working now.

Since the original question/problem has been addressed, I'm closing the issue.

closed 2024-07-21 by TommyMurphyTM1234