riscv-collab/riscv-gnu-toolchain #1517: GCC Version

Adam11072000, 2024-07-19:
Hey guys,

i need the toolchain branch/commit where the gcc  version was 11.2.0 and it was stable.

thanks

TommyMurphyTM1234, 2024-07-19:
What about the CI binary builds? 

- https://github.com/riscv-collab/riscv-gnu-toolchain/releases?page=7

GCC 11.2.0 was released on July 28 2021 so a CI build after that date may be what you need?

- https://gcc.gnu.org/gcc-11/

TommyMurphyTM1234, 2024-07-19:
There's also the xPack binary builds but only 11.3.0-1 is listed there for some reason:

- https://xpack.github.io/dev-tools/riscv-none-elf-gcc/
  - https://xpack.github.io/dev-tools/riscv-none-elf-gcc/releases/

ilg-ul, 2024-07-20:
>  only 11.3 release listed there is 11.3.0-1 for some reason (maybe @ilg-ul recalls why?):

There were some issues with building GCC 11 on macOS, and at that time I did not have a clear idea where to get the required patches from.

Now I updated the workflow, got all required patches for all current versions (11.5, 12.4, 13.2, 14.1) and I hope I can make a new set of releases during the next few days.

TommyMurphyTM1234, 2024-07-20:
@Adam11072000 - did any of the above address your issue?

TommyMurphyTM1234, 2024-07-21:
> well, i do not need a new build. can you provide me with a 11.2 build?

Not sure who you're addressing that to but just to say that I can't. Did you check the CI release builds to see if there's a suitable 11.2.0 there? If not then you'll probably have to do as I suggested and check the commit/PR/release history and try to pull all the relevant historical components and build them. I suspect that this may not be simple/straightforward.

> somehow guys using openTitan built their whole system upon gcc 11.2.

I don't know anything about that project but maybe they can advise you on how/where to get a RISC-V GCC 11.2.0 toolchain?

But I would really question the merits and advisability of using such an old version of the RISC-V toolchain given how much has changed and how many issues addressed in the meantime.

Finally, since this is not really an issue with the `riscv-gnu-toolchain` I'm inclined to think that it should probably be closed.

Adam11072000, 2024-07-21:
i will try to check the releases.
thanks!

TommyMurphyTM1234, 2024-07-25:
Any update @Adam11072000?
Did you find a suitable 11.2.0 toolchain in the CI releases or elsewhere? 
Did you check with the OpenTitan community about this issue?

Adam11072000, 2024-07-25:
the issue was in my makefiles.
i succeeded in compiling with no relation to the toolchain version
thanks

TommyMurphyTM1234, 2024-07-25:
Thanks, so you can close the issue now, I presume?

closed 2024-08-01 by TommyMurphyTM1234